ros2/ros2 #539: rclcpp failed to publish message

realdream, 2018-07-13:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04
- Installation type:
  - from dource 
- Version or commit hash:
  - release bouncy commit id bd2e0194
- DDS implementation:
  -  Fast-RTPS
- Client library (if applicable):
  - rclcpp
- network 
  - 5.8G wireless router

#### Steps to reproduce issue

on first PC
```
ros2 run image_tools showimage
```
on second PC
```
ros2 run image_tools cam2image -b
```

#### result 
the cam2image node crash after a while
```
[INFO] [cam2image]: Publishing image #6820
[INFO] [cam2image]: Publishing image #6821
[INFO] [cam2image]: Publishing image #6822
[INFO] [cam2image]: Publishing image #6823
[INFO] [cam2image]: Publishing image #6824
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  failed to publish message: cannot publish data, at /home/gitlab-runner/builds/dd724f49/1/ROS/ros2_repo/ros2_xenial_x86_64_all/src/ros2/rmw_fastrtps/rmw_fastrtps_cpp/src/rmw_publish.cpp:53, at /home/gitlab-runner/builds/dd724f49/1/ROS/ros2_repo/ros2_xenial_x86_64_all/src/ros2/rcl/rcl/src/rcl/publisher.c:241
```

sloretz, 2018-07-19:
I'm unable to reproduce this on bionic.

It comes from [rmw_publish.cpp:53](https://github.com/ros2/rmw_fastrtps/blob/7d3ff15e40999d878fb67104987b3d654d48db17/rmw_fastrtps_cpp/src/rmw_publish.cpp#L53) which happens when [eprosima::fastrtps::PublisherImpl::create_new_change_with_params()](https://github.com/eProsima/Fast-RTPS/blob/54f563a697e7de5e3c924a99fa1b2ce3877f69e7/src/cpp/publisher/PublisherImpl.cpp#L83) returns false.

@richiware may I ask if you have any ideas about why a this call might have failed? If it can be reproduced, what info would help determine the cause?

richiware, 2018-07-20:
By default in ros2 the one in charge of sending data in fastrtps is an asynchronous thread. Do you know which is the 
depth of the history?  If history's depth is 1 and in the moment of writing a new sample the asynchronous thread didn't 
send the previous one then there is no space in history to store the new sample and the write function fails.

We are pending on implement a timeout for this cases, a maximum blocking time where write function waits until there is 
space on the history or fails.

x-integrator, 2018-07-24:
It's crashing also on Win10 and after 5001 "Burgers", but w/o any message, simply crashing. The problem is perhaps somewhere else - it eats memory, about 250kB per Burger. It looks like new image rendering is always into newly allocated memory. On the screenshot shortly before crash with allocated 1GB RAM, when started there is about 20MB.

[image]

dirk-thomas, 2018-08-02:
> If history's depth is 1 and in the moment of writing a new sample the asynchronous thread didn't
send the previous one then there is no space in history to store the new sample and the write function fails.

@richiware I would have expected the pending sample to be overwritten (meaning "lost" / never send) )when a new one is being published. Why should the call itself fail?

richiware, 2018-08-07:
@dirk-thomas You're right. I don't know what I was thinking the other day. Using KEEP_LAST is the expected behaviour you 
described.  Using KEEP_ALL it's the behaviour I was decribing and it doesn't depends on history's depth, but in resource 
limits max_samples.

I was trying to reproduce the issue but I couldn't. I will continue.

dirk-thomas, 2018-08-17:
> I was trying to reproduce the issue but I couldn't. I will continue.

@richiware Any update on this issue?

richiware, 2018-08-27:
Sorry for the delay answering. We were on vacations. Before leaving the office I wasn't able to reproduce this issue.  
This week we will work on this. @miguelcompany will help me on this.

MiguelCompany, 2018-08-31:
We found the main issue is that samples are not automoatically removed from the history in this case (1 reliable publisher - 1 best effort subscriber), and started eProsima/Fast-RTPS#249 for this and other cases.

dirk-thomas, 2018-09-03:
@realdream can you please retry it with the latest version of FastRTPS to confirm that eProsima/Fast-RTPS#249 fixes the problem for you.

dirk-thomas, 2018-09-11:
I will go ahead and close this issue assuming the problem has been resolved.

closed 2018-09-11 by dirk-thomas

quhezheng, 2019-09-15:
> We found the main issue is that samples are not automoatically removed from the history in this case (1 reliable publisher - 1 best effort subscriber), and started [eProsima/Fast-RTPS#249](https://github.com/eProsima/Fast-RTPS/pull/249) for this and other cases.

We are suffering from the same issue on latest Fast-RTPS (https://github.com/ros2/Fast-RTPS).
So the issue is yet fixed

MiguelCompany, 2019-09-16:
@quhezheng 

> latest Fast-RTPS (https://github.com/ros2/Fast-RTPS)

The official repo for Fast-RTPS is https://github.com/eProsima/Fast-RTPS

> * Operating System:
>   Ubuntu 16.04  & QNX (no different form Ubuntu)
> * Installation type:
>   from source
> * Version or commit hash:
>   release crystal

Crystal release is pointing to eProsima/Fast-RTPS [v1.7.2](https://github.com/eProsima/Fast-RTPS/tree/v1.7.2), which includes eProsima/Fast-RTPS#350 that solves a communication issue on QNX.

This issue is not reproduced on crystal release. A similar issue is present on master (see ros2/rmw_fastrtps#317), but it is not due to the reasons discovered on this one.

dirk-thomas, 2019-09-16:
> latest Fast-RTPS (https://github.com/ros2/Fast-RTPS)

I just deleted the obsolete fork.

akshaysarvesh25, 2019-11-06:
I faced the same issue with FastRTPS. I was trying to send an array of size : 250000 Float64 values (2000 kbytes) from one computer to another (both running ROS2-Dashing) and faced the same error : `terminate called after throwing an instance of 'rclcpp::exceptions::RCLError' what(): failed to publish message: cannot publish data, at /home/akshay/ros2_ws/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/rmw_publish.cpp:52, at /home/akshay/ros2_ws/src/ros2/rcl/rcl/src/rcl/publisher.c:257`

The issue was solved by reducing the size of the data. The data was an array of size : 1500 Float64 values (12kbytes).

realdream, 2020-06-10:
@dirk-thomas 
this issue can still be reproduce in latest foxy release in docker....., seems not fully fixed
```
[INFO] [1591768725.981513109] [cam2image]: Publishing image #30094
[INFO] [1591768725.993903670] [cam2image]: Publishing image #30095
[INFO] [1591768725.995653899] [cam2image]: Publishing image #30096
[INFO] [1591768726.000690989] [cam2image]: Publishing image #30097
[INFO] [1591768726.012872481] [cam2image]: Publishing image #30098
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  failed to publish message: cannot publish data, at /tmp/binarydeb/ros-foxy-rmw-fastrtps-shared-cpp-1.0.1/src/rmw_publish.cpp:53, at /tmp/binarydeb/ros-foxy-rcl-1.1.5/src/rcl/publisher.c:291

```

realdream, 2020-06-10:
@dirk-thomas should we reopen this issue for further verification? seems easier reproduce over two pc through wifi than over local loopback. A temporary fix is add "try catch" for each publishing.

dirk-thomas, 2020-07-16:
@MiguelCompany Any update on this?

dirk-thomas, 2020-07-30:
@MiguelCompany @richiware Friendly ping.

ivanpauno, 2020-08-27:
@MiguelCompany @richiware friendly ping

IkerLuengo, 2020-08-28:
@dirk-thomas @ivanpauno, somehow this went under our radar. I'll reproduce on Foxy and share my findings.

IkerLuengo, 2020-08-31:
I don't seem able to reproduce the issue with the latest ROS2 master commit [91d8a43](https://github.com/ros2/ros2/commit/91d8a43ed28edbe04f4ea361c493b19758e35877)

@realdream. in order to be able to further analyze the issue, can you please specify:
 a) If the error occurs consistently
 b) Which commit was used on the reproduction
 c) If the steps to reproduce the problem have changed from the ones described at the beginning of the issue, please, specify that also.

clalancette, 2020-09-17:
@IkerLuengo FYI, I've assigned the issue to you so it doesn't come up on our tracking board.  Feel free to unassign if you don't want to be assigned to it.