ros2/ros2 #541: Bouncy release binary `local_seup.bat` fails with "The system cannot find the path specified."

calvertdw, 2018-07-13:
## Bug report

- Operating System:
Windows 10 build 1803
- Installation type:
https://github.com/ros2/ros2/releases/download/release-bouncy/ros2-bouncy-windows-AMD64.zip

#### Steps to reproduce issue
Download and "Extract All..." to `F:\dev\ros2bouncybin`.
```
F:\dev\ros2bouncybin
λ local_setup.bat
The system cannot find the path specified.
```

#### Additional information

I have tried setting a few debug variables I saw from https://github.com/ros2/ros2/issues/400:
```
F:\dev\ros2bouncybin
λ echo %COLCON_TRACE_SETUP_FILES%
1

F:\dev\ros2bouncybin
λ echo %COLCON_TRACE%
1

F:\dev\ros2bouncybin
λ local_setup.bat
The system cannot find the path specified.
```

calvertdw, 2018-07-13:
Commenting `@echo off` I get more output, including some context:
```for /F %p in ('""c:\python36\python.exe" "F:\dev\ros2bouncybin\_local_setup_util.py" --merged-install"') do (
if "!_colcon_ordered_packages!" NEQ "" set "_colcon_ordered_packages=!_colcon_ordered_packages!;"
 set "_colcon_ordered_packages=!_colcon_ordered_packages!%p"
)
The system cannot find the path specified.

endlocal   & (set "_ordered_packages=" )
```

mikaelarguedas, 2018-07-13:
Can you confirm what version of python you have installed?
Similar issues were reported when users had python 3.7 installed while ROS 2 bouncy was compiled against Python 3.6.5

calvertdw, 2018-07-13:
```
F:\dev\ros2bouncybin
λ choco list --local-only | grep python
python 3.6.5
python3 3.6.5
```

calvertdw, 2018-07-13:
Maybe this should be a bug for colcon, that it does not communicate the source of this error effectively.

calvertdw, 2018-07-13:
After repairing my python installation, everything works. Thanks @mikaelarguedas!

closed 2018-07-13 by calvertdw

mikaelarguedas, 2018-07-13:
Great :+1: 

Can you provide the steps you followed to fix your downgrade ? (That can be useful to other users that face the same issue)

calvertdw, 2018-07-17:
1. `choco uninstall python python3`
1. Delete `F:\Python37`
1. Delete any other `PythonXX` dirs
1. `choco list --local-only | grep python`, make sure there is no python installed
1. `choco install python --version 3.6.5`
1. Install all pip stuff again from https://github.com/ros2/ros2/wiki/Installation