ros2/ros2 #542: macOS tarball for Bouncy is corrupt/incomplete

dljsjr, 2018-07-13:
Downloading the macOS Release Tarball found here: https://github.com/ros2/ros2/releases will yield an incomplete archive.

Attempting to extract from the command line hangs; attempting to extract using the built-in archive utility returns an "Operation Not Permitted" error code, but using The Unarchiver (https://itunes.apple.com/us/app/the-unarchiver/id425424353?mt=12) gives back a more useful "archive is incomplete" error message.

I'd be willing to believe what The Unarchiver says, given that the file size for the macOS tarball is an order of magnitude smaller than the Linux and Windows releases.

Karsten1987, 2018-07-13:
I can confirm this behavior. I also run into the problem (`Operation Not Permitted`).

When extracting the archive with `tar xfz ros2-bouncy-macos-x86_64.tar.bz2`, I have a semi-complete workspace:

```
-rw-r--r--@   1 karsten  staff      7 Jul  2 16:17 .colcon_install_layout
-rw-r--r--@   1 karsten  staff      0 Jul  2 16:17 COLCON_IGNORE
-rw-r--r--@   1 karsten  staff   4638 Jul  2 17:13 _local_setup_util.py
drwxr-xr-x@  22 karsten  staff    748 Jul  2 17:13 bin/
drwxr-xr-x@   6 karsten  staff    204 Jul  2 16:18 cmake/
drwxr-xr-x@   3 karsten  staff    102 Jul  2 16:23 examples/
drwxr-xr-x@  75 karsten  staff   2550 Jul  2 17:13 include/
drwxr-xr-x@ 308 karsten  staff  10472 Jul  2 17:13 lib/
-rw-r--r--@   1 karsten  staff   3463 Jul  2 17:13 local_setup.bash
-rw-r--r--@   1 karsten  staff   1897 Jul  2 17:13 local_setup.ps1
-rw-r--r--@   1 karsten  staff   3885 Jul  2 17:13 local_setup.sh
-rw-r--r--@   1 karsten  staff   3915 Jul  2 17:13 local_setup.zsh
drwxr-xr-x@   3 karsten  staff    102 Jul  2 16:51 opt/
```

When then trying to `source local_setup.bash`, I get the following:
```
 ➭ source local_setup.bash 
Traceback (most recent call last):
  File "/Users/karsten/workspace/ros2-osx/_local_setup_util.py", line 137, in <module>
    main()
  File "/Users/karsten/workspace/ros2-osx/_local_setup_util.py", line 18, in main
    packages = get_packages(Path(__file__).parent, args.merged_install)
  File "/Users/karsten/workspace/ros2-osx/_local_setup_util.py", line 42, in get_packages
    for p in (prefix_path / subdirectory).iterdir():
  File "/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/pathlib.py", line 1079, in iterdir
    for name in self._accessor.listdir(self):
  File "/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/pathlib.py", line 387, in wrapped
    return strfunc(str(pathobj), *args)
FileNotFoundError: [Errno 2] No such file or directory: '/Users/karsten/workspace/ros2-osx/share/colcon-core/packages'
```

mikaelarguedas, 2018-07-13:
Can someone try that? the artifact from the bouncy release is at https://ci.ros2.org/job/ci_packaging_osx/41/

wjwwood, 2018-07-13:
It extracts fine. Testing that it works will be hard for me just due to the state of my machine (python version etc...), can anyone else just do a spot check?

Karsten1987, 2018-07-13:
Works fine for me.
I am able to extract the archive with standard tools and run the talker demo node.

I am going to close this.

@dljsjr Please feel free to re-open this in case the new archive still troubles you.
Thanks for reporting!

closed 2018-07-13 by Karsten1987

mikaelarguedas, 2018-07-13:
@dljsjr I reuploaded the archive, it is available at https://github.com/ros2/ros2/releases/tag/release-bouncy

Note that it is expected that the macOS archive is much smaller than the other platforms

dljsjr, 2018-07-17:
@mikaelarguedas seem to be working, thanks! I was just basing my comment off the Ardent release where the macOS tarball was on the same order of magnitude as the other tarballs.