ros2/ros2 #543: Building `rviz_ogre_vendor` fails in Windows ROS 2 bouncy build with "patch.exe" pop-up

calvertdw, 2018-07-13:
## Bug report

When I build ros2, a "patch.exe" window pops up and stays there. The build will not finish until clicking the "X" to close it, when the package `rviz_ogre_vendor` immediately fails and the build completes with 1 failure.

- Operating System:
  - Windows 10 1803
- Installation type:
  - source
- Version or commit hash:
  - bouncy
- DDS implementation:
  - all 3 installed

#### Steps to reproduce issue
```
colcon build --merge-install
```
[image]

Click the red "X".

```bash
[Processing: rviz_ogre_vendor]
[Processing: rviz_ogre_vendor]
[Processing: rviz_ogre_vendor]
Failed   <<< rviz_ogre_vendor   [ Exited with code 1 ]

Summary: 182 packages finished [3230.51s]
  1 package failed: rviz_ogre_vendor
  6 packages not processed

F:\dev\ros2bouncy>
```

#### Additional information

This issue appears to be isolated to "patch.exe" running with Administrator privileges.

In an admin terminal:
[image]

In an normal terminal the `--version` command quickly pops up and closes a window instead of outputting the version:
[image]

wjwwood, 2018-07-13:
You need to run the build in an "run as administrator" cmd.exe to prevent that. There's a troubleshooting item for this:

https://github.com/ros2/ros2/wiki/Windows-Development-Setup#patchexe-opens-a-new-command-window-and-asks-for-administrator

calvertdw, 2018-07-13:
Oh! Well look at that. Thanks!

wjwwood, 2018-07-13:
Sure, no problem :) It's a pretty weird problem, but hopefully we'll move away from needing to patch ogre at some point.

I'll close for now, but can reopen if need be.

calvertdw, 2018-07-13:
Oddly, after deleting `build` and `install` (is there a clean command?) I rebuilt with the admin console and `rviz_ogre_vendor` still fails with:

```
  -- extracting... done
  No update step for 'ogre-master-ca665a6'
  Performing patch step for 'ogre-master-ca665a6'
  patching file CMake/Dependencies.cmake
  Assertion failed: hunk, file ../patch-2.5.9-src/patch.c, line 354
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(171,5): error MSB6006: "cmd.exe" exited with code 3. [F:\dev\ros2bouncy\build\rviz_ogre_vendor\ogre-master-ca665a6.vcxproj]
Done Building Project "F:\dev\ros2bouncy\build\rviz_ogre_vendor\ogre-master-ca665a6.vcxproj" (default targets) -- FAILED.
Done Building Project "F:\dev\ros2bouncy\build\rviz_ogre_vendor\ALL_BUILD.vcxproj" (default targets) -- FAILED.

Build FAILED.
```

mikaelarguedas, 2018-07-13:
> is there a clean command?

There is no clean command at the moment (https://github.com/colcon/colcon-core/issues/37) but
`rmdir build install /s /q` should be equivalent.

I'll reopen this for now. I'm not familiar with the issue, maybe @wjwwood faced this before?

calvertdw, 2018-07-13:
I found that cygwin hosts a 2.7.4 version of patch.
https://cygwin.com/cgi-bin2/package-cat.cgi?file=x86_64%2Fpatch%2Fpatch-2.7.4-1&grep=patch

(The latest appears to be 2.7.6: http://git.savannah.gnu.org/cgit/patch.git)

After installing the cygwin version patch works in non-admin prompt. Rebuilding now...

calvertdw, 2018-07-13:
It seems to have gotten past the patch file, but `rviz_ogre_vendor` still fails.
```
  No update step for 'ogre-master-ca665a6'
  Performing patch step for 'ogre-master-ca665a6'
  patching file CMake/Dependencies.cmake
  Performing configure step for 'ogre-master-ca665a6'
  -- ogre-master-ca665a6 configure command succeeded.  See also F:/dev/ros2bouncy/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6-stamp/ogre-master-ca665a6-configure-*.log
  Performing build step for 'ogre-master-ca665a6'
  CMake Error at F:/dev/ros2bouncy/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6-stamp/ogre-master-ca665a6-build-Release.cmake:16 (message):
    Command failed: 1
  
     'N:/CMake/bin/cmake.exe' '--build' '.' '--config' 'Release'
  
    See also
  
      F:/dev/ros2bouncy/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6-stamp/ogre-master-ca665a6-build-*.log
  
  
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(171,5): error MSB6006: "cmd.exe" exited with code 1. [F:\dev\ros2bouncy\build\rviz_ogre_vendor\ogre-master-ca665a6.vcxproj]
Done Building Project "F:\dev\ros2bouncy\build\rviz_ogre_vendor\ogre-master-ca665a6.vcxproj" (default targets) -- FAILED.
Done Building Project "F:\dev\ros2bouncy\build\rviz_ogre_vendor\ALL_BUILD.vcxproj" (default targets) -- FAILED.

Build FAILED.
```
```
Summary: 107 packages finished [1112.42s]
  1 package failed: rviz_ogre_vendor
  7 packages aborted: lifecycle_msgs pendulum_msgs rcl_interfaces rosgraph_msgs std_msgs std_srvs test_msgs
  74 packages not processed
```
I guess I will now try with admin console.

calvertdw, 2018-07-13:
Yep! I just noticed those log files. Unfortunately those are in the `build` directory, so I've got to wait on this thing to finish again.

calvertdw, 2018-07-13:
As for cygwin I installed the bare minimum and deleted everything the bin folder so that it only contains `patch.exe` and some dlls it needs, so it should be "self-contained" for me.

calvertdw, 2018-07-13:
@wjwwood here is the log.

```
    1023 Warning(s)
    59 Error(s)

Time Elapsed 00:10:05.80
```

[ogre-master-ca665a6-build-out.log](https://github.com/ros2/ros2/files/2194462/ogre-master-ca665a6-build-out.log)

calvertdw, 2018-07-13:
I realized from https://stackoverflow.com/a/41254042/1070333 that I might need to add Python to my VS 2017 installation. I'm going to try that and report back.

calvertdw, 2018-07-16:
So I tried with this, but same errors.
[image]

calvertdw, 2018-07-16:
I can also verify that the patch worked in `F:\dev\ros2bouncy\build\rviz_ogre_vendor\ogre-master-ca665a6-prefix\src\ogre-master-ca665a6\CMake\Dependencies.cmake`
[image]

calvertdw, 2018-07-16:
For reference also ,from `configure-out`. It does see that python is installed. I have tried looking for an example log on your CI with no luck so I'm not sure what features might be required.
```
-- The following external packages were located on your system.
-- This installation will have the extra features provided by these packages.
+ freetype
+ OpenGL
+ OpenGL 3+
+ OpenGL ES 1.x
+ OpenGL ES 2.x
+ OpenGL ES 3.x
+ DirectX9
+ DirectX11
+ POCO
+ Python
-----------------------------------------------------------------------------
-- The following OPTIONAL packages could NOT be located on your system.
-- Consider installing them to enable more features from this software.
+ zlib: Simple data compression library <http://www.zlib.net>
+ freeimage: Support for commonly used graphics image formats <http://freeimage.sourceforge.net>
+ cg: C for graphics shader language <http://developer.nvidia.com/object/cg_toolkit.html>
+ boost: Boost (general) <http://boost.org>
+ boost-thread: Used for threading support <http://boost.org>
+ tbb: Threading Building Blocks <http://www.threadingbuildingblocks.org/>
+ GLSL Optimizer: GLSL Optimizer <http://github.com/aras-p/glsl-optimizer/>
+ HLSL2GLSL: HLSL2GLSL <http://hlsl2glslfork.googlecode.com/>
+ OpenEXR: Load High dynamic range images <http://www.openexr.com/>
+ SDL2: Simple DirectMedia Library needed for input handling in samples <https://www.libsdl.org/>
+ Doxygen: Tool for building API documentation <http://doxygen.org>
+ Softimage: Softimage SDK needed for building XSIExporter <FALSE>
```

calvertdw, 2018-07-16:
Going though "build-out" I found the first error which is interesting and maybe the cause of the problem:
```
ogre-master-ca665a6-build\Components\Python\OgrePYTHON_wrap.cxx : 
fatal error C1128: number of sections exceeded object file format limit: compile with /bigobj
```

wjwwood, 2018-07-16:
Interesting, I haven't seen that and our CI doesn't have that issue. It might a new problem that's only showing up when setting up Windows from scratch.

@Martin-Idel-SI have any of your guys seen this issue before?

I'm not sure what adding Python via Visual Studio might do, since we're also installing it from `choco`, but it sounds like you have the same issue regardless of where Python was coming from.

Perhaps you could manually patch the CMake code of Ogre (or pass a CMake variable to it) to avoid building the Python wrappers which we don't use (I wasn't even aware that this was an option for Ogre).

calvertdw, 2018-07-16:
> Perhaps you could manually patch the CMake code of Ogre (or pass a CMake variable to it) to avoid building the Python wrappers which we don't use (I wasn't even aware that this was an option for Ogre).

@wjwwood I'm trying to figure out how to do that.

wjwwood, 2018-07-16:
@calvertdw you could add CMake arguments here:

https://github.com/ros2/rviz/blob/09905a3db7df9e9ae8ac888b6d9a0c1923200203/rviz_ogre_vendor/CMakeLists.txt#L131-L142

But you'd have to figure out if there is one, otherwise you'll have to patch it by adding to our patch file (or creating a new one).

calvertdw, 2018-07-16:
After adding a comment on Python to the patch file, I am down to one error.

[ogre-ca665a6-patch-2.diff.txt](https://github.com/ros2/ros2/files/2199312/ogre-ca665a6-patch-2.diff.txt)

The remaining error is:
`Unable to open file F:\dev\ros2bouncy\build\rviz_ogre_vendor\ogre-master-ca665a6-prefix\src\ogre-master-ca665a6-build\java\src\org\Ogre\SWIGTYPE_p_Ogre__multimapT_Ogre__TextureUnitState__TextureEffectType_Ogre__TextureUnitState__TextureEffect_std__lessT_Ogre__TextureUnitState__TextureEffectType_t_t__type.java: No such file or directory`

wjwwood, 2018-07-16:
I haven't seen that one either. I don't think I have Java installed on my Windows machine. I'm surprised that it's doing SWIG related stuff at all. I'd have to dig the build file out of one of our nightlies or CI to compare. I'll try to do that later, but for now another patch to remove the Java bindings generation might be best.

calvertdw, 2018-07-16:
Okay with the following patch that excludes Java, `rviz_ogre_vendor` succeeds but now I have a failure in `rviz_rendering`.

[ogre-ca665a6-patch-3.diff.txt](https://github.com/ros2/ros2/files/2199511/ogre-ca665a6-patch-3.diff.txt)

```
CMake Error at F:/dev/ros2bouncy/install/share/rviz_ogre_vendor/cmake/rviz_ogre_vendor-extras.cmake:127 (set_target_properties):
  Property INTERFACE_LINK_LIBRARIES may not contain link-type keyword
  "optimized".  The INTERFACE_LINK_LIBRARIES property may contain
  configuration-sensitive generator-expressions which may be used to specify
  per-configuration rules.
Call Stack (most recent call first):
  F:/dev/ros2bouncy/install/share/rviz_ogre_vendor/cmake/rviz_ogre_vendorConfig.cmake:30 (include)
  CMakeLists.txt:24 (find_package)
```

wjwwood, 2018-07-16:
Can you make this diff:

```diff
--- /Users/william/Downloads/rviz_ogre_vendor-extras.cmake.txt
+++ /Users/william/Downloads/rviz_ogre_vendor-extras.cmake.txt (Unsaved)
@@ -124,6 +124,7 @@
     if(APPLE)
       list(APPEND _extra_interface_link_libraries "-framework Cocoa")
     endif()
+    message(FATAL_ERROR "_extra_interface_link_libraries: ${_extra_interface_link_libraries}")
     set_target_properties(rviz_ogre_vendor::OgreMain
       PROPERTIES
         "INTERFACE_LINK_LIBRARIES" "${_extra_interface_link_libraries}"

```

And post the output from there? It might be that one of the inputs to the `INTERFACE_LINK_LIBRARIES` differs on your system, perhaps due to additional things being installed? Hopefully I can tell from the output.

calvertdw, 2018-07-16:
I do have a lot installed. I tried using a Windows Docker container to build in a clean environment, but that was looking like a long road after a while. (https://github.com/ihmcrobotics/ihmc-open-robotics-software/blob/feature/bouncy-windows-and-linux-support/ihmc-interfaces/src/rosdocker/windows/Dockerfile)

wjwwood, 2018-07-16:
That would be cool if we could get it to work. I haven't even looked into it yet though.

calvertdw, 2018-07-16:
Here is your output:
```
--- stderr: rviz_rendering
CMake Error at F:/dev/ros2bouncy/install/share/rviz_ogre_vendor/cmake/rviz_ogre_vendor-extras.cmake:127 (message):
  _extra_interface_link_libraries:
  optimized;F:/dev/ros2bouncy/install/opt/rviz_ogre_vendor/lib/freetype.lib;debug;F:/dev/ros2bouncy/install/opt/rviz_ogre_vendor/lib/freetyped.lib;F:/dev/ros2bouncy/install/share/rviz_ogre_vendor/cmake/../../../opt/rviz_ogre_vendor/lib/zlibstatic.lib;opengl32;glu32
Call Stack (most recent call first):
  F:/dev/ros2bouncy/install/share/rviz_ogre_vendor/cmake/rviz_ogre_vendorConfig.cmake:30 (include)
  CMakeLists.txt:24 (find_package)
```

wjwwood, 2018-07-16:
So for what ever reason the instance of `freetype` built by Ogre is both release and debug, which is causing issues. We'll have to harden that code to handle that case. I doesn't usually come up because we don't switch from release to debug or vice versa on Windows without deleting the build folder.

As a workaround you can try just deleting one of those two files (`freetype.lib` or `freetyped.lib`).

I'll make an issue on the rviz repository that this case is possible and therefore should be handled.

calvertdw, 2018-07-16:
Oh... That is probably my fault because I built with Debug in Visual Studio GUI as I was debugging earlier linking issues. 😅

wjwwood, 2018-07-16:
That's ok, it should be handled, if that work around doesn't work, try deleting the whole build folder for `rviz_ogre_vendor`. Here's an issue for that: https://github.com/ros2/rviz/issues/339

calvertdw, 2018-07-16:
Neither of those workarounds worked. They yeilded:
```
--- stderr: rviz_rendering
CMake Error at F:/dev/ros2bouncy/install/share/rviz_ogre_vendor/cmake/rviz_ogre_vendor-extras.cmake:127 (message):
  _extra_interface_link_libraries:
  optimized;F:/dev/ros2bouncy/install/opt/rviz_ogre_vendor/lib/freetype.lib;debug;F:/dev/ros2bouncy/install/opt/rviz_ogre_vendor/lib/freetyped.lib;F:/dev/ros2bouncy/install/share/rviz_ogre_vendor/cmake/../../../opt/rviz_ogre_vendor/lib/zlibstatic.lib;opengl32;glu32
Call Stack (most recent call first):
  F:/dev/ros2bouncy/install/share/rviz_ogre_vendor/cmake/rviz_ogre_vendorConfig.cmake:30 (include)
  CMakeLists.txt:24 (find_package)
```
I'm going to delete fully `install` and `build` , build, and get back to you tomorrow.

calvertdw, 2018-07-17:
Well, unfortunately, still getting:
```
--- stderr: rviz_rendering
CMake Error at F:/dev/ros2bouncy/install/share/rviz_ogre_vendor/cmake/rviz_ogre_vendor-extras.cmake:127 (message):
  _extra_interface_link_libraries:
  F:/dev/ros2bouncy/install/opt/rviz_ogre_vendor/lib/freetype.lib;F:/dev/ros2bouncy/install/share/rviz_ogre_vendor/cmake/../../../opt/rviz_ogre_vendor/lib/zlibstatic.lib;opengl32;glu32
Call Stack (most recent call first):
  F:/dev/ros2bouncy/install/share/rviz_ogre_vendor/cmake/rviz_ogre_vendorConfig.cmake:30 (include)
  CMakeLists.txt:24 (find_package)

---
Failed   <<< rviz_rendering     [ Exited with code 1 ]
Aborted  <<< rmw_implementation

Summary: 100 packages finished [841.26s]
  1 package failed: rviz_rendering
  1 package aborted: rmw_implementation
  1 package had stderr output: rviz_rendering
  87 packages not processed
```

calvertdw, 2018-07-17:
OH! Is that just the message we added that's failing the build? haha

calvertdw, 2018-07-17:
Yes, in that case deleting `freetyped.lib` did the trick.

calvertdw, 2018-07-17:
So, in summary the solutions to this issue are:

1. Using this stripped down cygwin version of patch: [patch-2.7.4-cygwin.zip](https://github.com/ros2/ros2/files/2202420/patch-2.7.4-cygwin.zip)
2. Using this updated patch file in `rviz_ogre_vendor`: [ogre-ca665a6-patch-fix.diff.txt](https://github.com/ros2/ros2/files/2202421/ogre-ca665a6-patch-fix.diff.txt)

calvertdw, 2018-07-17:
My build was successful.