ros2/ros2 #545: Why intra-process communication isn't more efficient than standalone nodes

realdream, 2018-07-17:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04
- Installation type:
  - from source
- Version or commit hash:
  - release bouncy
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

I'm following the tutorial for Intra-process communication
```
ros2 run intra_process_demo image_pipeline_all_in_one
```
use netload for monitoring network usage
```
sudo netload lo
```
what confusing me is:
1 intra-process communication still cost heavy load on local loopback 
2 I tried to run three standalone node(camera_node,watermark_node and image_view_node) and I can't see any difference on cpu usage compared to intra-process way. 

It cost almost 100% cpu on my target platform to transmit  image topic (1080p@30hz).
Is there any way to make improvement? it seems like the topic still publish through dds even in intra-process way?

scott-eddy, 2018-07-18:
I can comment a little as to what is happening here having looked at the code, note however that I'm not an author nor maintainer of any of the ROS2 project. 

> 1 intra-process communication still cost heavy load on local loopback
Intra-process communications still provide a message to the underlying middleware, in the case of intra-process configured nodes co-existing with inter-process nodes.  The logic for this can be found in the various `publish` methods of rclcpp/publisher.hpp.

Ultimately all types of calls to `publish` end up in [`publish(std::unique_ptr<MessageT, MessageDeleter> & msg)`](https://github.com/ros2/rclcpp/blob/master/rclcpp/include/rclcpp/publisher.hpp#L189), the logic here immediately performs `do_inter_process_publish()`, only then does the logic for storing the message intra_process kick in.  So even if configured for intra_process you are going to be giving the message to the underlying middleware.  This is also explicitly  mentioned in the [Intra Process Communication Demo](https://github.com/ros2/ros2/wiki/Intra-Process-Communication):

> Avoiding Unnecessary Interprocess Publishes
Currently we're relying on the middleware, the DDS vendor, to avoid publishing to the wire unnecessarily, but no matter what we have to give the middleware a copy of the user's message. We could avoid this copy given to the system if we could know if it is needed. We can do this currently by checking to see if there are any non intra process subscriptions currently attached to the publisher. The problem with this becomes apparent when we go to implement latching, which we'll see more about below.

> 2 I tried to run three standalone node(camera_node,watermark_node and image_view_node) and I can't see any difference on cpu usage compared to intra-process way.

This likely is due to the intrinsic cost of allocating and copying 1080p images at 30Hz.  This is going to be expensive and is required in order to give intra_process subscribers a message that they have ownership of (unique_ptr).  There are some plans to use shared copies of messages but right now that is not implemented.  Again from the intra_process example:

> In other parts of the system we've worked really hard to allow users to avoid memory allocation. This has performance benefits and may be required for real-time or embedded scenarios. We cannot currently do that with intra process. Mostly this is because we haven't had time to figure out the right interfaces, but the general problem is that if a message needs to be delivered to more than one subscription, or if the user gives us a const & or const shared_ptr when publishing, we need to make a copy. And the destination of the copy is currently created using new and is not configurable. We expect to resolve that in the future.

sloretz, 2018-07-19:
FYI there is discussion about `intra-process` in ros2/rclcpp#504

> 2 I tried to run three standalone node(camera_node,watermark_node and image_view_node) and I can't see any difference on cpu usage compared to intra-process way.

Using intra-process should be more efficient. Would you mind posting a copy/pasteable example of with the commands run and how performance is being measured?

realdream, 2018-07-20:
Thank you for your reply! 

I have written a simple test package https://github.com/realdream/ros2_basic_test
It's a simple image transmitter without any image_show or cout message

### standalone node
```
ros2 launch basic_test test_pub_sub.launch.py
```
then I got cpu usage from top
```
PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND                               
11001 yrk       20   0  661884 220608  11772 S  40.1  1.4   0:12.14 standalone_sub                        
11000 yrk       20   0  881704  94504  15124 S  31.1  0.6   0:12.40 standalone_pub         
```
### intra-process
```
ros2 run basic_test composition
```
then I got cpu usage from top
```
PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND     
11105 yrk       20   0 1772084 651240  14836 S  73.3  4.0   0:9.65 composition    
```
### result
The intra-process way cost more cpu & memory.

It seems that
1. Subscriber did get the zero-copy message in intra-process way, we can assure that by checking msg address.
2. However another copy of  the message still publish & subscribe through the network by the dds even if there is no inter-process node participate in the topic. that cost heavy load on local loopback and unnecessary cpu & ram usage.

gilduz, 2020-03-05:
Is this issue not solved? I'm facing the same problematics with ros dashing diamata.

hidmic, 2020-04-23:
AFAIK intraprocess communication has changed significantly since. Have you tried Eloquent Elusor or `master` (upcoming Foxy Fitzroy)?

realdream, 2020-06-11:
Intra-process communication works smoothly in foxy now. great job!

closed 2020-06-11 by realdream