ros2/ros2 #546: Cannot launch RViz 2 on macOS High Sierra (missing Ogre .dylib)

dljsjr, 2018-07-17:
## Bug report

**Required Info:**

- Operating System:
  - macOS "High Sierra"10.13.5 (17F77)
- Installation type:
  - Binaries
- Version or commit hash:
  - N/A
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

1. Install macOS binaries following the instructions here: https://github.com/ros2/ros2/wiki/OSX-Install-Binary
2. Source the correct shell setup for your environment
3. `ros2 run rviz2 rviz2`

#### Expected behavior

RViz 2 should launch

#### Actual behavior

RViz 2 fails to launch with the following error message:

```shell
dyld: Library not loaded: @rpath/libOgreMain.1.10.11.dylib
  Referenced from: /Users/dstephen/ros2_install/ros2-osx/lib/rviz2/rviz2
  Reason: image not found
```

#### Additional information

SIP is definitely disabled; output of `csrutil status` shows `disabled`.

wjwwood, 2018-07-18:
Are you using `bash` or `zsh`? Can you try just `rviz2`?

dljsjr, 2018-07-18:
@wjwwood I'm using zsh. Running just `rviz2` has the same result.

wjwwood, 2018-07-18:
Can you try `bash`? (I'm also running `zsh`, let me try that too)

wjwwood, 2018-07-18:
I see the same thing... dang I tested the binary only with `bash` during the release I guess:

```
% rviz2
dyld: Library not loaded: @rpath/libOgreMain.1.10.11.dylib
  Referenced from: /tmp/ros2-osx/bin/rviz2
  Reason: image not found
[1]    66283 abort      rviz2
```

@dirk-thomas are you aware of any changes in `colcon` that might have addressed this already since bouncy?

mikaelarguedas, 2018-07-19:
> are you aware of any changes in colcon that might have addressed this already since bouncy?

Do you mean in colcon itself or on the zsh side?

on the `zsh` side I don't think anything changed since the release (except an unrelated PR earlier today)

dhood, 2018-07-19:
that looks like the issue you get when you have a different library version than what the binaries were built against. @dljsjr what's your ogre version?

mikaelarguedas, 2018-07-19:
> that looks like the issue you get when you have a different library version than what the binaries were built against. @dljsjr what's your ogre version?

The surprising thing is that we ship our own version of Ogre...

dljsjr, 2018-07-19:
I can confirm that it does launch with `bash`

dljsjr, 2018-07-19:
@dhood I don't have an additional user or system Ogre installation, it's using whatever is vendored.

dhood, 2018-07-19:
alright, didn't realise it's specific to zsh. Never mind me!

wjwwood, 2018-07-19:
I opened an issue about it, but I don't have time to dig into it right now: https://github.com/colcon/colcon-zsh/issues/12

dljsjr, 2018-07-19:
I see that this is fixed, is there a nightly I can download that will allow me to use .zsh? Or do I need to build from source.

mikaelarguedas, 2018-07-19:
The archive from https://ci.ros2.org/job/ci_packaging_osx/48/ should have the fix in it.

Otherwise we're about to do a bouncy patch release. The artifacts of the patch release should have this fixed as well

dljsjr, 2018-07-20:
Sweet thanks!

mikaelarguedas, 2018-07-20:
> Sweet thanks!

:+1: thanks for reporting the issue

> Bad news is: ROS2 CLI doesn't seem to work

Yeah one of the main changes of this patch release is that we upgraded to Python3.7 to fix https://github.com/ros2/ros2/issues/526

So I think you now need to upgrade `python` for the Python based tools to work

dljsjr, 2018-07-20:
Easy enough, I have 3.7 installed I just pinned brew to 3.6 for ROS 2.

mikaelarguedas, 2018-07-20:
do you have argcomplete installed from pip ?
Does upgrading it fixes the issue ?

(side note: we do not receive notifications on comment edits so it's likely for us to miss them, especially on closed issues)

dljsjr, 2018-07-20:
@mikaelarguedas reinstalling argcomplete fixed it.