ros2/ros2 #547: `ros1_bridge` not included in Bouncy release on Windows

calvertdw, 2018-07-17:
## Bug report

From https://github.com/ros2/ros1_bridge/blob/master/README.md:

> The bridge provided with the prebuilt ROS 2 binaries includes support for common ROS interfaces (messages/services), such as the interface packages listed in the ros2/common_interfaces repository and tf2_msgs.

- Operating System:
  - Windows 10 build 1803
- Installation type:
  - binary
- Version or commit hash:
  - bouncy
- DDS implementation:
  - all 3 installed

#### Steps to reproduce issue
```
F:\dev\ros2bouncybin
λ local_setup.bat

F:\dev\ros2bouncybin
λ call "N:\opensplice67\HDE\x86_64.win64\release.bat"
"<<< Vortex OpenSplice HDE Release 6.7.180404OSS For x86_64.win64, Date 2018-06-05 >>>"

F:\dev\ros2bouncybin
λ call "C:\Program Files\rti_connext_dds-5.3.1\resource\scripts\rtisetenv_x64Win64VS2017.bat"
###############################################################################

         (c) Copyright, Real-Time Innovations, All rights reserved.

                           RTI Connext DDS 5.3.1

###############################################################################

F:\dev\ros2bouncybin
λ call "C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Auxiliary\Build\vcvars64.bat"
**********************************************************************
** Visual Studio 2017 Developer Command Prompt v15.7.5
** Copyright (c) 2017 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x64'

F:\dev\ros2bouncybin
λ ros2 run ros1_bridge dynamic_bridge -- --help
Package 'ros1_bridge' not found
```

#### Additional information

Rviz 2 launches fine, so it seems my install is good.

gavanderhoorn, 2018-07-17:
Afaik the bridge is not included as it needs a proper ROS1 installation to be present, and that is not possible on Windows.

calvertdw, 2018-07-17:
Haha of course 😅

closed 2018-07-17 by calvertdw