ros2/ros2 #552: message transmits failed with local loop-back only

realdream, 2018-07-27:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04
- Installation type:
  - from source 
- Version or commit hash:
  - release bouncy
- DDS implementation:
  -  Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

disconnect all the network
```
$ifconfig
lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:840500 errors:0 dropped:0 overruns:0 frame:0
          TX packets:840500 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:106407657 (106.4 MB)  TX bytes:106407657 (106.4 MB)

wlp2s0    Link encap:Ethernet  HWaddr 9c:b6:d0:89:89:5d  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:16720 errors:0 dropped:0 overruns:0 frame:0
          TX packets:4864 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:5719273 (5.7 MB)  TX bytes:1098261 (1.0 MB)
```
then run the minimal talker&listener
```
ros2 run demo_nodes_cpp talker
ros2 run demo_nodes_cpp listener
```
listener can not receive any message
ros2 topic/node is also not working

sloretz, 2018-08-01:
If I understand correctly, the loopback interface needs to be configured to support multicast. I don't have a complete answer, but this [stackoverflow post](https://stackoverflow.com/questions/16007311/multicasting-on-loopback-device) is a start.

With just the loopback interface talker/listener don't communicate, and commands like `ros2 topic list` or `ros2 node list` show nothing. If I enable multicast over loopback then `ros2 topic list` and `ros2 node list` work, but no messages are received by `ros2 topic echo /chatter`.

```
route add -net 224.0.0.0 netmask 240.0.0.0 dev lo
ifconfig lo multicast
```

Maybe another route needs to be added for topics to communicate? It seems like the solution to this will be documenting how to configure the system rather than changing to ros2 code.

clalancette, 2018-10-04:
All right, I'm pretty sure this is the same problem as what is in https://github.com/ros2/rmw_fastrtps/issues/228.  I'm going to close this out since we've done a bit more work over there.

closed 2018-10-04 by clalancette