ros2/ros2 #553: Official "fat" zip for easy ROS2 installation on Windows

anhosi, 2018-07-27:
## Feature request

#### Feature description
In addition to the Windows binary release in its current form which requires a couple of dependencies to be installed in the system a "fat" zip including all necessary dependencies should be provided.
Such a zip would simplify the installation to unzipping and maybe running a post-unzip script.

Especially for corporate Windows environments where a user may not have admin permissions and centrally enforced updates might interfere with the system requirements of the current installation instructions, this would make using ROS2 much easier.

#### Implementation considerations
We have built such a fat zip for the bouncy.0 release. Unfortunately we cannot share the result but only our steps taken:
- Install [WinPython](http://winpython.github.io/) Zero instead of choco python (reason: winpython can be made relocatable)
- Use WinPythons pip to install all python dependencies from the ros2 installation instructions
- Run `make_winpython_movable`
- Install the remaining choco packages and OpenSSL as listed in the installation instructions (the installation folders can be moved).
- Download and unzip ROS2 binary release zip.
- Create a `Dependencies` folder inside the unzipped ros2 archive.
- Move the WinPython installation, all installed choco packages and OpenSSL into the Dependencies folder
- Patch `local_setup.bat`
  - set python interpreter:
    ```set "_colcon_python_executable=%~dp0\Dependencies\WinPython\python-3.6.5.amd64\python.exe"```
  - Add dependencies to path:
    ```call:_colcon_prefix_bat_prepend_unique_value PATH "%~dp0Dependencies\tinyxml2\lib;%~dp0Dependencies\OpenSSL-Win64\bin;%~dp0Dependencies\WinPython\python-3.6.5.amd64;%~dp0Dependencies\WinPython\python-3.6.5.amd64\Scripts"```
- Add a post-unzip script that patches the absolute path of the python interpreter at ros2 built time with correct path to WinPython. (This script needs to be run once after unzipping.)
- Paperworks: add NOTICE file listing all included software and file containing all 3rd party licenses in order to comply with license requirements.
- Zip the expanded ros2 folder containing all these additions

jespersmith, 2018-08-17:
I would be interested in this feature as well. Installing ROS2 on Windows is currently quite involved, making it hard to deploy to non-developers.

anhosi, 2018-11-19:
@Bimble sorry I missed your question.

If it helps I'm happy to test a fat zip.