ros2/ros2 #555: Wiki update suggestion

billchenxi, 2018-08-02:
## Feature request

#### Feature description

First of all, thank you! 

Since ROS2 is still kind in the phase of development, so a lot of users might want to install ROS2 and ROS1 parallel on the computer. 

#### Implementation considerations

sloretz, 2018-08-02:
Hi @billchenxi , I'm not sure if I understand. Is the suggestion to add a wiki page showing how to install ROS1 and ROS2 side by side?

mikaelarguedas, 2018-08-02:
Thanks @billchenxi for reaching out.

ROS 1 and ROs 2 are side by side installable.

Do you have a specific change in mind to make on the wiki ?

mikaelarguedas, 2018-08-16:
Closing due to no response.

@billchenxi feel free to comment here with more information and we can reopen

closed 2018-08-16 by mikaelarguedas