ros2/ros2 #560: Building package which contains interface files (.msg) packages failed

minggangw, 2018-08-09:
## Bug report

**Required Info:**

- Operating System:
Windows 10

- Installation type:
Install from binary packages https://ci.ros2.org/view/packaging/job/packaging_windows/lastSuccessfulBuild/artifact/ws/ros2-package-windows-AMD64.zip

- Version or commit hash:
[#1087](https://ci.ros2.org/view/packaging/job/packaging_windows/1087/)
- DDS implementation:
Default

- Client library (if applicable):

#### Steps to reproduce issue
* Build some package which contains interface files (.msg)

#### Expected behavior
* Build successfully

#### Actual behavior
* Errors were found in output, it seems that `rti_connext_dds-5.3.1' not found

```bash
CMake Warning at C:/ros2-windows/share/rmw_connext_shared_cpp/cmake/ament_cmake_export_libraries-extras.cmake:62 (message):
  Package 'rmw_connext_shared_cpp' exports the library 'C:/Program
  Files/rti_connext_dds-5.3.1/lib/x64Win64VS2017/rticonnextmsgcppd.lib' which
  doesn't exist
Call Stack (most recent call first):
  C:/ros2-windows/share/rmw_connext_shared_cpp/cmake/rmw_connext_shared_cppConfig.cmake:38 (include)
  C:/ros2-windows/share/rmw_connext_cpp/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  C:/ros2-windows/share/rmw_connext_cpp/cmake/rmw_connext_cppConfig.cmake:38 (include)
  C:/ros2-windows/share/rmw_implementation_cmake/cmake/call_for_each_rmw_implementation.cmake:61 (find_package)
  C:/ros2-windows/share/rosidl_generator_py/cmake/rosidl_generator_py_get_typesupports.cmake:26 (call_for_each_rmw_implementation)
  C:/ros2-windows/share/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake:27 (rosidl_generator_py_get_typesupports)
  C:/ros2-windows/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:38 (include)
  C:/ros2-windows/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:169 (ament_execute_extensions)
  CMakeLists.txt:22 (rosidl_generate_interfaces)
```
#### Additional information

* This issue does not happen on Ubuntu/macOS
* I did not execute `set "NDDSHOME=C:\Program Files\rti_connext_dds-5.3.1"`

minggangw, 2018-08-09:
Here is the package I used for testing https://github.com/RobotWebTools/rclnodejs/tree/develop/test/rclnodejs_test_msgs

and the error reported on Appveyor https://ci.appveyor.com/project/minggangw/rclnodejs/build/develop-806#L306

mikaelarguedas, 2018-08-09:
These CMake messages are only warnings and are expected as you don't have Connext sourced, this should not cause a build failure.

Could you provide a full console output or the message with the error that makes the build fail ?
Thanks!

minggangw, 2018-08-10:
Indeed, there are only warnings, but failed in the end. The whole output is below (you can check this [online](https://ci.appveyor.com/project/minggangw/rclnodejs/build/develop-806#L302)):

```bash
colcon build
Starting >>> rclnodejs_test_msgs
[Processing: rclnodejs_test_msgs]
--- stderr: rclnodejs_test_msgs
CMake Warning at C:/ros2-windows/share/rmw_connext_shared_cpp/cmake/ament_cmake_export_include_directories-extras.cmake:11 (message):
  Package 'rmw_connext_shared_cpp' exports the include directory 'C:/Program
  Files/rti_connext_dds-5.3.1/include' which doesn't exist
Call Stack (most recent call first):
  C:/ros2-windows/share/rmw_connext_shared_cpp/cmake/rmw_connext_shared_cppConfig.cmake:38 (include)
  C:/ros2-windows/share/rmw_connext_cpp/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  C:/ros2-windows/share/rmw_connext_cpp/cmake/rmw_connext_cppConfig.cmake:38 (include)
  C:/ros2-windows/share/rmw_implementation_cmake/cmake/call_for_each_rmw_implementation.cmake:61 (find_package)
  C:/ros2-windows/share/rosidl_generator_py/cmake/rosidl_generator_py_get_typesupports.cmake:26 (call_for_each_rmw_implementation)
  C:/ros2-windows/share/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake:27 (rosidl_generator_py_get_typesupports)
  C:/ros2-windows/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:38 (include)
  C:/ros2-windows/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:169 (ament_execute_extensions)
  CMakeLists.txt:22 (rosidl_generate_interfaces)
CMake Warning at C:/ros2-windows/share/rmw_connext_shared_cpp/cmake/ament_cmake_export_include_directories-extras.cmake:11 (message):
  Package 'rmw_connext_shared_cpp' exports the include directory 'C:/Program
  Files/rti_connext_dds-5.3.1/include/ndds' which doesn't exist
Call Stack (most recent call first):
  C:/ros2-windows/share/rmw_connext_shared_cpp/cmake/rmw_connext_shared_cppConfig.cmake:38 (include)
  C:/ros2-windows/share/rmw_connext_cpp/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  C:/ros2-windows/share/rmw_connext_cpp/cmake/rmw_connext_cppConfig.cmake:38 (include)
  C:/ros2-windows/share/rmw_implementation_cmake/cmake/call_for_each_rmw_implementation.cmake:61 (find_package)
  C:/ros2-windows/share/rosidl_generator_py/cmake/rosidl_generator_py_get_typesupports.cmake:26 (call_for_each_rmw_implementation)
  C:/ros2-windows/share/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake:27 (rosidl_generator_py_get_typesupports)
  C:/ros2-windows/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:38 (include)
  C:/ros2-windows/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:169 (ament_execute_extensions)
  CMakeLists.txt:22 (rosidl_generate_interfaces)
CMake Warning at C:/ros2-windows/share/rmw_connext_shared_cpp/cmake/ament_cmake_export_libraries-extras.cmake:62 (message):
  Package 'rmw_connext_shared_cpp' exports the library 'C:/Program
  Files/rti_connext_dds-5.3.1/lib/x64Win64VS2017/nddsc.lib' which doesn't
  exist
Call Stack (most recent call first):
  C:/ros2-windows/share/rmw_connext_shared_cpp/cmake/rmw_connext_shared_cppConfig.cmake:38 (include)
  C:/ros2-windows/share/rmw_connext_cpp/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  C:/ros2-windows/share/rmw_connext_cpp/cmake/rmw_connext_cppConfig.cmake:38 (include)
  C:/ros2-windows/share/rmw_implementation_cmake/cmake/call_for_each_rmw_implementation.cmake:61 (find_package)
  C:/ros2-windows/share/rosidl_generator_py/cmake/rosidl_generator_py_get_typesupports.cmake:26 (call_for_each_rmw_implementation)
  C:/ros2-windows/share/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake:27 (rosidl_generator_py_get_typesupports)
  C:/ros2-windows/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:38 (include)
  C:/ros2-windows/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:169 (ament_execute_extensions)
  CMakeLists.txt:22 (rosidl_generate_interfaces)
CMake Warning at C:/ros2-windows/share/rmw_connext_shared_cpp/cmake/ament_cmake_export_libraries-extras.cmake:62 (message):
  Package 'rmw_connext_shared_cpp' exports the library 'C:/Program
  Files/rti_connext_dds-5.3.1/lib/x64Win64VS2017/nddscore.lib' which doesn't
  exist
Call Stack (most recent call first):
  C:/ros2-windows/share/rmw_connext_shared_cpp/cmake/rmw_connext_shared_cppConfig.cmake:38 (include)
  C:/ros2-windows/share/rmw_connext_cpp/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  C:/ros2-windows/share/rmw_connext_cpp/cmake/rmw_connext_cppConfig.cmake:38 (include)
  C:/ros2-windows/share/rmw_implementation_cmake/cmake/call_for_each_rmw_implementation.cmake:61 (find_package)
  C:/ros2-windows/share/rosidl_generator_py/cmake/rosidl_generator_py_get_typesupports.cmake:26 (call_for_each_rmw_implementation)
  C:/ros2-windows/share/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake:27 (rosidl_generator_py_get_typesupports)
  C:/ros2-windows/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:38 (include)
  C:/ros2-windows/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:169 (ament_execute_extensions)
  CMakeLists.txt:22 (rosidl_generate_interfaces)
CMake Warning at C:/ros2-windows/share/rmw_connext_shared_cpp/cmake/ament_cmake_export_libraries-extras.cmake:62 (message):
  Package 'rmw_connext_shared_cpp' exports the library 'C:/Program
[... cut ...]
  C:/ros2-windows/share/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake:27 (rosidl_generator_py_get_typesupports)
  C:/ros2-windows/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:38 (include)
  C:/ros2-windows/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:169 (ament_execute_extensions)
  CMakeLists.txt:22 (rosidl_generate_interfaces)
---
Failed   <<< rclnodejs_test_msgs	[ Exited with code 1 ]
Summary: 0 packages finished [58.4s]
  1 package failed: rclnodejs_test_msgs
  1 package had stderr output: rclnodejs_test_msgs
```

mikaelarguedas, 2018-08-10:
Unfortunately your appveyor jobs print only the content of the stderr stream to console. But MSVC doesnt print the build errors to stderr (only the CMake warnings), so you need to print the stdout output to see the actual error message.

By default colcon prints only stderr to console, you can get the stdout by either looking at the content of the log file (`log/rclnodejs_test_msgs/stdout_stderr.log`) or by telling colcon to print both stdout and stderr to console by using the  `--event-handlers console_cohesion+` flag. Without this there is no telling what the actual error is.

minggangw, 2018-08-10:
I see, thanks! I am going to check it.

mikaelarguedas, 2018-08-16:
@minggangw were you able to get more context about the build failure ?

minggangw, 2018-08-17:
@mikaelarguedas Sorry for the delay, I addedd the arguments `--event-handlers console_cohesion+` and got the detailed error message. As the message is too long, I only pasted some of it below. If you want check yourself, the links are:

*  [windows](https://ci.appveyor.com/project/minggangw/rclnodejs/build/develop-827#L584) - failed
*  [ubuntu](https://api.travis-ci.org/v3/job/417087591/log.txt) - passed
*  [macOS](https://circleci.com/gh/minggangw/rclnodejs-1/725) - passed

```bash
ClCompile:
  C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.14.26428\bin\HostX86\x64\CL.exe /c /IC:\proj\rclnodejs\build\rclnodejs_test_msgs\rosidl_generator_c /I"C:\ros2-windows\include" /I"C:\Tools\vcpkg\installed\x64-windows\include" /nologo /W3 /WX- /diagnostics:classic /O2 /Ob2 /D WIN32 /D _WINDOWS /D NDEBUG /D ROSIDL_GENERATOR_C_BUILDING_DLL_rclnodejs_test_msgs /D "CMAKE_INTDIR=\"Release\"" /D rclnodejs_test_msgs__rosidl_generator_c_EXPORTS /D _WINDLL /D _MBCS /Gm- /MD /GS /fp:precise /Zc:wchar_t /Zc:forScope /Zc:inline /Fo"rclnodejs_test_msgs__rosidl_generator_c.dir\Release\\" /Fd"rclnodejs_test_msgs__rosidl_generator_c.dir\Release\vc141.pdb" /Gd /TC /FC /errorReport:queue C:\proj\rclnodejs\build\rclnodejs_test_msgs\rosidl_generator_c\rclnodejs_test_msgs\msg\static_array_non_primitives__functions.c
  static_array_non_primitives__functions.c
Link:
  C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.14.26428\bin\HostX86\x64\link.exe /ERRORREPORT:QUEUE /OUT:"C:\proj\rclnodejs\build\rclnodejs_test_msgs\Release\rclnodejs_test_msgs__rosidl_generator_c.dll" /INCREMENTAL:NO /NOLOGO /LIBPATH:"C:\Tools\vcpkg\installed\x64-windows\lib" /LIBPATH:"C:\Tools\vcpkg\installed\x64-windows\lib\manual-link" "C:\ros2-windows\Lib\builtin_interfaces__rosidl_typesupport_c.lib" "C:\ros2-windows\Lib\builtin_interfaces__rosidl_typesupport_cpp.lib" "C:\ros2-windows\Lib\builtin_interfaces__rosidl_generator_c.lib" "C:\ros2-windows\Lib\builtin_interfaces__rosidl_typesupport_fastrtps_c.lib" "C:\ros2-windows\Lib\builtin_interfaces__rosidl_typesupport_fastrtps_cpp.lib" "C:\ros2-windows\Lib\builtin_interfaces__rosidl_typesupport_introspection_c.lib" "C:\ros2-windows\Lib\builtin_interfaces__rosidl_typesupport_introspection_cpp.lib" "C:\ros2-windows\Lib\rosidl_typesupport_c.lib" "C:\ros2-windows\Lib\rosidl_typesupport_cpp.lib" "C:\ros2-windows\Lib\rosidl_generator_c.lib" "C:\ros2-windows\Lib\rosidl_typesupport_introspection_c.lib" "C:\ros2-windows\Lib\rosidl_typesupport_introspection_cpp.lib" "C:\ros2-windows\Lib\rosidl_generator_c.lib" "C:\ros2-windows\Lib\rosidl_typesupport_introspection_c.lib" "C:\ros2-windows\Lib\rosidl_typesupport_introspection_cpp.lib" kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib "C:\Tools\vcpkg\installed\x64-windows\lib\*.lib" /MANIFEST /MANIFESTUAC:"level='asInvoker' uiAccess='false'" /manifest:embed /PDB:"C:/proj/rclnodejs/build/rclnodejs_test_msgs/Release/rclnodejs_test_msgs__rosidl_generator_c.pdb" /SUBSYSTEM:CONSOLE /TLBID:1 /DYNAMICBASE /NXCOMPAT /IMPLIB:"C:/proj/rclnodejs/build/rclnodejs_test_msgs/Release/rclnodejs_test_msgs__rosidl_generator_c.lib" /MACHINE:X64  /machine:x64 /DLL rclnodejs_test_msgs__rosidl_generator_c.dir\Release\static_array_non_primitives__functions.obj
     Creating library C:/proj/rclnodejs/build/rclnodejs_test_msgs/Release/rclnodejs_test_msgs__rosidl_generator_c.lib and object C:/proj/rclnodejs/build/rclnodejs_test_msgs/Release/rclnodejs_test_msgs__rosidl_generator_c.exp
  rclnodejs_test_msgs__rosidl_generator_c.vcxproj -> C:\proj\rclnodejs\build\rclnodejs_test_msgs\Release\rclnodejs_test_msgs__rosidl_generator_c.dll
AppLocalFromInstalled:
  C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -ExecutionPolicy Bypass -noprofile -File "C:\Tools\vcpkg\scripts\buildsystems\msbuild\applocal.ps1" "C:\proj\rclnodejs\build\rclnodejs_test_msgs\Release\rclnodejs_test_msgs__rosidl_generator_c.dll" "C:\Tools\vcpkg\installed\x64-windows\bin" "rclnodejs_test_msgs__rosidl_generator_c.dir\Release\rclnodej.DA0431C3.tlog\rclnodejs_test_msgs__rosidl_generator_c.write.1u.tlog" "rclnodejs_test_msgs__rosidl_generator_c.dir\Release\vcpkg.applocal.log"
FinalizeBuildStatus:
  Deleting file "rclnodejs_test_msgs__rosidl_generator_c.dir\Release\rclnodej.DA0431C3.tlog\unsuccessfulbuild".
  Touching "rclnodejs_test_msgs__rosidl_generator_c.dir\Release\rclnodej.DA0431C3.tlog\rclnodejs_test_msgs__rosidl_generator_c.lastbuildstate".
Done Building Project "C:\proj\rclnodejs\build\rclnodejs_test_msgs\rclnodejs_test_msgs__rosidl_generator_c.vcxproj" (default targets).
Project "C:\proj\rclnodejs\build\rclnodejs_test_msgs\rclnodejs_test_msgs.vcxproj" (3) is building "C:\proj\rclnodejs\build\rclnodejs_test_msgs\rclnodejs_test_msgs__rosidl_typesupport_c.vcxproj" (6) on node 1 (default targets).
PrepareForBuild:
  Creating directory "rclnodejs_test_msgs__rosidl_typesupport_c.dir\Release\".
  Creating directory "rclnodejs_test_msgs__rosidl_typesupport_c.dir\Release\rclnodej.19F03075.tlog\".
InitializeBuildStatus:
  Creating "rclnodejs_test_msgs__rosidl_typesupport_c.dir\Release\rclnodej.19F03075.tlog\unsuccessfulbuild" because "AlwaysCreate" was specified.
CustomBuild:
  Generating C type support dispatch for ROS interfaces
VcpkgTripletSelection:
  Using triplet "x64-windows" from "C:\Tools\vcpkg\installed\x64-windows\"
ClCompile:
[... cut ...]
C:\proj\rclnodejs\build\rclnodejs_test_msgs\rosidl_generator_py\rclnodejs_test_msgs\rclnodejs_test_msgs_s__rosidl_typesupport_introspection_c.pyd : fatal error LNK1120: 11 unresolved externals [C:\proj\rclnodejs\build\rclnodejs_test_msgs\rclnodejs_test_msgs__rosidl_typesupport_introspection_c__pyext.vcxproj]
Done Building Project "C:\proj\rclnodejs\build\rclnodejs_test_msgs\rclnodejs_test_msgs__rosidl_typesupport_introspection_c__pyext.vcxproj" (default targets) -- FAILED.
Done Building Project "C:\proj\rclnodejs\build\rclnodejs_test_msgs\ALL_BUILD.vcxproj" (default targets) -- FAILED.
Build FAILED.
```

mikaelarguedas, 2018-08-17:
```
C:\Python37\libs\python37.lib : warning LNK4272: library machine type 'x86' conflicts with target machine type 'x64' [C:\proj\rclnodejs\build\rclnodejs_test_msgs\rclnodejs_test_msgs__rosidl_typesupport_introspection_c__pyext.vcxproj]
C:\proj\rclnodejs\build\rclnodejs_test_msgs\rosidl_generator_py\rclnodejs_test_msgs\rclnodejs_test_msgs_s__rosidl_typesupport_introspection_c.pyd : fatal error LNK1120: 11 unresolved externals [C:\proj\rclnodejs\build\rclnodejs_test_msgs\rclnodejs_test_msgs__rosidl_typesupport_introspection_c__pyext.vcxproj]
Done Building Project "C:\proj\rclnodejs\build\rclnodejs_test_msgs\rclnodejs_test_msgs__rosidl_typesupport_introspection_c__pyext.vcxproj" (default targets) -- FAILED.
Done Building Project "C:\proj\rclnodejs\build\rclnodejs_test_msgs\ALL_BUILD.vcxproj" (default targets) -- FAILED.
Build FAILED.
```

This looks to be related to the build environment configuration rather than ROS 2 code.
Can you make sure that you have the 64bit version of the python libraries installed?

---

As this doesn't look to be related to ROS 2 code, I'm going to close this issue.
Feel free to comment here if you still face the issue after investigating the python libraries issue and we can reopen

closed 2018-08-17 by mikaelarguedas

minggangw, 2018-08-20:
Hi @mikaelarguedas It was the wrong Python version which caused this error and I messed them up on the Appveyor. Thanks for your help.