ros2/ros2 #561: error while loading shared libraries: libopencv_calib3d3.so.3.2: cannot open shared object file

DGCHAO, 2018-08-11:
I have a error message when I execute roslaunch turtlebot3_autorace_instinsic_camera_calibration.launch
process[republish-1]: started with pid [3405]
process[camera/image_proc-2]: started with pid [3406]
/opt/ros/kinetic/lib/image_proc/image_proc: error while loading shared libraries: libopencv_calib3d3.so.3.2: cannot open shared object file: No such file or directory
[camera/image_proc-2] process has died [pid 3406, exit code 127, cmd /opt/ros/kinetic/lib/image_proc/image_proc _approximate_sync:=true image_raw:=image __name:=image_proc __log:=/home/burger/.ros/log/ccdd3c4e-9ba4-11e8-9c4b-000c29da7948/camera-image_proc-2.log].
log file: /home/burger/.ros/log/ccdd3c4e-9ba4-11e8-9c4b-000c29da7948/camera-image_proc-2*.log
I think is libopencv_calib3d3.so.3.2 course this error,but I can't solve it.My ROS version is kinetic.This is my first time using OpenCV.I want reinstall of opencv3.If I look into /opt/ros/kinetic/lib, i see libopencv_calib3d.so, libopencv_calib3d.so.3.3, and libopencv_calib3d.so.3.3.1, but not what I need libopencv_calib3d.so.3.2 .

How do I solve this?

DGCHAO, 2018-08-11:
I have try modify /etc/ld.so.conf and then refresh and deleting the devel and build folders and redoing catkin_make would fix it.But is not work.

dirk-thomas, 2018-08-11:
I am not sure if you have seen the [ROS support guidelines](http://wiki.ros.org/Support) but we kindly ask to raise questions like this on [answers.ros.org](http://answers.ros.org/) instead.

The rational for asking all kind of questions in a single place is:
* There are many more people reading questions there so your chances on getting an answer and in a timely manner are much higher.
* In the future other users will search there for similar problems and can find your question and the potential answers.
* The issue tracker in this repo is used to track bugs, feature requests, etc. But your question is not such an action item.
* Also this repo is not related to your problem in any way.

Therefore I will close this ticket here for now. If you have asked your question on answers.ros.org please feel free to add a link to the question to this ticket so that future readers can find the related question.

closed 2018-08-11 by dirk-thomas