ros2/ros2 #562: [Feature request] Node grouping or public/private topic

realdream, 2018-08-14:
## Feature request
#### Feature description

I think ROS2 need a way to insulate the node and also the topic.
like a group of nodes, only nodes in same group can communicate directly. communication between different group should make some specification or some like public topic / private topic.

in ros1 we can set up different `$ROS_MASTER_URI` and use different port to run multi master
which result in:
nodes with same url & port `can` communicate with each other
nodes with different url & port `can not` communicate with each other

it is useful when you have some unconnected data flow and you want to insulate some heavy load topic like `tf`

when come to ROS2, it become more complicated
the topic is valid in entire LAN,
in some common use case, nodes will get the wrong topic from their neighbor
- Same code run on different machine
- Colleagues developing/debugging same nodes
- multi robot with different namespace can run together but will get heavy load on `tf` topic

#### Implementation considerations
Using different port like ROS1 can be a way, but is it possible to make an interface in rcl level?

wjwwood, 2018-08-14:
You can set the `ROS_DOMAIN_ID` to different values in order to "physically" disconnect two ROS graphs in ROS 2. You should also be able to have two nodes in the same process which are each in different domains, see the option in the underlying C API:

http://docs.ros2.org/bouncy/api/rcl/structrcl__node__options__t.html#a9e72fdcd773eae2416d22a661e5ef16f

Unfortunately there's no way to pass this through the C++ or Python API (yet), but you can set it for an entire process using the `ROS_DOMAIN_ID` environment variable. The option just needs to be added to make it programmatic.

Please consider closing this and possibly opening follow up issues that are more specific if the `ROS_DOMAIN_ID` addresses your immediate issue.

There's a bit of documentation on this here:

https://github.com/ros2/ros2/wiki/Overview-of-ROS-2-concepts#discovery

realdream, 2018-08-14:
@wjwwood thank you very much!

closed 2018-08-14 by realdream