ros2/ros2 #563: Bouncy Patch release 2

sloretz, 2018-08-15:
Ticket to aggregate the list of tickets that should be fixed and / or backported

## Tickets to address:

## PRs to backport
- [ ] ros2/common_interfaces#54

## Repositories to version bump and release

## Rerun bloom

wjwwood, 2018-08-20:
As with the last patch release, I would propose that we just fast-forward the packages in the `launch` repo, since only bug fixes and pure additions to the API have been made, see:

https://github.com/ros2/launch/compare/0.5.2...master

mikaelarguedas, 2018-08-20:
Sounds good to me: added `launch` to the list of repos to bump and release

mikaelarguedas, 2018-08-20:
CI with these PRs

Debug:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

Release
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

Packaging:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

Xenial:
Debug:
* Linux [image]
* Linux-aarch64 [image]
Release
* Linux [image]
* Linux-aarch64 [image]
Packaging:
* Linux [image]
* Linux-aarch64 [image]

mikaelarguedas, 2018-08-22:
Artifacts for testing:

Linux: [image]
Linux aarch64: [image]

macOS: [image]

Windows: [image]
Windows Debug: [image]

Xenial: [image]
Xenial aarch64: [image]

wjwwood, 2018-08-22:
The artifacts may not have the latest launch stuff in them due to me forgetting this: https://github.com/ros2/launch/pull/125

I'll check now and help with testing one of the architectures.

mikaelarguedas, 2018-08-22:
Issues that should be fixed in this release:
- https://github.com/ros2/common_interfaces/pull/54
- https://github.com/ros2/ros2cli/issues/127
- https://github.com/ros2/rviz/issues/341
- https://github.com/ros2/launch/issues/89

sloretz, 2018-08-22:
Xenial testing with https://ci.ros2.org/job/ci_packaging_linux/140/ in progress. Here is what I intend to test.

<details>
<summary>TODO: Can include `sensor_msgs/point_cloud2_iterator.hpp` and compile</summary>

```c++
#include <sensor_msgs/point_cloud2_iterator.hpp>

int main()
{
  return 0;
}
```

```cmake

cmake_minimum_version(3.5 REQUIRED)

project(include_test)

find_package(sensor_msgs)

add_executable(test_exec main.cpp)
target_link_libraries(test_exec ${sensor_msgs_LIBRARIES})
target_include_dirs(test_exec ${sensor_msgs_INCLUDE_DIRS})
```

</details>

<details>
<summary>TODO: Can filter parameter list by prefix</summary>

</details>

<details>
<summary>TODO: Can echo a truncated message</summary>

</details>

<details>
<summary>TODO: Can run random ros2cli commands</summary>

</details>

<details>
<summary>TODO: Executor should only allow a node to be added to it once</summary>

</details>

<details>
<summary>TODO: Once removed from an executor the entities shouldn't get callbacks</summary>

</details>

<details>
<summary>TODO: Publishing a LaserScan message with range > intensity</summary>

Could follow instructions in ros2/rviz#341

</details>

mikaelarguedas, 2018-08-22:
New set of archives using repos file from https://github.com/ros2/ros2/pull/565 (including launch fixes)

Linux: [image]
Linux aarch64: [image]

macOS: [image]

Windows: [image]
Windows Debug: [image]

Xenial: [image]
Xenial aarch64: [image]

Repos file from https://github.com/ros2/ros2/pull/565 is now available for source testing as well:

- [ ] Linux Bionic
- [ ] Linux Xenial
- [ ] Linux Bionic aarch64
- [ ] Linux Xenial aarch64
- [ ] MacOS
- [ ] Windows
- [ ] Windows Debug

wjwwood, 2018-08-22:
I did all the tests that @sloretz did, but also:

- run any launch file and ctrl-c it to see that it exits on the first ctrl-c
  - e.g. `ros2 launch demo_nodes_cpp talker_listener.launch.py`

wjwwood, 2018-08-22:
Thanks, by the way, for the detailed how-to @sloretz!

sloretz, 2018-08-22:
Xenial testing with https://ci.ros2.org/job/ci_packaging_linux/142/ looks good

<details>
<summary>Launch closes successfully after CTRL+C  </summary>
This platform isn't expected to hang even without the patch. As expected it works fine.

```
$ ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py
[INFO] [launch]: process[dummy_map_server-1]: started with pid [5164]
[INFO] [launch]: process[robot_state_publisher-2]: started with pid [5165]
[INFO] [launch]: process[dummy_joint_states-3]: started with pid [5166]
[INFO] [launch]: process[dummy_laser-4]: started with pid [5167]
Initialize urdf model from file: /home/developer/workspaces/ros2-linux/share/dummy_robot_bringup/launch/single_rrbot.urdf
Parsing robot urdf xml string.
Link single_rrbot_link1 had 1 children
Link single_rrbot_link2 had 1 children
Link single_rrbot_link3 had 2 children
Link single_rrbot_camera_link had 0 children
Link single_rrbot_hokuyo_link had 0 children
got segment single_rrbot_camera_link
got segment single_rrbot_hokuyo_link
got segment single_rrbot_link1
got segment single_rrbot_link2
got segment single_rrbot_link3
got segment world
Adding fixed segment from world to single_rrbot_link1
Adding moving segment from single_rrbot_link1 to single_rrbot_link2
Adding moving segment from single_rrbot_link2 to single_rrbot_link3
Adding fixed segment from single_rrbot_link3 to single_rrbot_camera_link
Adding fixed segment from single_rrbot_link3 to single_rrbot_hokuyo_link
[INFO] [dummy_laser]: angle inc:        0.004363
[INFO] [dummy_laser]: scan size:        1081
[INFO] [dummy_laser]: scan time increment:      0.000028
^C[WARNING] [launch.LaunchService]: user interrupted with ctrl-c (SIGINT)
signal_handler(2)
[INFO] [launch]: process[dummy_map_server-1]: process has finished cleanly
[INFO] [launch]: process[robot_state_publisher-2]: process has finished cleanly
[INFO] [launch]: process[dummy_laser-4]: process has finished cleanly
signal_handler(2)
signal_handler(2)
[ERROR] [launch]: process[dummy_joint_states-3] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [launch]: sending signal 'SIGTERM' to process[dummy_joint_states-3]
[ERROR] [launch]: process[dummy_joint_states-3] process has died [pid 5166, exit code -15, cmd '/home/developer/workspaces/ros2-linux/lib/dummy_sensors/dummy_joint_states'].
$
```

```
$ ros2 launch demo_nodes_cpp talker_listener.launch.py
[INFO] [launch]: process[talker-1]: started with pid [5200]
[INFO] [launch]: process[listener-2]: started with pid [5201]
[INFO] [talker]: Publishing: 'Hello World: 1'
[INFO] [listener]: I heard: [Hello World: 1]
[INFO] [talker]: Publishing: 'Hello World: 2'
[INFO] [listener]: I heard: [Hello World: 2]
^C[WARNING] [launch.LaunchService]: user interrupted with ctrl-c (SIGINT)
signal_handler(2)
signal_handler(2)
[INFO] [launch]: process[talker-1]: process has finished cleanly
[INFO] [launch]: process[listener-2]: process has finished cleanly
main|developer@470bab580aa2:~/workspaces$
```

</details>

mikaelarguedas, 2018-08-22:
Thanks @sloretz for the detailed steps :+1:

For Linux users: the `ros1_bridge` issue can be tested by following the steps listed in the PR: https://github.com/ros2/ros1_bridge/pull/135

sloretz, 2018-08-23:
@mikaelarguedas the `ros1_bridge` instructions work fine on xenial using lunar and https://ci.ros2.org/job/ci_packaging_linux/142/

mjcarroll, 2018-08-23:
Bionic  testing with debs from: http://repo.ros2.org/ubuntu/testing/

- [ ] Can include `sensor_msgs/point_cloud2_iterator.hpp` and compile
- [ ] Can filter parameter list by prefix
- [ ] Can echo a truncated message
- [ ] Can run random ros2cli commands
- [ ] Executor only allows a node to be added to it once
- [ ] Once removed from an executor the entities don't get callbacks
- [ ] Able to visualize a laserscan with no intensities and ranges > 1
- [ ] Launch closes successfully after CTRL+C

mjcarroll, 2018-08-23:
Windows Release testing:

- [ ] Can include `sensor_msgs/point_cloud2_iterator.hpp` and compile
- [ ] Can filter parameter list by prefix
- [ ] Can echo a truncated message
- [ ] Can run random ros2cli commands
- [ ] Executor only allows a node to be added to it once
- [ ] Once removed from an executor the entities don't get callbacks
- [ ] Able to visualize a laserscan with no intensities and ranges > 1
- [ ] Launch closes successfully after CTRL+C

mikaelarguedas, 2018-08-23:
Windows Debug from binaries:

tested everything but:
- Able to visualize a laserscan with no intensities and ranges > 1

I tested:
- Launch closes successfully after CTRL+C

For talker_listener it works fine everytime
For `dummy_robot_bringup.launch.py` I need to Ctrl-C it 2 times for it to exit
<details>

```
[INFO] [launch]: process[dummy_map_server.EXE-1]: started with pid [11128]
[INFO] [launch]: process[robot_state_publisher.EXE-2]: started with pid [2276]
[INFO] [launch]: process[dummy_joint_states.EXE-3]: started with pid [7132]
[INFO] [launch]: process[dummy_laser.EXE-4]: started with pid [6416]
Initialize urdf model from file: C:\dev\ros2-windows\share\dummy_robot_bringup\launch\single_rrbot.urdf
Parsing robot urdf xml string.
Link single_rrbot_link1 had 1 children
Link single_rrbot_link2 had 1 children
Link single_rrbot_link3 had 2 children
Link single_rrbot_camera_link had 0 children
Link single_rrbot_hokuyo_link had 0 children
got segment single_rrbot_camera_link
got segment single_rrbot_hokuyo_link
got segment single_rrbot_link1
got segment single_rrbot_link2
got segment single_rrbot_link3
got segment world
[INFO] [dummy_laser]: angle inc:        0.004363
[INFO] [dummy_laser]: scan size:        1081
[INFO] [dummy_laser]: scan time increment:      0.000028
Adding fixed segment from world to single_rrbot_link1
Adding moving segment from single_rrbot_link1 to single_rrbot_link2
Adding moving segment from single_rrbot_link2 to single_rrbot_link3
Adding fixed segment from single_rrbot_link3 to single_rrbot_camera_link
Adding fixed segment from single_rrbot_link3 to single_rrbot_hokuyo_link
C:\dev\ros2-windows\Lib\site-packages\launch\launch_service.py:299: DeprecationWarning: The 'warn' method is deprecated, use 'warning' instead
  _logger.warn(base_msg)
[WARNING] [launch.LaunchService]: user interrupted with ctrl-c (SIGINT)
signal_handler(2)
[INFO] [launch]: process[dummy_joint_states.EXE-3]: process has finished cleanly
[INFO] [launch]: process[robot_state_publisher.EXE-2]: process has finished cleanly
[INFO] [launch]: process[dummy_map_server.EXE-1]: process has finished cleanly
[INFO] [launch]: process[dummy_laser.EXE-4]: process has finished cleanly
signal_handler(2)
signal_handler(2)
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "C:\dev\ros2-windows\Lib\site-packages\launch\launch_service.py", line 61, in close_loop
    loop.close()
  File "C:\Python37\lib\asyncio\proactor_events.py", line 528, in close
    self._proactor.close()
  File "C:\Python37\lib\asyncio\windows_events.py", line 775, in close
    if not self._poll(1):
  File "C:\Python37\lib\asyncio\windows_events.py", line 703, in _poll
    status = _overlapped.GetQueuedCompletionStatus(self._iocp, ms)
  File "C:\dev\ros2-windows\Lib\site-packages\launch\utilities\signal_management.py", line 120, in __on_sigint
    __original_sigint_handler(signum, frame)
KeyboardInterrupt
```

</details>

wjwwood, 2018-08-23:
> For talker_listener it works fine everytime

That's interesting, perhaps I'm misremembering, but I think that it might be that Windows ctrl-c works but only if the processes you're running behave. The escalation of signals is not a thing on Windows. So I might have been wrong this morning @mjcarroll, sorry about that.

Feel free to open issues against [`launch`](https://github.com/ros2/launch/issues) for these things, but I don't think it should block the patch release.

mikaelarguedas, 2018-08-24:
The patch release is out: https://discourse.ros.org/t/ros-2-bouncy-patch-2-release/5835

Closing this

closed 2018-08-24 by mikaelarguedas

wjwwood, 2018-08-24:
Thanks @mikaelarguedas!