ros2/ros2 #568: microRTPSAgent problem with libfastrtps.so.1 in ROS2.

OpusK, 2018-08-28:
## Bug report

**Required Info:**

- Operating System:
  - Raspbian Stretch, Ubuntu MATE 16.04
- Installation type:
  - source
- Version or commit hash:
  - master or latest-release
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
This bug is related to microRTPSAgent.
Please refer to the issue below for details.
 - [Segmentation fault](https://github.com/eProsima/micro-RTPS-agent/issues/3)

#### Expected behavior
Normal operation.

#### Actual behavior
When I try to access a topic in ROS2 after creating a topic with microRTPSAgent, "segmentation fault" error occurs in microRTPSAgent.

#### Additional information
I do not know the exact cause, but I have problems using ROS2 and microRTPSAgent on raspberryPi or Upboard. If I do not install ROS2 and use microRTPSAgent, it works fine.
There is no problem in ROS2 itself, but it seems that there is a problem with the linked library used by microRTPSAgent.

I thought that there was a problem with the libfastrtps.so.1 file in ROS2.0 and replaced it with the libfastrtps.so.1 file generated when FastRTPS was installed, and it was executed well without a "segmentation fault" error.

If I have installed FastRTPS and have installed ROS2.0(bouncy) in a binary way:
$ cd /opt/ros/bouncy/lib
$ sudo mv libfastrtps.so.1 libfastrtps.so.2   #for backup
$ sudo cp /usr/local/lib/libfastrtps.so.1 ./

I do not know what is different between FastRTPS and the libfastrtps.so.1 file that ROS 2.0 has, so I do not know if this is the right solution.

However, if there are differences between the two files, and you are aware of these issues, please update them.

OpusK, 2018-08-29:
Sorry, I found a problem with the solution I posted.
With the above method, the agent operates normally, but a "segmentation fault" occurs in ROS2.

Of course, since the Agent is operating normally, I can receive the topic normally with ROS2 in the remote.

wjwwood, 2018-08-29:
So can this be closed?

OpusK, 2018-08-30:
Hi, @wjwwood 

There is an alternative, but the problem is not resolved.
I do not know if it's right to leave an issue here, since I do not know how you manage rmw and FastRTPS when you develop ROS2. Of course, I also left this issue on eProsima.

In summary, the problem is: If I install ROS2 on an SBC such as RaspberryPi (2G RAM) or Upboard (2G RAM) and try to use microRTPSAgent, I get a "segmentation fault".

In other words, microRTPS to support XRCE cannot be used simultaneously with ROS2 in some SBCs.

I have attempted my own attempt as mentioned above, but eventually there is a problem that only one side works and the other side does not work.

wjwwood, 2018-08-30:
So if you remove the `microRTPSAgent` from being used (just use `ros2 topic pub` and `ros2 topic echo` on the two machines) so you still have a problem?

If so, we need more information, e.g.: which OS, what version of ROS 2, etc. And I'd recommend you open the issue on `rclcpp` if you're using C++ or `rclpy` if you're using Python and include a backtrace from `gdb` for the segmentation fault. We cannot debug "it segfaults".

OpusK, 2018-08-30:
> So if you remove the microRTPSAgent from being used (just use ros2 topic pub and ros2 topic echo on the two machines) so you still have a problem?

If I remove only microRTPSAgent in the following situation, ROS2 commands do not work normally. In particular, when I run "ros2 topic list" command, a "segmentation fault" error occurs.
```
If I have installed FastRTPS and have installed ROS2.0(bouncy) in a binary way:
$ cd /opt/ros/bouncy/lib
$ sudo mv libfastrtps.so.1 libfastrtps.so.2 #for backup
$ sudo cp /usr/local/lib/libfastrtps.so.1 ./
```

I wrote the OS and version information on the first post in accordance with the issue form.
As you say, I will issue gdb information to the corresponding repository issue according to the language I use. Thank you.

routiful, 2018-08-30:
I attached debug result in RaspberryPi with above environments which @OpusK set

Commands : **ros2 run --prefix 'gdb -ex run --args' demo_nodes_cpp talker**

```
pi@raspberrypi:~ $ ros2 run --prefix 'gdb -ex run --args' demo_nodes_cpp talker
GNU gdb (Raspbian 7.12-6) 7.12.0.20161007-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/pi/ros2_ws/install/demo_nodes_cpp/lib/demo_nodes_cpp/talker...(no debugging symbols found)...done.
Starting program: /home/pi/ros2_ws/install/demo_nodes_cpp/lib/demo_nodes_cpp/talker 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x768e03b4 in std::vector<std::shared_ptr<eprosima::fastrtps::rtps::TransportDescriptorInterface>, std::allocator<std::shared_ptr<eprosima::fastrtps::rtps::TransportDescriptorInterface> > >::operator=(std::vector<std::shared_ptr<eprosima::fastrtps::rtps::TransportDescriptorInterface>, std::allocator<std::shared_ptr<eprosima::fastrtps::rtps::TransportDescriptorInterface> > > const&)
    () from /home/pi/ros2_ws/install/fastrtps/lib/libfastrtps.so.1
```

dirk-thomas, 2018-08-30:
The information is still not sufficient. Please use a debug build of the code in question as well as provide a full backtrace (not only the last frame).

OpusK, 2018-10-20:
Resolved by [eProsima](https://github.com/eProsima/Micro-XRCE-DDS-Agent/issues/3)

closed 2018-10-20 by OpusK