ros2/ros2 #569: Node ignores yaml configuration when in non-global namespace

christianrauch, 2018-09-01:
## Bug report

Moving a node into a namespace prevents setting parameters via the yaml configuration and blocks `ros2 param list`.

**Required Info:**

- Operating System: Ubuntu 18.04
- Installation type: binary
- Version or commit hash: bouncy
- DDS implementation: Fast-RTPS
- Client library (if applicable): rclcpp

#### Steps to reproduce issue
- move the talker example node ([talker.cpp](https://github.com/ros2/demos/blob/master/demo_nodes_cpp/src/topics/talker.cpp)) into a namespace (e.g. `talker_ns`) via inheriting from `Node("talker", "talker_ns")`:
```C++
#include <chrono>
#include <cstdio>
#include <memory>
#include <string>

#include "rclcpp/rclcpp.hpp"

#include "std_msgs/msg/string.hpp"

using namespace std::chrono_literals;

class Talker : public rclcpp::Node
{
public:
  explicit Talker(const std::string & topic_name) : Node("talker", "talker_ns")
  {
    msg_ = std::make_shared<std_msgs::msg::String>();

    // get parameter from yaml
    int lala;
    bool ret = get_parameter("some_int", lala);
    std::cout << "some_int: " << lala << " ("<<ret<<")" << std::endl;

    auto publish_message =
      [this]() -> void {
        msg_->data = "Hello World: " + std::to_string(count_++);
        RCLCPP_INFO(this->get_logger(), "Publishing: '%s'", msg_->data.c_str())
        pub_->publish(msg_);
      };

    rmw_qos_profile_t custom_qos_profile = rmw_qos_profile_default;
    custom_qos_profile.depth = 7;
    pub_ = this->create_publisher<std_msgs::msg::String>(topic_name, custom_qos_profile);
    timer_ = this->create_wall_timer(1s, publish_message);
  }

private:
  size_t count_ = 1;
  std::shared_ptr<std_msgs::msg::String> msg_;
  rclcpp::Publisher<std_msgs::msg::String>::SharedPtr pub_;
  rclcpp::TimerBase::SharedPtr timer_;
};

int main(int argc, char * argv[])
{
  rclcpp::init(argc, argv);
  auto node = std::make_shared<Talker>("chatter");
  rclcpp::spin(node);
  rclcpp::shutdown();
  return 0;
}
```
- save yaml configuration from: https://github.com/ros2/ros2/wiki/Node-arguments as `demo_params.yaml`
- run node with configuration file: `ros2 run demo_nodes_cpp talker __params:=demo_params.yaml`
- check parameters in another terminal: `ros2 param list`

#### Expected behavior
- terminal with node should show: `some_int: 42 (1)`
- `ros2 param list` should list parameters

#### Actual behavior
- terminal with node shows: `some_int: 0 (0)`
- `ros2 param list` blocks

#### Additional information
If the yaml configuration to set or the API call to retrieve the parameter needs to be adapted, this should be mentioned in the documentation. However, `ros2 param list` should not block in any case.

christianrauch, 2018-09-01:
I missed the documentation at: https://github.com/ros2/rcl/blob/master/rcl_yaml_param_parser/README.md.

The parameters of the yaml file need also be moved into the namespace:
```yaml
talker_ns:
    talker:
        ros__parameters:
            some_int: 42
            a_string: "Hello world"
            some_lists:
                some_integers: [1, 2, 3, 4]
                some_doubles : [3.14, 2.718]
```
However, I guess that the blocking of `ros2 param list` is still unexpected behaviour.

dirk-thomas, 2018-09-01:
You don't need a custom `talker` for this. You can just push the existing `talker` node into a namespace using a command line argument: `__ns:=/talker_ns`.

The problem of not considering the namespace of a node in various cases is being addressed: https://github.com/ros2/rmw/issues/142 After these patches have been merged this can be rechecked if it is still blocking.

closed 2018-09-06 by dirk-thomas