ros2/ros2 #570: Bazel Build System for ROS2 ?

datlife, 2018-09-03:
## Universal Build System
**Required Info:**
- Operating System: All
- Installation type: Binary/Source
- Version or commit hash: N/A
- DDS implementation: N/A
- Client library (if applicable): Python, C++ ( potentially Golang, or any language supported by Bazel)

## Feature request
#### Feature description

Given existing multiple building tools for ROS, described in this blog post - ["Universal Build Tool"](https://design.ros2.org/articles/build_tool.html), I feel confused and not sure how to start as a ROS user ( I have developed a few packages in ROS 1). `Colcon` seems like a solution to address this issue. 

Still, I wonder if there is any plan to integrate Bazel in the future build system for ROS2? I found it much easier to work with compared to CMake.

Thanks,

gavanderhoorn, 2018-09-04:
Are you looking to build all of ROS2 using Bazel (ie: use Bazel as the driver for the entire build), or do you want to be able to specify Bazel as the build tool for a single/a nr of packages?

At least the latter seems possible with [colcon/colcon-ros-bazel](https://github.com/colcon/colcon-ros-bazel).

nuclearsandwich, 2018-09-05:
> Given existing multiple building tools for ROS, described in this blog post - "Universal Build Tool", I feel confused and not sure how to start as a ROS user ( I have developed a few packages in ROS 1). Colcon seems like a solution to address this issue.

colcon is a build tool which is capable of building ros workspaces with one command invocation from the user. When I was first getting familiar with ROS I had difficulty distinguishing between build *systems* and build *tools*. It's a difficult distinction to make since most, if not all, build systems also have an associated build tool. But if you think of a Makefile without `make` or a the contents of a CMakeLists.txt without `cmake` or bazel's BUILD file you'd be getting into build system territory. Software to describe and manage the build lifecycle of other software. A build tool is a user interface to that software. I think of colcon as a "high level" interface which wraps the lower level build tools used by each package applying common conventions as well as passing down desired options to all packages.

> I wonder if there is any plan to integrate Bazel in the future build system for ROS2? I found it much easier to work with compared to CMake.

It's likely that ament_cmake will remain the recommended build system for C/C++ ROS 2 packages for the foreseeable future and if you're getting started with ROS 2 that's what I would recommend.

nuclearsandwich, 2018-09-05:
> Based on the definition in the design article most build systems do not provide a build tool. make and cmake are not build tools (following that definition) since they only operate on a single package.

Thanks for the correction. Haven't re-read the article recently and my intuitive definition didn't include that caveat. Editing above.

Theosakamg, 2018-09-05:
A `colcon` extension exists to build a bazel package (regardless of ROS2) with [` colcon-bazel`] (https://github.com/colcon/colcon-bazel).
Another `colcon` extension to compile the ROS2 package with bazel at [` colcon-ros-bazel`] (https://github.com/colcon/colcon-ros-bazel) but in WIP state.
(need to finish * .bzl rules for colcon workspace)

nuclearsandwich, 2018-09-06:
@datlife can you please let us know whether the discussion above answers your question so we can resolve the issue. I'd also invite you to ask future questions on https://answers.ros.org/ which is better suited to the Question & Answer format than GitHub Issues. Thanks!

datlife, 2018-09-07:
Hi @nuclearsandwich  and @all,

I think the `colcon-ros-bazel` seems to be a solution for my need now. I appreciate your all inputs and advice! 

Please close this issue at anytime. :+1:  . And yes, I will post to answers.ros.org in the future.