ros2/ros2 #572: ROS Buildfarm improvements for Crystal

nuclearsandwich, 2018-09-04:
* [ ] CI / CD jobs on build.ros2.org
  * [ ] Allow nightly jobs to be configured with a .repos file
  * [ ] Allow a CI/devel job that crosses multiple repos (can require common branch name)
  * [ ] Add incremental CI job that starts from the most recent nightly artifacts and overlays source from a .repos file (Focus on Linux first and get other platforms online later)
* [ ] Update the ros_bootstrap repo and any properties that need to directly interact with it.
* [ ] Update Jenkins hosts to latest LTS (https://github.com/ros-infrastructure/buildfarm_deployment/issues/193)
* [ ] Preserve old .deb files in repository
* [ ] Allow use of nvidia-docker (for gazebo or rviz)

nuclearsandwich, 2018-11-17:
> do these updates/changes also include a migration from aptly from reprepro (for "Preserve old .deb files in repository")?
I did some work in September migrating the bootstrap repository to aptly and found a few show-stoppers preventing us from easily adopting it for the buildfarm. To do package preservation in while exploring the possibility of working on those with aptly upstream we're in the process of spinning up a separate snapshot repository host which will store snapshots of repositories at each sync.

gavanderhoorn, 2018-11-17:
@nuclearsandwich wrote:
> I did some work in September migrating the bootstrap repository to aptly and found a few show-stoppers preventing us from easily adopting it for the buildfarm.

Would you still know what those were? I'd be interested to know.

nuclearsandwich, 2018-11-17:
> Would you still know what those were? I'd be interested to know.

The biggest one is https://github.com/aptly-dev/aptly/issues/757 for how we handle python releases. Publishing speed when dealing with multiple distributions and strangely memory usage were also observed anecdotally but I didn't have the tools at hand to actually monitor. I'm hoping to get things written up when we have the snapshot repo announced.

dirk-thomas, 2019-01-17:
@nuclearsandwich Please update this ticket - at least the title which currently mentioned Crystal.

nuclearsandwich, 2019-02-01:
> [x] Preserve old .deb files in repository

While we had originally hoped to integrate this feature into the ROS buildfarm itself the tools we intended to use to achieve it didn't quite meet our needs. I've just announced the [ROS snapshot repository](https://discourse.ros.org/t/announcing-ros-snapshot-repositories/7705) which will provide access to older binary packages.