ros2/ros2 #573: Sourcing setup.bash on ROS2 takes considerably longer compared to ROS1

christianrauch, 2018-09-08:
## Bug report

Sourcing the default ROS2 workspace via `source /opt/ros/bouncy/setup.bash` takes much longer than sourcing a ROS1 workspace. This makes setting it by default in `.bashrc` uncomfortable.

**Required Info:**

- Operating System: Ubuntu 18.04
- Installation type: binary
- Version or commit hash: bouncy

#### Steps to reproduce issue
- install `ros-bouncy-desktop`
- measure the time it takes to source the default workspace: `time source /opt/ros/bouncy/setup.bash`

#### Expected behavior
Sourcing a default ROS1 workspace via `time source /opt/ros/melodic/setup.bash` is done without noticeable delay (0.1 seconds). It, therefore, can be added to the user's `.bashrc` to have a source environment with every new terminal.

#### Actual behavior
In ROS2 sourcing the default workspace takes a considerable amount of time (more than 2 seconds on my PC). Sourcing the default workspace by default in the `.bashrc` is therefore not an option. On less powerfull devices like the Raspberry Pi, sourcing takes even longer (more than 10 seconds).

wjwwood, 2018-09-17:
> Form the name of the source files, I would assume that local_setup.bash is sourcing every package individually (as observed), but setup.bash should source the whole workspace at once.

`local_setup.bash` sources the environment hooks for just one package or just one workspace (depending on where it was located. `setup.bash` will source environment hooks for the current package or workspace _and_ it's dependencies. In both cases each file needs to be sourced by the setup file. There's no single file that aggregates all environment hooks for all packages.

Seanmatthews, 2019-01-08:
I came across this thread looking for the same answer to this awful core problem. I don't know what the real solution should be, but I sped up the sourcing a tad by adding an ampersand at the end of this line in `install/local_setup.bash`:

`_colcon_prefix_bash_source_script "$COLCON_CURRENT_PREFIX/share/${_colcon_package_name}/package.bash"`

dirk-thomas, 2019-01-08:
> adding an ampersand at the end of this line ...

What you are suggesting isn't going to work. While the invocation is faster it simply doesn't set the environment variable it is supposed to. You can't source scripts in the background since the current shell needs to wait for the script in order to update the environment variables.

routiful, 2019-03-19:
Does anybody handle this bug?

EricCousineau-TRI, 2019-03-26:
Just to weigh in, this is the first things that struck me negatively when starting to tinker with ros2.

For others to explore potential solutions, is there a suggested toy benchmark in which people can tinker with and some pointers to what source to play with?

Random thoughts: If this is strongly coupled to decoupling [*], might there be a way to speed up composition?
Can info collection be delegated to python if the file matches some pattern in which the script is more declarative than imperative? (e.g. if the user doesn't supply their own template, put some token near the top, and attempt to fast aggregation beforehand?)

[*] Realized this phrase means nothing. I write it, nonetheless!

Levi-Armstrong, 2019-06-02:
@dirk-thomas We are currently integrating colcon into the ROS Qt Creator plugin and also noticed the pause due to sourcing. In the IDE I can run this in the background sometimes but not always and when finished update the IDE environment, but currently every time I execute commands that need the environment I go ahead and resource the workspace which I believe is excessive. I am trying to understand when a user should resource the workspace? Is it simple like after new package are introduced to the workspace or is it more complicated than that?

clalancette, 2019-06-03:
@Levi-Armstrong Sourcing the workspace generally does things like setup PYTHONPATH, LD_LIBRARY_PATH, and things of that nature.  So you definitely need to do it when introducing new packages so it can set up the environment for the new package.  If you are just iterating on a package by changing the source code, adding/removing libraries, and things of that nature, sourcing the workspace once when you start working is generally sufficient[1].  I'd say that you generally don't need to continually source the workspace.

[1] There are more complicated scenarios where this isn't exactly true, but for the most part it should work and is how I do development.

Levi-Armstrong, 2019-06-03:
@clalancette thank you for the feedback. That is typically been my experience too when developing. What I will do is add an option to the setting which enables auto sourcing workspace and have this on by default. I will then add a button to the project build setting that when clicked will source the workspace and update the IDE environment for those who want to manually source the workspace when needed and they can turn off auto sourcing.

christianrauch, 2019-07-20:
As far as I understand, the source script in ROS2 recursively traverses all packages and calls their source scripts (as shown by `AMENT_TRACE_SETUP_FILES=1 source /opt/ros/dashing/setup.bash`) to gather the environment variables. This involves multiple scripts (about 5) per colcon package. I assume that this is necessary since packages can be relocated, i.e. they can be installed in a separated or merged workspace.

For the binary distribution, which is installed as a merged workspace, the effective change of the environment basically comes down to
```
LD_LIBRARY_PATH=/opt/ros/dashing/opt/yaml_cpp_vendor/lib:/opt/ros/dashing/opt/rviz_ogre_vendor/lib:/usr/lib:/opt/ros/dashing/lib
AMENT_PREFIX_PATH=/opt/ros/dashing
OSPL_VERBOSITY=2
OSPL_URI=file:///opt/ros/dashing/share/opensplice_cmake_module/config/ros_ospl.xml
OSPL_HOME=/usr
OSPL_INFOFILE=<stdout>
ROS_VERSION=2
ROS_PYTHON_VERSION=3
OSPL_ERRORFILE=<stderr>
PYTHONPATH=/opt/ros/dashing/lib/python3.6/site-packages
PATH=/opt/ros/dashing/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ROS_DISTRO=dashing
OSPL_TMPL_PATH=/usr/etc/opensplice/idlpp
```
For a workspace with separated packages, I assume that `LD_LIBRARY_PATH`, `PATH` and `PYTHONPATH` will have multiple entries.

Instead of recursively gathering environment variables from all shell scripts, wouldn't it be possible to cache these variables (as shown above) once a new package gets installed and only use a simple source script with this cache environment from there one?
Alternatively, there should be an option to not recursively iterate through a merged workspace and just set the standard library and executable paths. I believe this is how `virtualenv` is separating environments too.

felixendres, 2019-08-07:
Could this be replaing /opt/ros_$ROS_DISTRO/install/setup.bash with a file that effectively stores the changes to the environment settings like @christianrauch posted? A script to update would be required and run e.g. from the postprocessing script in ros2 Debian packages. So unless something new is installed from .deb packages in "/opt/ros_...", the base setup.bash would be static.

felixendres, 2019-08-09:
I've implemented the caching of environment and found that the bash completion needs to be sourced separately, otherwise the approach works. It works by introducing the caching in local_setup.bash.

The first time all env vars prefixed with ROS, or containing "/ros2_$ROS_DISTRO/" in their values are written to a unique cache file (in the user's home directory, if the directory containing local_setup.bash is not writable). Also, the list of packages (of which the configuration was sourced) is written to a file.

The next time, if the list of packages is the same as before, the going through all packages is skipped and the environment vars are read from file. Finally the configuration of ros2cli is run because of the bash completions.

It works very well for me, when replacing the respective "local_setup.bash" of the system-installation and my workspaces sourcing is instantanous.

Caveat: If environment variables (e.g., PATH) are changed after first usage, before next usage, they are overwritten.

The cache-enabled local_setup.bash is attached as [new_local_setup.bash.txt](https://github.com/ros2/ros2/files/3486293/new_local_setup.bash.txt) and can directly replace /opt/ros2_$ROS_DISTRO/install/local_setup.bash or that of any workspace.
Original to patch would be https://github.com/colcon/colcon-bash/blob/master/colcon_bash/shell/template/prefix.bash.em, but because of the caveat, I am hestiant to create a pull request.

hdino, 2019-08-15:
Since the long loading time of terminals is really annoying when sourcing setup.bash I wanted to try the caching approach. However, my dashing installation on Ubuntu 18.04 from the official repository doesn't have an install folder and replacing local_setup.bash in /opt/ros/dashing caused an error on my fresh system (I think a python script was missing). Therefore, as a temporary workaround I put everything in a script which I source:
```sh
export AMENT_PREFIX_PATH=/opt/ros/dashing
export ROS_DISTRO=dashing
export ROS_PYTHON_VERSION=3
export ROS_VERSION=2

if [ -z "$LD_LIBRARY_PATH" ]; then
  export LD_LIBRARY_PATH=/opt/ros/dashing/opt/yaml_cpp_vendor/lib:/opt/ros/dashing/opt/rviz_ogre_vendor/lib:/opt/ros/dashing/lib
else
  export LD_LIBRARY_PATH=/opt/ros/dashing/opt/yaml_cpp_vendor/lib:/opt/ros/dashing/opt/rviz_ogre_vendor/lib:/opt/ros/dashing/lib:$LD_LIBRARY_PATH
fi

if [ -z "$PATH" ]; then
  export PATH=/opt/ros/dashing/bin
else
  export PATH=/opt/ros/dashing/bin:$PATH
fi

if [ -z "$PYTHONPATH" ]; then
  export PYTHONPATH=/opt/ros/dashing/lib/python3.6/site-packages
else
  export PYTHONPATH=/opt/ros/dashing/lib/python3.6/site-packages:$PYTHONPATH
fi

source /opt/ros/dashing/share/ros2cli/environment/ros2-argcomplete.bash
```

Of course, this is limited to a default installation, but maybe it's useful for someone as long as there is no better working solution.

hdino, 2019-08-20:
Since sourcing the setup files of other workspaces is as slow as the main setup, I extended my script to allow sourcing those as well: [ros2_quick_setup.sh](https://github.com/bam-de/ros2_tools/blob/master/ros2_quick_setup.sh)

Usage is (e.g. in .bashrc):
```sh
source PATH/ros2_quick_setup.sh
setup_ros2_environment /opt/ros/ dashing
add_ros2_workspace /home/user/my_workspace
```

Though this is still a workaround, having one common script to source arbitrary workspaces might be a good idea for further development.

dirk-thomas, 2019-08-20:
While these hacks might work for your use cases locally they don't satisfy the requirement we have for the environment setup. Please see #764 which explicitly enumerated the goals and also references some work in progress to address this problem.

hdino, 2019-08-20:
I know. However, as long as the setup scripts take several seconds even for small workspaces (two packages, each includes only one library/executable) on relatively fast systems (recent Core i5, Samsung 850 Pro SSD), these hacks might be an option to setup the environment. Of course, they were not intended to replace anything and should not be used after these problems have been fixed.

dirk-thomas, 2019-08-20:
> as long as the setup scripts take several seconds even for small workspaces (two packages, each includes only one library/executable) on relatively fast systems (recent Core i5, Samsung 850 Pro SSD),

This is certainly not a common performance for such a setup. For an isolated workspace with ~200 pkgs it takes me 1-2s to source on Ubuntu on a Lenovo P50 using an SSD.

Are you sourcing the `local_setup` file or `setup`? Which platform are you on?

Karsten1987, 2019-08-20:
If I may chip in: If I build my ros2 workspace (274 pkgs) in isolation on my 2016 macbook pro, sourcing the workspace takes more than 20 seconds.

Sourcing it with a `merge-install` setup:
```
 ➭ time source install/local_setup.bash
not found: "/Users/karsten/workspace/osrf/ros2_full/install/share/rttest/local_setup.sh"
not found: "/Users/karsten/workspace/osrf/ros2_full/install/share/rttest/local_setup.bash"
not found: "/Users/karsten/workspace/osrf/ros2_full/install/share/tlsf/local_setup.sh"
not found: "/Users/karsten/workspace/osrf/ros2_full/install/share/tlsf/local_setup.bash"
not found: "/Users/karsten/workspace/osrf/ros2_full/install/share/tlsf_cpp/local_setup.sh"
not found: "/Users/karsten/workspace/osrf/ros2_full/install/share/tlsf_cpp/local_setup.bash"
not found: "/Users/karsten/workspace/osrf/ros2_full/install/share/pendulum_control/local_setup.sh"
not found: "/Users/karsten/workspace/osrf/ros2_full/install/share/pendulum_control/local_setup.bash"

real	0m9.085s
user	0m4.685s
sys	0m3.311s
```

dirk-thomas, 2019-08-20:
@Karsten1987 If I recall correctly on your macOS the title bar of your terminal tab was constantly being updated while the sourcing was ongoing. That seems to indicate another problem which is probably slowing down the process. I would suggest to first look into that / resolve that and then measure again.

hdino, 2019-08-20:
I just did a fresh install of Ubuntu 18.04 on a Ryzen machine with a fast PCIe SSD (native, not in a virtual machine) and dashing from the official repository. Results:
```
time source /opt/ros/dashing/setup.bash

real	0m1,353s
user	0m1,087s
sys	0m0,325s
```
```
time source /opt/ros/dashing/local_setup.bash 

real	0m1,119s
user	0m0,971s
sys	0m0,193s
```

Example workspace containing just `example_interfaces` and `examples`:
```
time source install/setup.bash 

real	0m1,409s
user	0m1,132s
sys	0m0,326s
```
```
time source install/local_setup.bash 

real	0m0,192s
user	0m0,174s
sys	0m0,022s
```

The machine I mentioned above takes about twice as long.

dirk-thomas, 2019-08-20:
> Example workspace containing just `example_interfaces` and `examples`:

While your workspace has only these two items sourcing `install/setup.bash` implicitly also sources `/opt/ros/dashing/local_setup.bash`. The actual time for this small workspace is < 0.2s as you measured.

These results sound pretty common.

hdino, 2019-08-20:
Well, it's about the total time a terminal needs to start if I put these things in, e.g., `.bashrc`. Start up times of 1.4 s (or ~3 s on the slower machine) just for sourcing `/opt/ros/dashing/setup.bash` can be quite annoying.

dirk-thomas, 2019-08-20:
I understand your point of view. That is why work is ongoing to improve the performance.

But in the conversation you mentioned "several seconds even for ... two packages". I just asked for the additional context to clarify that it is actually not the time for only two packages but numerous (likely hundreds) installed Dashing packages. Otherwise we would needed to troubleshoot why your performance was different (which at the end it isn't).

dirk-thomas, 2019-10-04:
All changes have landed for Eloquent.

closed 2019-10-04 by dirk-thomas

LouisChen1905, 2020-03-18:
> What you are suggesting isn't going to work. While the invocation is faster it simply doesn't set the environment variable it is supposed to. You can't source scripts in the background since the [...]

you can add 'wait' after the last child processes. It saves much time.

EricCousineau-TRI, 2020-03-18:
@LouisChen1905 I don't believe that will work for the same reason that Dirk mentioned: using ampersand will use a different process, and `wait` will simply block execution in the parent, but it will not try to "retrieve" the child processes environment variables and such.

Please run this script to see:
```sh
#!/bin/bash
set -euxo pipefail

# Use subshells () to get separate bash processes.
# The syntax ${var:-unset value} will print the value of ${var} if set;
# otherwise, it will print "unset value" (or whatever else you type there).

set-env() {
    export MY_VARIABLE=hello
    echo "Set MY_VARIABLE=hello"
}

# Direct usage, works.
(
    set-env
    # Works
    echo ${MY_VARIABLE:-did not work}
)

# Indirect usage via forking, does not work.
(
    set-env &
    wait
    echo ${MY_VARIABLE:-did not work}    
)
```

Running this inside of script:
```
$ ./env_example.sh 
+ set-env
+ export MY_VARIABLE=hello
+ MY_VARIABLE=hello
+ echo 'Set MY_VARIABLE=hello'
Set MY_VARIABLE=hello
+ echo hello
hello
+ wait
+ set-env
+ export MY_VARIABLE=hello
+ MY_VARIABLE=hello
+ echo 'Set MY_VARIABLE=hello'
Set MY_VARIABLE=hello
+ echo did not work
did not work
```

FYI I tend to try out solutions from GitHub blindly sometimes, so it's helpful if you have the time to vet it first!

hdino, 2020-05-05:
I just did a fresh install of Ubuntu 18.04 and Eloquent from the official repository, i.e. `ros-eloquent-ros-workspace` version `0.8.0-1bionic.20191213.040754` is installed.
```
time source /opt/ros/eloquent/setup.bash 

real	0m2,167s
user	0m1,407s
sys	0m0,787s
```

This is still too slow to do it in every terminal I open.

EricCousineau-TRI, 2020-05-05:
@hdino Just curious; would you be up for trying to profile the execution?
https://stackoverflow.com/a/20855353/7829525
That might help to pinpoint the bottlenecks.

You may also want to ensure that you cut out any excess cruft and isolate the execution, e.g.:
```sh
alias bash-isolate='env -i HOME=$HOME DISPLAY=$DISPLAY SHELL=$SHELL TERM=$TERM USER=$USER PATH=/usr/local/bin:/usr/bin:/bin bash --norc'

# Inject this with instrumentation for profiling
bash-isolate -c 'time source /opt/ros/eloquent/setup.bash'
```
(copy-pasta'd `bash-isolate` from my gist, https://gist.github.com/EricCousineau-TRI/6870970d68f75d23061177a52932a5d1#file-bash_helper_functions-sh-L34 )

hdino, 2020-05-06:
@EricCousineau-TRI I used the first profiling method (`set -x` and `date`) and got the following results:
```
time bash-isolate -c ./benchmark.sh 

real	0m1,762s
user	0m1,242s
sys	0m0,516s
```
[benchmark.sh.txt](https://github.com/ros2/ros2/files/4587553/benchmark.sh.txt) (GitHub doesn't allow .sh)
[profiling.log](https://github.com/ros2/ros2/files/4587526/profiling.log)

The second column contains the accumulated time, the first one the time differences.

The command that takes the longest is in line 6528:
```
0.439764	1.5125	+++++ _ament_ordered_commands='export AMENT_PREFIX_PATH="/opt/ros/eloquent:$AMENT_PREFIX_PATH"
```

richichiare, 2020-05-20:
Hello, due to hardware restrictions, I am still encountering slowdowns when sourcing the setup file in my workspace. Can you give me hints on how to speed-up the process? Thank you.

fujitatomoya, 2020-05-21:
I do not think that we could accept the following as product, since boot time is key especially consumer devices, which reflects on user experience.

- Raspi4/Ubuntu18.04/Eloquent

```
# time source /opt/ros/eloquent/setup.bash

real    0m1.491s
user    0m0.727s
sys     0m0.814s
```

- NVIDIA Jetson Nano/Ubuntu18.04/Eloquent

```
# time source /opt/ros/eloquent/setup.bash

real    0m1.291s
user    0m0.520s
sys     0m0.248s
```

since this is closed, is there any related issue?

richichiare, 2020-05-21:
Is there a way to speed-up the sourcing process in ros2 Eloquent? For example, is there a way to do the sourcing only once (e.g., at boot time), rather than repeating the process for all bashes?
In another thread @dirk-thomas said that Eloquent has better performances (he said that sourcing ~300 pkgs takes about half a second) but still in my NVIDIA Jetson Nano with Ubuntu 18.04 and Eloquent, sourcing ~200 pkgs takes dozens of seconds.
Do you have any advice for reducing this sourcing time?

clalancette, 2020-05-21:
Basically someone needs to take a look at the profiling that @hdino kindly did in https://github.com/ros2/ros2/issues/573#issuecomment-624699716 (or redo it themselves), and spend some time looking at ways to reduce the time spent in those functions.  Do feel free to open another issue to track the fact that it is too slow for your use case, but please provide as much information as possible

dirk-thomas, 2020-10-21:
> I do not think that we could accept the following as product

@fujitatomoya I think everyone agrees that faster would be desired. If you have specific suggestion to improve the time please don't hesitate to open a new ticket or even better pull request.

> Is there a way to speed-up the sourcing process in ros2 Eloquent? For example, is there a way to do the sourcing only once (e.g., at boot time), rather than repeating the process for all bashes?

@richichiare With the changes proposed in colcon/colcon-core#403 and ament/ament_package#120 you should be able to source a workspace once (with `COLCON_TRACE=1` or `AMENT_TRACE_SETUP_FILES=1`) and pipe the output into a file. As long as the starting environment is the same and the content of the workspace hasn't changed you should be able to source the generated file instead of the usual setup file (which should be much faster) to setup the same environment.