ros2/ros2 #574: Latest source code build failure on macOS

youtalk, 2018-09-09:
## Bug report

**Required Info:**

- Operating System:
  - macOS High Sierra
- Installation type:
  - from source
- Version or commit hash:
  - HEAD
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

```
$ colcon build --symlink-install
```

#### Expected behavior

Build success.

#### Actual behavior

Build failure. I found the following error message.

```
...
CMake Error in CMakeLists.txt:
  Imported target "resource_retriever::resource_retriever" includes
  non-existent path

    "/Users/osrf/jenkins-agent/workspace/ci_packaging_osx/ws/install/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.
...
```

#### Additional information

----
## Feature request

#### Feature description

I think `jenkins-agent` is only for CI build so that we should remove the related packages.

#### Implementation considerations

mikaelarguedas, 2018-09-11:
@youtalk there is no `jenkins-agent` hard coded anywhere in the code. Are you sure you are building ROS 2 from source?

Could you provide the exact staps you used to download and compile ROS 2?

youtalk, 2018-09-12:
Oh it's my fault. I used [the latest artifacts](
https://ci.ros2.org/job/ci_packaging_osx/lastSuccessfulBuild/artifact/ws/ros2-package-osx-x86_64.tar.bz2) from ci.ros2.org and overlayed its workspace to another workspace for compilation.
I'm sorry.

youtalk, 2018-09-22:
Without underlay using the nightly artifacts, the build was failure by the same error message.
I found the reason why the error occurred.

```
$ colon build --symlink-install
...
Starting >>> rcl
--- stderr: rcl
CMake Error at test/CMakeLists.txt:9 (find_package):
  By not providing "Findosrf_testing_tools_cpp.cmake" in CMAKE_MODULE_PATH
  this project has asked CMake to find a package configuration file provided
  by "osrf_testing_tools_cpp", but CMake did not find one.

  Could not find a package configuration file provided by
  "osrf_testing_tools_cpp" with any of the following names:

    osrf_testing_tools_cppConfig.cmake
    osrf_testing_tools_cpp-config.cmake

  Add the installation prefix of "osrf_testing_tools_cpp" to
  CMAKE_PREFIX_PATH or set "osrf_testing_tools_cpp_DIR" to a directory
  containing one of the above files.  If "osrf_testing_tools_cpp" provides a
  separate development package or SDK, be sure it has been installed.

---
Failed   <<< rcl	[ Exited with code 1 ]
...
```

The CMakeLists.txt of the test dir of `rcl` package has `ors'_testing_tools_cpp` but it could not be found.
https://github.com/ros2/rcl/blob/master/rcl/test/CMakeLists.txt#L8

youtalk, 2018-09-22:
Ooops, `ros2.repos` contains `osrf_testing_tools_cpp` package so I think it may be my `vcs pull` failure. Sorry for the confusion.
https://github.com/ros2/ros2/blob/master/ros2.repos#L38

closed 2018-09-22 by youtalk