ros2/ros2 #576: Why there are so many ament things in colcon? Will they remain forever in the future releases?

qkx515, 2018-09-11:
For example, using ros2 pkg create, The CMakeLists.txt it generate automatically looks like
```
cmake_minimum_required(VERSION 3.5)
project(testpackage)

# Default to C99
if(NOT CMAKE_C_STANDARD)
  set(CMAKE_C_STANDARD 99)
endif()

# Default to C++14
if(NOT CMAKE_CXX_STANDARD)
  set(CMAKE_CXX_STANDARD 14)
endif()

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
endif()

# find dependencies
find_package(ament_cmake REQUIRED)
# uncomment the following section in order to fill in
# further dependencies manually.
# find_package(<dependency> REQUIRED)

if(BUILD_TESTING)
  find_package(ament_lint_auto REQUIRED)
  # the following line skips the linter which checks for copyrights
  # remove the line when a copyright and license is present in all source files
  set(ament_cmake_copyright_FOUND TRUE)
  # the following line skips cpplint (only works in a git repo)
  # remove the line when this package is a git repo
  set(ament_cmake_cpplint_FOUND TRUE)
  ament_lint_auto_find_test_dependencies()
endif()

ament_package()
```
Why ament remains here? Does this mean even in future releases, we will always use amend as a tradition?

Thanks,
Kaixian QU

mikaelarguedas, 2018-09-11:
`colcon` is a build tool that can build packages of multiple build systems (python, cmake, ament_cmake, ament_python, catkin...).

The build system used for ROS 1 is `catkin`.
The build systems used for ROS 2 are `ament_cmake` and `ament_python`.

More details can be found at http://design.ros2.org/articles/build_tool.html.

I'm going to close this question, in the future we encourage to raise questions like this on [answers.ros.org](http://answers.ros.org/) instead, following the [ROS support guidelines](http://wiki.ros.org/Support).

closed 2018-09-11 by mikaelarguedas

qkx515, 2018-09-11:
Thank you so much. Sorry for bothering you! :-)

ZhenshengLee, 2023-12-25:
more info and discussions can be found in https://discourse.ros.org/t/ros2-build-tools-systems/35173