ros2/ros2 #577: Docs v2

tfoote, 2018-09-12:
* Add the ability for a package to offer .rst documentation that the index will ingest, render, and search over (e.g., add a new url tag to package.xml that specifies a place to fetch the .rst files)
* Add the ability to upload/specify in the index non-package-specific .rst documentation

dirk-thomas, 2018-09-13:
I would add one bullet:
* transfer the content of the ROS 2 wiki into .rst file in a repository which are being ingested by this

tfoote, 2018-09-26:
The site is under test now at https://index.ros.org/

It has hydro through melodic as well as ardent and bouncy
It supports additional readme elements from the package.xml
It renders readmes from the package root and repository root.

There's an external documentation site demostration using:  https://github.com/ros2/ros2_documentation for content.

hidmic, 2019-01-10:
I think we should either re-scope this issue or close it if we can call it done.

tfoote, 2019-01-15:
rst is working and the content is migrated, closing.

closed 2019-01-15 by tfoote