ros2/ros2 #581: ROS2 fails installation on MAC on package rviz_ogre_vendor

ignacioalvmar, 2018-10-02:
## Bug report

**Required Info:**

- Operating System:
  MacOS High Sierra v 10.13.4
- Installation type:
  from Source
- Version or commit hash:
  ROS 2 Bouncy Bolson - 2018-08-24
- DDS implementation:
  Fast-RTPS (default)
- Client library (if applicable):
  N/A

#### Steps to reproduce issue
Following installation instructions from [ros2 wiki](https://github.com/ros2/ros2/wiki/OSX-Development-Setup)

Install dependencies with brew:
```
brew install cmake cppcheck eigen pcre poco python3 tinyxml wget
brew install asio tinyxml2
brew install opencv
```
Install rviz dependencies
```
 brew install qt freetype assimp
 export CMAKE_PREFIX_PATH=$CMAKE_PREFIX_PATH:/usr/local/opt/qt
```
Install other dependencies:
```
python3 -m pip install argcomplete catkin_pkg colcon-common-extensions coverage empy flake8 flake8-blind-except flake8-builtins flake8-class-newline flake8-comprehensions flake8-deprecated flake8-docstrings flake8-import-order flake8-quotes mock nose pep8 pydocstyle pyparsing setuptools vcstool
```
Disable System Integrity Protection following [these instructions](https://developer.apple.com/library/archive/documentation/Security/Conceptual/System_Integrity_Protection_Guide/ConfiguringSystemIntegrityProtection/ConfiguringSystemIntegrityProtection.html)
Create workspace and clone repos:
```
mkdir -p ~/ros2_ws/src
cd ~/ros2_ws
wget https://raw.githubusercontent.com/ros2/ros2/release-latest/ros2.repos
vcs import src < ros2.repos
```
Build ROS2 code
```
cd ~/ros2_ws/
colcon build --symlink-install
```
#### Expected behavior
Complete installation of all ROS2 packages
#### Actual behavior

Failed installation at rviz_ogre_vendor module:
```
Summary: 114 packages finished [26min 51s]
  1 package failed: rviz_ogre_vendor
  3 packages aborted: actionlib_msgs geometry_msgs rcl
  11 packages had stderr output: connext_cmake_module rmw_connext_cpp rmw_connext_shared_cpp rmw_opensplice_cpp rosidl_typesupport_connext_c rosidl_typesupport_connext_cpp rosidl_typesupport_opensplice_c rosidl_typesupport_opensplice_cpp rttest rviz_ogre_vendor tlsf
  71 packages not processed
The file /Users/ignacioalvarez/ros2_ws/None does not exist.
```
#### Additional information
````
$ colcon build --symlink-install
Starting >>> ament_package
Starting >>> ament_flake8
Starting >>> fastcdr
Starting >>> poco_vendor
Starting >>> ament_cppcheck
Starting >>> gtest_vendor
Starting >>> osrf_pycommon
Starting >>> console_bridge
[... 1221 lines cut ...]
/Users/ignacioalvarez/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgreInstanceBatch.h:89: Warning 813: Warning for Ogre::InstanceBatch, base Ogre::MovableObject ignored. Multiple inheritance is not supported in Java.
/Users/ignacioalvarez/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgreSimpleRenderable.h:51: Warning 813: Warning for Ogre::SimpleRenderable, base Ogre::Renderable ignored. Multiple inheritance is not supported in Java.
/Users/ignacioalvarez/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgreSharedPtr.h:342: Warning 503: Can't wrap 'operator unspecified_bool_type' unless renamed to a valid identifier.
/Users/ignacioalvarez/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgreMesh.h:769: Warning 516: Overloaded method Ogre::Mesh::getEdgeList(unsigned short) const ignored,
/Users/ignacioalvarez/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgreMesh.h:760: Warning 516: using Ogre::Mesh::getEdgeList(unsigned short) instead.
/Users/ignacioalvarez/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgreMesh.h:769: Warning 516: Overloaded method Ogre::Mesh::getEdgeList() const ignored,
/Users/ignacioalvarez/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgreMesh.h:760: Warning 516: using Ogre::Mesh::getEdgeList() instead.
/Users/ignacioalvarez/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgreMesh.h:760: Warning 516: Overloaded method Ogre::Mesh::getEdgeList(unsigned short) ignored,
/Users/ignacioalvarez/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgreMesh.h:769: Warning 516: using Ogre::Mesh::getEdgeList(unsigned short) const instead.
/Users/ignacioalvarez/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgreMesh.h:88: Warning 813: Warning for Ogre::Mesh, base Ogre::AnimationContainer ignored. Multiple inheritance is not supported in Java.
/Users/ignacioalvarez/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgreMeshManager.h:60: Warning 813: Warning for Ogre::MeshManager, base Ogre::ManualResourceLoader ignored. Multiple inheritance is not supported in Java.
/Users/ignacioalvarez/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgrePass.h:102: Warning 503: Can't wrap 'operator ()' unless renamed to a valid identifier.
/Users/ignacioalvarez/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgrePass.h:547: Warning 516: Overloaded method Ogre::Pass::getTextureUnitState(unsigned short) const ignored,
/Users/ignacioalvarez/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgrePass.h:539: Warning 516: using Ogre::Pass::getTextureUnitState(unsigned short) instead.
/Users/ignacioalvarez/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgrePass.h:551: Warning 516: Overloaded method Ogre::Pass::getTextureUnitState(Ogre::String const &) const ignored,
/Users/ignacioalvarez/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgrePass.h:543: Warning 516: using Ogre::Pass::getTextureUnitState(Ogre::String const &) instead.
/Users/ignacioalvarez/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgrePass.h:1735: Warning 516: Overloaded method Ogre::Pass::getUserObjectBindings() const ignored,
/Users/ignacioalvarez/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgrePass.h:1729: Warning 516: using Ogre::Pass::getUserObjectBindings() instead.
/Users/ignacioalvarez/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgreTechnique.h:696: Warning 516: Overloaded method Ogre::Technique::getUserObjectBindings() const ignored,
/Users/ignacioalvarez/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgreTechnique.h:690: Warning 516: using Ogre::Technique::getUserObjectBindings() instead.
/Users/ignacioalvarez/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgreRenderTexture.h:55: Warning 516: Overloaded method Ogre::RenderTexture::copyContentsToMemory(Ogre::Box const &,Ogre::PixelBox const &,Ogre::RenderTarget::FrameBuffer) ignored,
/Users/ignacioalvarez/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgreRenderTarget.h:338: Warning 516: using copyContentsToMemory(Ogre::Box const &,Ogre::PixelBox const &,Ogre::RenderTarget::FrameBuffer) instead.
/Users/ignacioalvarez/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgreRenderTexture.h:55: Warning 516: Overloaded method Ogre::RenderTexture::copyContentsToMemory(Ogre::Box const &,Ogre::PixelBox const &) ignored,
/Users/ignacioalvarez/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgreRenderTarget.h:338: Warning 516: using copyContentsToMemory(Ogre::Box const &,Ogre::PixelBox const &) instead.
/Users/ignacioalvarez/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgreRenderTarget.h:338: Warning 516: Overlo

clalancette, 2018-10-03:
Hm, interesting.  I don't see this exact error when trying to build on my High Sierra machine, but the build is failing for me in `rviz_default_plugins`.  Regardless, this looks like a problem we'll have to track down.  I'll mark it as a bug.

clalancette, 2018-10-08:
After looking at some incoming PRs, it looks like this is fixed by https://github.com/ros2/rviz/pull/356 and https://github.com/ros2/rclcpp/pull/566 .

routiful, 2018-10-28:
@clalancette Did you sovled build failed in `rviz_default_plugins`? 
I downloaded rviz and rclcpp in ros2 branch. But still it is failed.

asimonov, 2019-05-05:
I still see this error on MacOs High Sierra 1.13.

```
Finished <<< ament_cmake_pep8 [2.55s]
Finished <<< ament_cmake_pclint [2.56s]
Finished <<< ament_cmake_clang_format [3.10s]
--- stderr: rviz_ogre_vendor                                                                                                           
Undefined symbols for architecture x86_64:
  "Ogre::ZipArchive::ZipArchive(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, _zzip_plugin_io*)", referenced from:
      _wrap_new_ZipArchive(_object*, _object*) in OgrePYTHON_wrap.cxx.o
  "Ogre::EmbeddedZipArchiveFactory::addEmbbeddedFile(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, unsigned char const*, unsigned long, bool (*)(unsigned long, void*, unsigned long))", referenced from:
      _wrap_EmbeddedZipArchiveFactory_addEmbbeddedFile(_object*, _object*) in OgrePYTHON_wrap.cxx.o
  "Ogre::EmbeddedZipArchiveFactory::removeEmbbeddedFile(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      _wrap_EmbeddedZipArchiveFactory_removeEmbbeddedFile(_object*, _object*) in OgrePYTHON_wrap.cxx.o
  "Ogre::EmbeddedZipArchiveFactory::EmbeddedZipArchiveFactory()", referenced from:
      _wrap_new_EmbeddedZipArchiveFactory(_object*, _object*) in OgrePYTHON_wrap.cxx.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[5]: *** [lib/_Ogre.so] Error 1
make[4]: *** [Components/Python/CMakeFiles/_Ogre.dir/all] Error 2
make[4]: *** Waiting for unfinished jobs....
Undefined symbols for architecture x86_64:
  "Ogre::ZipArchive::ZipArchive(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, _zzip_plugin_io*)", referenced from:
      _Java_org_Ogre_OgreJNI_new_1ZipArchive_1_1SWIG_10 in OgreJAVA_wrap.cxx.o
      _Java_org_Ogre_OgreJNI_new_1ZipArchive_1_1SWIG_11 in OgreJAVA_wrap.cxx.o
  "Ogre::EmbeddedZipArchiveFactory::addEmbbeddedFile(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, unsigned char const*, unsigned long, bool (*)(unsigned long, void*, unsigned long))", referenced from:
      _Java_org_Ogre_OgreJNI_EmbeddedZipArchiveFactory_1addEmbbeddedFile in OgreJAVA_wrap.cxx.o
  "Ogre::EmbeddedZipArchiveFactory::removeEmbbeddedFile(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      _Java_org_Ogre_OgreJNI_EmbeddedZipArchiveFactory_1removeEmbbeddedFile in OgreJAVA_wrap.cxx.o
  "Ogre::EmbeddedZipArchiveFactory::EmbeddedZipArchiveFactory()", referenced from:
      _Java_org_Ogre_OgreJNI_new_1EmbeddedZipArchiveFactory in OgreJAVA_wrap.cxx.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[5]: *** [java/libs/libOgreJNI.jnilib] Error 1
make[4]: *** [Components/Java/CMakeFiles/OgreJNI.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [ogre-master-ca665a6-prefix/src/ogre-master-ca665a6-stamp/ogre-master-ca665a6-build] Error 2
make[1]: *** [CMakeFiles/ogre-master-ca665a6.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< rviz_ogre_vendor	[ Exited with code 2 ]
Aborted  <<< ament_lint_common
```

I think I am on Xcode 9.4.

```
(ros2_crystal) alexeysimonov@courage:~/dev/ros2_crystal_ws$ xcode-select -v
xcode-select version 2349.
(ros2_crystal) alexeysimonov@courage:~/dev/ros2_crystal_ws$ clang --version
Apple LLVM version 9.1.0 (clang-902.0.39.2)
Target: x86_64-apple-darwin17.7.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
(ros2_crystal) alexeysimonov@courage:~/dev/ros2_crystal_ws$ xcode-select -p
/Library/Developer/CommandLineTools
(ros2_crystal) alexeysimonov@courage:~/dev/ros2_crystal_ws$ pkgutil --pkg-info=com.apple.pkg.CLTools_Executables
package-id: com.apple.pkg.CLTools_Executables
version: 9.4.1.0.1.1528165917
volume: /
location: /
install-time: 1555893224
groups: com.apple.FindSystemFiles.pkg-group 
```

any advice?

asimonov, 2019-05-05:
just confirmed i see the same error with Xcode 9.4.1, 10.0 and 10.1