ros2/ros2 #582: ROS2 Performance Testing

mjcarroll, 2018-10-08:
Create a suite of performance tests for ROS2 as well as integrate performance testing into continuous integration or nightly testing.

- [ ] Write performance tests (e.g., ping-pong and burst for a variety of message types)
- [ ] Identify a couple of EC2 instance types to test on
- [ ] Establish performance benchmarks on those instance types
- [ ] Integrate performance test results into Jenkins

dirk-thomas, 2018-10-08:
It would be good to also test the performance of the intra process comm (C++ only).

chapulina, 2019-10-17:
Not sure if there are still plans to create a new performance test suite, but we've been working into integrating https://github.com/ApexAI/performance_test/ into the ROS2 buildfarm. @cottsay has been leading that effort.

Here's the current implementation: http://build.ros2.org/view/Eci/job/Eci__nightly-performance_ubuntu_bionic_amd64/

It's a work in progress and feedback is appreciated.

rotu, 2020-02-18:
@chapulina, @cottsay, cc: @joespeed What happened to the implementation on the ros2 build server?

chapulina, 2020-02-18:
The performance tests were put behind an authentication wall while they're being worked on. We want to prevent users from looking at the unfinished tests and taking the wrong conclusions. See this issue: https://github.com/ros2/buildfarm_perf_tests/issues/21

mjcarroll, 2020-06-11:
This issue has largely been covered and there are now nightly performance tests: 

Eloquent Nightly Performance: http://build.ros2.org/view/Eci/job/Eci__nightly-performance_ubuntu_bionic_amd64/plot/
Foxy Nightly Performance: http://build.ros2.org/view/Fci/job/Fci__nightly-performance_ubuntu_focal_amd64/plot/

I'm going to close this in favor of tracking enhancements here: https://github.com/ros2/buildfarm_perf_tests

closed 2020-06-11 by mjcarroll