ros2/ros2 #584: No amend_tools

m4xst3r, 2018-10-11:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04
- Installation type:
  - Via Debian Packages

#### Steps to reproduce issue

```
1. Install ROS2 
2. Create ROS2 Wosrkspace
3. D0 Ament Tutorial
```

#### Expected behavior
src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install
Build ament.py
#### Actual behavior
No ament_tools Folder in the ament Folder
#### Additional information
In the repo https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos there is als no ament_tools Folder.

----
## Feature request

#### Feature description

#### Implementation considerations

dirk-thomas, 2018-10-11:
Your information is inconsistent. At the top you mentioned that you installed using Debian packages. But below it looks like that you are trying to do a from source build.

The current master branch (and future released of ROS 2) do not support `ament_tools` anymore. Instead use `colcon` as described in the current instructions.

kshokn2, 2018-10-12:
I also tried the same method. There is no ament_tools directory in '~/ros2_ws/src/ament/' after using svc command. I found find ament_tools dir in this repo ( https://github.com/ament ).

After git clone, I got the error about asio when building FastRTPS. So maybe I succeeded to build  the half of progress.

m4xst3r, 2018-10-12:
Okay I have overlooked the Note regards the ament_tools, sorry about that. I sure did not Install from Source these are Steps I did to install ROS 2:

1. I installed ROS 2 Ardent for Ubuntu Xenial, with the instructions from this site: https://github.com/ros2/ros2/wiki/Linux-Install-Debians
2. I wanted to install the build Tool colcon but on this Page https://github.com/ros2/ros2/wiki/Colcon-Tutorial it refers to the ament_tutorial for ROS2 Releases before Bouncy

dirk-thomas, 2018-10-12:
@kshokn2 above I already described the reason why `ament_tools` isn't available in a recent workspace anymore.

@m4xst3r if you are using the Ardent release you can continue to use `ament_tools`. But then you can't mix that set of packages with anything from the `master` branch. You should stick to the `.repos` file from the `ardent` tag then. Otherwise you are mixing code from two very different times / states which are likely incompatible with each other.