ros2/ros2 #587: Failed to building rmw.

kshokn2, 2018-10-16:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 18.04
- Installation type:
  -  Debian Packages
- Version or commit hash:
  -  https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue
 I've followed the installation instructions and then used colcon to build it
```
mkdir -p ~/ros2_ws/src
cd ~/ros2_ws
wget https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos
vcs import ~/ros2_ws/src < ros2.repos
colcon build --symlink-install
```

#### Expected behavior
Complete building of ros2.0 packages
#### Actual behavior
failed to build rmw
```
Starting >>> rmw
Scanning dependencies of target rmw
[ 13%] Built target gmock
[ 17%] Building C object CMakeFiles/rmw.dir/src/convert_rcutils_ret_to_rmw_ret.c.o
[ 21%] Building C object CMakeFiles/rmw.dir/src/allocators.c.o
[ 30%] Built target gmock_main
[ 34%] Building C object CMakeFiles/rmw.dir/src/names_and_types.c.o
[ 39%] Building C object CMakeFiles/rmw.dir/src/sanity_checks.c.o
In file included from /home/shyoon/ros2_ws/src/ros2/rmw/rmw/include/rmw/types.h:30:0,
                 from /home/shyoon/ros2_ws/src/ros2/rmw/rmw/include/rmw/allocators.h:23,
                 from /home/shyoon/ros2_ws/src/ros2/rmw/rmw/src/allocators.c:15:
/home/shyoon/ros2_ws/src/ros2/rmw/rmw/include/rmw/serialized_message.h:24:10: fatal error: rcutils/types/char_array.h: No such file or directory
 #include "rcutils/types/char_array.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/rmw.dir/src/allocators.c.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/rmw.dir/build.make:62: recipe for target 'CMakeFiles/rmw.dir/src/allocators.c.o' failed
In file included from /home/shyoon/ros2_ws/src/ros2/rmw/rmw/include/rmw/types.h:30:0,
                 from /home/shyoon/ros2_ws/src/ros2/rmw/rmw/include/rmw/convert_rcutils_ret_to_rmw_ret.h:25,
                 from /home/shyoon/ros2_ws/src/ros2/rmw/rmw/src/convert_rcutils_ret_to_rmw_ret.c:15:
/home/shyoon/ros2_ws/src/ros2/rmw/rmw/include/rmw/serialized_message.h:24:10: fatal error: rcutils/types/char_array.h: No such file or directory
 #include "rcutils/types/char_array.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/rmw.dir/build.make:86: recipe for target 'CMakeFiles/rmw.dir/src/convert_rcutils_ret_to_rmw_ret.c.o' failed
make[2]: *** [CMakeFiles/rmw.dir/src/convert_rcutils_ret_to_rmw_ret.c.o] Error 1
In file included from /home/shyoon/ros2_ws/src/ros2/rmw/rmw/include/rmw/types.h:30:0,
                 from /home/shyoon/ros2_ws/src/ros2/rmw/rmw/include/rmw/names_and_types.h:28,
                 from /home/shyoon/ros2_ws/src/ros2/rmw/rmw/src/names_and_types.c:15:
/home/shyoon/ros2_ws/src/ros2/rmw/rmw/include/rmw/serialized_message.h:24:10: fatal error: rcutils/types/char_array.h: No such file or directory
 #include "rcutils/types/char_array.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/rmw.dir/build.make:134: recipe for target 'CMakeFiles/rmw.dir/src/names_and_types.c.o' failed
make[2]: *** [CMakeFiles/rmw.dir/src/names_and_types.c.o] Error 1
In file included from /home/shyoon/ros2_ws/src/ros2/rmw/rmw/include/rmw/types.h:30:0,
                 from /home/shyoon/ros2_ws/src/ros2/rmw/rmw/include/rmw/sanity_checks.h:26,
                 from /home/shyoon/ros2_ws/src/ros2/rmw/rmw/src/sanity_checks.c:15:
/home/shyoon/ros2_ws/src/ros2/rmw/rmw/include/rmw/serialized_message.h:24:10: fatal error: rcutils/types/char_array.h: No such file or directory
 #include "rcutils/types/char_array.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/rmw.dir/src/sanity_checks.c.o] Error 1
CMakeFiles/rmw.dir/build.make:158: recipe for target 'CMakeFiles/rmw.dir/src/sanity_checks.c.o' failed
CMakeFiles/Makefile2:323: recipe for target 'CMakeFiles/rmw.dir/all' failed
make[1]: *** [CMakeFiles/rmw.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Failed   <<< rmw	[ Exited with code 2 ]
```
#### Additional information

 If i change the address 'rcutils/types/char_array.h' more detailed , and then fail to link
```
Starting >>> rmw
Scanning dependencies of target rmw
[ 13%] Built target gmock
[ 17%] Building C object CMakeFiles/rmw.dir/src/allocators.c.o
[ 26%] Built target gmock_main
[ 30%] Building C object CMakeFiles/rmw.dir/src/convert_rcutils_ret_to_rmw_ret.c.o
[ 34%] Building C object CMakeFiles/rmw.dir/src/sanity_checks.c.o
[ 39%] Building C object CMakeFiles/rmw.dir/src/names_and_types.c.o
[ 43%] Building C object CMakeFiles/rmw.dir/src/node_security_options.c.o
[ 47%] Building C object CMakeFiles/rmw.dir/src/validate_namespace.c.o
[ 52%] Building C object CMakeFiles/rmw.dir/src/validate_full_topic_name.c.o
[ 56%] Building C object CMakeFiles/rmw.dir/src/validate_node_name.c.o
[ 60%] Linking C shared library librmw.so
[ 65%] Built target rmw
Scanning dependencies of target test_validate_node_name
Scanning dependencies of target test_serialized_message
Scanning dependencies of target test_validate_namespace
Scanning dependencies of target test_validate_full_topic_name
[ 69%] Building CXX object test/CMakeFiles/test_validate_node_name.dir/test_validate_node_name.cpp.o
[ 73%] Building CXX object test/CMakeFiles/test_validate_namespace.dir/test_validate_namespace.cpp.o
[ 78%] Building CXX object test/CMakeFiles/test_serialized_message.dir/test_serialized_message.cpp.o
[ 82%] Building CXX object test/CMakeFiles/test_validate_full_topic_name.dir/test_validate_full_topic_name.cpp.o
[ 86%] Linking CXX executable test_serialized_message
[ 91%] Linking CXX executable test_validate_namespace
[ 95%] Linking CXX executable test_validate_full_topic_name
CMakeFiles/test_serialized_message.dir/test_serialized_message.cpp.o: In function `test_serialized_message_default_initialization_Test::TestBody()':
test_serialized_message.cpp:(.text+0x2a): undefined reference to `rcutils_get_zero_initialized_char_array'
test_serialized_message.cpp:(.text+0x51): undefined reference to `rcutils_char_array_init'
test_serialized_message.cpp:(.text+0x220): undefined reference to `rcutils_char_array_fini'
CMakeFiles/test_serialized_message.dir/test_serialized_message.cpp.o: In function `test_serialized_message_resize_Test::TestBody()':
test_serialized_message.cpp:(.text+0x6a4): undefined reference to `rcutils_get_zero_initialized_char_array'
test_serialized_message.cpp:(.text+0x6cb): undefined reference to `rcutils_char_array_init'
test_serialized_message.cpp:(.text+0x8c2): undefined reference to `rcutils_char_array_resize'
test_serialized_message.cpp:(.text+0xc6b): undefined reference to `rcutils_char_array_resize'
test_serialized_message.cpp:(.text+0x11a5): undefined reference to `rcutils_char_array_fini'
collect2: error: ld returned 1 exit status
test/CMakeFiles/test_serialized_message.dir/build.make:98: recipe for target 'test/test_serialized_message' failed
make[2]: *** [test/test_serialized_message] Error 1
CMakeFiles/Makefile2:1180: recipe for target 'test/CMakeFiles/test_serialized_message.dir/all' failed
make[1]: *** [test/CMakeFiles/test_serialized_message.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[100%] Linking CXX executable test_validate_node_name
[100%] Built target test_validate_namespace
[100%] Built target test_validate_full_topic_name
[100%] Built target test_validate_node_name
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Failed   <<< rmw	[ Exited with code 2 ]
```

dirk-thomas, 2018-10-16:
There error message indicates that you are trying to build the latest code of the `rmw` package but don't have the latest version of the `rcutils` package. The missing file does exist in the expected location on the default branch (see https://github.com/ros2/rcutils/blob/ba1fe742b30d31f60b6b6ecbae442ebc6d8b69b9/include/rcutils/types/char_array.h).

You are likely building an inconsistent set of packages / repositories. If you want to build some packages using the latest code from the master branch you will likely also need to the latest code of dependencies and can't use the Debian packages of the latest release for them.

kshokn2, 2018-10-22:
@dirk-thomas 
Thank you for reply!! Sorry for delay.
Okay, I understand. I have a question about the method of ros2 building.
If I install ros2 with debian packages, don't i need to follow [this tutorial](https://github.com/ros2/ros2/wiki/Colcon-Tutorial) ?

dirk-thomas, 2018-10-22:
If you only want to use the existing Debian packages you don't need to follow that tutorial. That tutorial describes how to build the current default branches of ROS 2 repositories from source.