ros2/ros2 #589: RMW configuration from XML only

JuanCarlos-eProsima, 2018-10-17:
We have implemented a new feature to apply the QoS configuration for RMW only from XML files, avoiding the current configuration set by code. To do that we have opened these pull requests to the related projects:

- https://github.com/ros2/rmw/pull/152
- https://github.com/ros2/rmw_fastrtps/pull/230
- https://github.com/ros2/system_tests/pull/301

mjcarroll, 2018-11-12:
I have a few concerns about this set of changes.

1. I'm a little unclear on why we would use a flag to replace a configuration, rather than some sort of automatic resolution. In this case, in order to cause the configuration to change, the code must be changed in order to make a custom configuration usable.
2. It also introduces a flag that assumes that RMW configuration can be done from XML, rather than some generic external mechanism.
3. This will only work for FastRTPS.

I think that a better approach would be to allow an external configuration set options that are not otherwise set via the `rmw_qos_profile_t` struct, and then provide another mechanism for externally overriding the options that are set in code.  We can then express this in a way that is not XML/FastRTPS specific.

JuanCarlos-eProsima, 2018-11-22:
Hi @mjcarroll,
What would you think about creating an environment variable to enable this '*XML Only*' management?
This approach only would affect "rmw_fastrtps", keeping the same code for the other packages.

dirk-thomas, 2018-11-22:
There could be a FastRTPS specific environment variable which overrides all settings. While I don't think that this approach would be portable it would certainly be doable. That wouldn't affect any of the RMW API though since it would be done out-of-band in `rmw_fastrtps` with no knowledge to the interface or higher level code. It would probably also mean that introspection tools like `ros2 node info` won't be able to show the overridden values but still show the user the configuration chosen in the source code which might be misleading.

Therefore we think a generally supported policy as described above would make more sense.

richiware, 2018-11-26:
Your approach will be "an external configuration set options". I'm not sure how to approach this. Sorry but I have a limited knowledge of ROS2 architecture. I can only think in three ways:

* Using a high-level API function (example: a function in RCLCPP). But I think this functionality is out-of-topic for this API level.

* Using a direct RMW function. But as far as I know, there is not direct access to RMW layer, at least using a high-level API as RCLCPP.

* Using a external mechanism from ROS2, as an OS mechanism like environment variables. For example a `RMW_LEAVE_MIDDLEWARE_DEFAULT_QOS'. This variable might serve to warning rmw middlware implementation to leave default middleware qos.

dirk-thomas, 2018-11-28:
In order to encourage writing reusable components I don't think the first two options are viable. Imo it is not desired to require changing the code (either calling `rclcpp` or `rmw`) to affect such configuration.

When deploying a component (rather then writing it) the user should have the ability to a) specify additional configuration which amends the options chosen in the code (e.g. a QoS setting not supported by ROS 2) and b) specify overriding configuration values (e.g. increasing the history size or changing a topic from best effort to reliable or the other way around).

Therefore adding an option `config_only_from_xml` which required code to be changed isn't aligned with the described use case. An environment variable would be one way to provide such configurations mentioned in a) and b). Since there is no standard for that atm this at least for now needs to be a rmw specific solution.

JuanCarlos-eProsima, 2018-11-29:
Hi @dirk-thomas ,

Thank you for the clarification. We are going to apply the "Environment Variable" option to don't affect any other package and we are going to use the name `RMW_LEAVE_MIDDLEWARE_DEFAULT_QOS'.

dirk-thomas, 2018-11-29:
Since this is not a cross rmw option the environment variable should be FastRTPS specific.

JuanCarlos-eProsima, 2018-11-29:
ok, we rename the variable to `RMW_FASTRTPS_LEAVE_MIDDLEWARE_DEFAULT_QOS' to show that it's only applied to rmw_fastrtps

nuclearsandwich, 2018-12-06:
Is this issue resolved now that https://github.com/ros2/rmw_fastrtps/pull/243 has merged?

closed 2018-12-06 by mjcarroll