ros2/ros2 #590: Memory management improvements for Crystal

wjwwood, 2018-10-24:
This is a meta ticket to collect some related tasks that have been targeted for the Crystal release and that are related to memory management within ROS 2 and it's various layers.

- Expand usage of `rcutils_allocator_t`
  - `rmw`: all rmw functions which may allocate memory internally need to take an allocator so it can be passed down from the `rcl` functions which call it
  - `rosidl` for C types: all code generators (including `rosidl_generator_c` as well as middleware specific code generators) should be updated to take allocators where it makes sense (e.g. array initialization functions)
  - `rosidl` for C++ types: the std::allocator pattern is already in use, but it should be audited to make sure it's being used correctly
- Provide mechanism to use memory allocated by the middleware (get a message loan from the middleware, fill it with data, use it to publish, return the loan)
- Remove unnecessary copies and memory allocations in publish/take calls (more generally in calls made in "steady state" of typical nodes)
- Audit documentation for declarations about dynamic memory usage
  - `rcutils`
  - `rmw`
  - `rosidl`
  - `rcl`

mjcarroll, 2018-11-19:
A few unnecessary allocations were discovered in `rmw_fastrtps`: https://github.com/eProsima/Fast-RTPS/pull/332

dirk-thomas, 2019-01-17:
@wjwwood  Please update this ticket - at least the title which currently mentions Crystal.

gonzodepedro, 2019-01-21:
Memory allocation audit was conducted (see: /ros2/rcl#366), issues found are documented [here](https://gist.github.com/gonzodepedro/e3a4c2d496ef3cbe605e9d53bef61555)

wjwwood, 2022-06-02:
This issue is out of date, I'm just going to close it for now.

closed 2022-06-02 by wjwwood