ros2/ros2 #593: Migrate uuid_msgs to ros2

jacobperron, 2018-10-25:
## Feature request

#### Feature description

The plan is to use UUIDs at identifiers for goals in ROS 2 actions.
Since goal IDs will be included in various message/service types (e.g. goal requests and responses) and there already exists a ROS message package [uuid_msgs](https://github.com/ros-geographic-info/unique_identifier/tree/master/uuid_msgs), this is a proposal to migrate the package to the ros2 org.

#### Implementation considerations

There is already a ROS 2 port of `uuid_msgs` pending (ros-geographic-info/unique_identifier#13), but I propose we add the message definitions to a new repo **ros2/unique_identifier** and package `unique_identifier_msgs`.

Along with the package containing the ROS message definition, we should also have some language-specific libraries for working UUIDs that could go in separate packages. For example, the structure of the repo might look like:

- unique_identifier
  - unique_identifier_c
  - unique_identifier_cpp
  - unique_identifier_msgs
  - unique_identifier_python

#### Resources

- https://discourse.ros.org/t/uuid-msgs-porting-to-ros2
- https://github.com/ros-geographic-info/unique_identifier
- ROS 2 Action Plan: https://github.com/ros2/ros2/issues/583

stonier, 2018-10-29:
+1 to a pure msgs repository and separate language specific support for those.

stonier, 2018-10-29:
Looks like @jack-oquin has been off the radar for a while though :( No github activity since April and no response to that PR for a week either. I'd be inclined to move on and make this happen. We can retroactively work back in with Jack if he returns, but I suspect if he does, he'll be fine with the move anyway.

@tfoote @mikaelarguedas I think `uuid_msgs` would be a useful candidate for `common_interfaces` with supporting libraries as other repo's in the ros2 org. What process is there for making that decision? 

Whether we have it as part of the ros2 org or elsewhere, I'm happy to help if @jacobperron doesn't move on it more quickly than I am able to.

jacobperron, 2018-10-31:
@stonier It's lower on my priority list, help is appreciated if you get to it first.

jack-oquin, 2018-11-01:
> Looks like @jack-oquin <https://github.com/jack-oquin> has been off the
> radar for a while though :( No github activity since April and no response
> to that PR for a week either. I'd be inclined to [...]

Sure am.

stonier, 2018-11-04:
Thanks for chiming in @jack-oquin. Hope life is treating you well.

@jacobperron I'll wait till you have dropped something in the ros2 organisation. If that's looking like taking some time, let me know. I'm blocked on releasing several packages till that is there.

jacobperron, 2018-11-05:
@stonier Created a repo if you want to point your PR there.

stonier, 2018-11-18:
@jacobperron checking in to see if I can help moving this forward. I think the checklist would be something like:

1. Bloom `unique_identifier_msgs` for `crystal`
2. Create a ros2/unique_identifier repo
3. Split python and cpp implementations into `unique_identifier_py`, `unique_identifier_cpp` packages
4. Bloom unique identifier repository (so something is available asap)
5. Remove the boost dependency from the cpp implementation / c implementation(?)

I could have an example of 3. very quickly (partway through it already) and could do the blooming as well.

jacobperron, 2018-11-18:
@stonier 

I will make sure `unique_identifier_msgs` gets bloomed after #609.

Regarding the `unique_identifier` repo/packages, I'm not actively working on this, so help is appreciated. The question becomes should we add it to ros2 org? In the ros2 org, I don't think we will release it until the boost dependency is removed and I'm not sure yet if it will be used in any core packages. It's possible it could be used in the action client libraries (or rcl_action itself) (@sloretz @hidmic thoughts?).
I noticed there is already a fork: https://github.com/ros2/unique_identifier. If it turns out to be useful for the action implementations then I'm happy to work on it and review any PRs to help move it along.

In any case, if you would like to proceed we can always fork any work done back into the ros2 org.

PR #609 (2018-11-21): Add unique_identifier_msgs
Resolves #593 

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]
closed 2018-11-21 by jacobperron