ros2/ros2 #594: disagree about making old ros2 wiki decommissioned completely

itfanr, 2018-10-31:
## Bug report

#### Expected behavior
Ros2 wiki used to be  maintained at github with repository.

#### Actual behavior
Ros2 wiki has been decommissioned completely and moved to https://index.ros.org/doc/ros2.
The page  https://index.ros.org/doc/ros2 is under construction but the decommission of old ros2 wiki make a lot of urls become dead link.

## Feature request
Keep old Ros2 wiki at github and give readers some words like:
 "the doc is outdated, please move to xxx url"

dirk-thomas, 2018-10-31:
Each previous wiki page contain a deep link to the new location of the content so it should be easy to reach for users. Keeping the duplicate content in the wiki seems to only have disadvantages (readers missing the note about being outdated or trying to edit the content). What is the motivation for your request to keep the old content?

itfanr, 2018-11-01:
I think opensouce project shoud keep compatibility and shoud always add docs and links instead of deleting useful links significantly.

closed 2018-11-01 by itfanr