ros2/ros2 #595: Cannot receive large block data messages

IcanChen, 2018-11-01:
## Cannot receive large block data messages

**Required Info:**

- Operating System:
  - Ubuntu 18.04 bouncy

#### Steps to reproduce issue

publish an nav_msgs/msg/OccupancyGrid type message; when the msg data exceeds a certain size, it will not be received. for example：40m*40m occupancygrid map.

#### Expected behavior
Run cmd: ros2 topic echo  /topic_name
display the topic data.
#### Actual behavior
can not display the topic data.

IcanChen, 2018-11-01:
For “Sending large data”, Fast-RTPS is described as follows:
[image]

But how to set ASYNCHRONOUS_PUBLISH_MODE in ROS2.0 ?

dirk-thomas, 2018-11-01:
Please produce ready-to-execute reproducible steps.

Can you successfully subscribe to the message in another node (rather then the command line tool)? Does the image demo (https://index.ros.org/doc/ros2/Quality-of-Service/) work for you since it is able to send large successfully in our tests?

> But how to set ASYNCHRONOUS_PUBLISH_MODE in ROS2.0 ?

There is no need to configure FastRTPS for this - the rmw implementation already does that behind the scene

IcanChen, 2018-11-02:
I fixed. Just need to set default rmw_qos_profile.

closed 2018-11-02 by IcanChen

kangleorc, 2021-02-03:
how did you fixed the problem？