ros2/ros2 #596: cannot get topic from ros1_bridge by ros2 topic list

itfanr, 2018-11-01:
## Bug report

**Required Info:**

- Operating System:
   Ubuntu 16.04.1

- Installation type:
  ros bouncy:  binaries from ros2 repo
  ros1_bridge: compile by myself

- DDS implementation:
   rmw_implementation used: Fast-RTPS

- Client library (if applicable):
rclcpp

#### Steps to reproduce issue
1. start a ros1 project
2. start ros1_bridge:
 ` ros2 run ros1_bridge dynamic_bridge `
3. list topic and service at ros2 side:
`ros2 topic list`
`ros2 service list`

#### Expected behavior
get topics from terninal

#### Actual behavior
Can only got service, that is topic list is empty

#### Additional information
I got topic cotent by:
`ros2 run demo_nodes_cpp listener -t some_topic`

#### Implementation considerations
There are some bugs in command `ros2 topic`. I am a beginner at ros2, so I cannot fix it by myself now.

dirk-thomas, 2018-11-01:
The `dynamic_bridge` does conceptionally not support this use case. I'm works by monitoring both sides of the bridge and only when there is a publisher on one side and a subscriber in the other side it initiates the bridging of that topic.

In your case there are publisher in the ROS 1 side but no subscribers in the ROS 2 side. Therefore the dynamic bridge doesn't bridge the topics yet. That is mostly for efficiency reasons since there is no need to do so since nobody is interested in the data.

You should be able to.`ros2 topic echo` the data though since that will create a subscriber which will initiate the bridge.

Another option is to use a different bridge - one of the static or parametrized ones.

itfanr, 2018-11-01:
@dirk-thomas  Thank you. I will look into ros1_bridge later, or I will  make all topic bridged and test the ros1_bridge.

closed 2018-11-01 by dirk-thomas