ros2/ros2 #597: change terminology of static and (un)bounded dynamic arrays

dirk-thomas, 2018-11-02:
When we wrote the design article about [interface definitions](http://design.ros2.org/articles/interface_definition.html) we needed to distinguish three different kind of arrays and came up with these terms:
* "static array": arrays with a static size
* "dynamic unbounded arrays": arrays with a dynamic size and no upper boundary
* "dynamic bounded arrays": arrays with a dynamic size and an upper boundary

In preparation of the migration to [IDL](https://www.omg.org/spec/IDL/About-IDL/) we should change our terminology to match the specification.

* `array` equates to "static array" from above
* (unbounded) `sequence` equates to "dynamic unbounded arrays" from above
* `bounded sequence` equates to "dynamic bounded arrays" from above

This ticket will serve as a meta ticket for PRs renaming types, functions, variables, files, etc.

dirk-thomas, 2018-11-02:
All referenced PRs are ready for review.

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

wjwwood, 2018-11-02:
Renaming sounds fine to me.

mjcarroll, 2018-11-02:
Looks like there were a few places where `array` -> `sequence` caused us to violate line lengths.

jacobperron, 2018-11-02:
Should ROS message types in [rcl_interfaces/test_msgs](https://github.com/ros2/rcl_interfaces/tree/master/test_msgs/msg) be renamed for consistency?

dirk-thomas, 2018-11-02:
* Fixed the header guard in https://github.com/ros2/rosidl/pull/313/commits/14a5b3ee9a6411c77ec06eb50c80356e01425803#diff-e120ebd828429d798c09b4002e0667aeR15.
* Fixed the line length in https://github.com/ros2/system_tests/pull/307/commits/6157b1fe7cb57cc8276ea063a96e2e9c88798079#diff-831fe03f2d54d3547eae0bc2f01db942R1158.

[image]

> Should ROS message types in rcl_interfaces/test_msgs be renamed for consistency?

That could be done. Since it doesn't affect the API touched by the IDL work I didn't include it in this set of patches.

dirk-thomas, 2018-11-02:
And another linter fix: https://github.com/ros2/rosidl/pull/313/commits/2d11c370cd1a034fd76e2292bdc6fb02eff0970c#diff-e989a753d48150acb8507440277b41b4R15

CI only running the linter of the failing test: [image]

closed 2018-11-02 by dirk-thomas