ros2/ros2 #598: Unable to Link with Bouncy Release

DLu, 2018-11-05:
# Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04 via VirtualBox
- Installation type: Binary
- Version or commit hash:
  - ros-bouncy-ament-tools: 0.5.0-0bionic.20180719.223430
  - python3-colcon-core: 0.3.12-1
  - rclcpp - 0.5.1-0bionic.20180823.142109
- DDS implementation: Default
- Client library (if applicable): rclcpp

#### Steps to reproduce issue
 * Clone this repo into a ros2 workspace: https://github.com/DLu/simple_ros2_example (ros2 branch, which is default)
 * `colcon build --symlink-install --packages-select awesome_library fantastic_node`

#### Expected behavior
Link library from awesome_library into fantastic_node.

#### Actual behavior
Does not link.
```
fnode.cpp:(.text+0x17d): undefined reference to `awesome_library::Awesomeness::Awesomeness()'
```
See [answers.ros.org](https://answers.ros.org/question/307412/ros2-linking/) for more info.

#### Additional information
 * It worked for @allenh1 but using [everything from source](https://github.com/DLu/simple_ros2_example/pull/1#issuecomment-435508651), so something may be missing from the latest release of something.

dirk-thomas, 2018-11-05:
For my this fails to build with the following error message:

```
/tmp/ws/src/simple_ros2_example/fantastic_node/src/fnode.cpp:2:10: fatal error: awesome_library/awesomeness.h: No such file or directory
 #include <awesome_library/awesomeness.h>
```

Looking at the CMake code of `awesome_library` that makes sense since it doesn't export any include directories. The line which tries to do so uses a variable `include_dirs` which doesn't exist: https://github.com/DLu/simple_ros2_example/blob/885417d0bb6c85fd072e8c038329452b9b2c21bd/awesome_library/CMakeLists.txt#L10

Changing the line to the following will fix the wrong export:

```
ament_export_include_directories(include)
```

Also the install rule is incorrect - please check the install space of `awesome_library`. You want either of the following options to not install the header to `<install>/awesome_library/include/awesome_library/awesome_library/awesomeness.h`:

```
install(
  DIRECTORY include/
  DESTINATION include)
```

or

```
install(
  DIRECTORY include/${PROJECT_NAME}/
  DESTINATION include/${PROJECT_NAME})
```

dirk-thomas, 2018-11-05:
Please avoid cross posting the same questions in multiple locations.

closed 2018-11-05 by dirk-thomas