ros2/ros2 #599: ROS2 Fails to build at rviz_ogre_vendor on MacOS Mojave (10.14)

zackbright, 2018-11-12:
## Bug report

**Required Info:**

- Operating System:
  - MacOS Mojave (10.14)
- Installation type:
  - from source
- Version or commit hash:
  - release-latest as of 11/10/18
[5a7323f](https://github.com/ros2/ros2/commit/5a7323fd3dfdc66c45c9c2d841a2d412764f5184)
- DDS implementation:
  -  Fast-RTPS
- Client library (if applicable):
  - rviz_ogre_vendor

#### Steps to reproduce issue

Build from source per the instructions on:
[OSX-Development-Setup](https://index.ros.org/doc/ros2/OSX-Development-Setup/)

No additional middleware installed.
Using a machine running MacOS Mojave (10.14).
All dependencies installed per the instructions above.
Using xcode10 may be important per build issues with Ogre as they tend to have [some](https://github.com/OGRECave/ogre/issues/817) [issues](https://github.com/OGRECave/ogre/issues/894) when new releases of Xcode come out.

#### Expected behavior
Successful build without error.

#### Actual behavior

Build fails with error:

```
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[5]: *** [lib/_Ogre.so] Error 1
make[4]: *** [Components/Python/CMakeFiles/_Ogre.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [ogre-master-ca665a6-prefix/src/ogre-master-ca665a6-stamp/ogre-master-ca665a6-build] Error 2
make[1]: *** [CMakeFiles/ogre-master-ca665a6.dir/all] Error 2
make: *** [all] Error 2
make: INTERNAL: Exiting with 9 jobserver tokens available; should be 8!
---
Failed   <<< rviz_ogre_vendor	[ Exited with code 2 ]
```

#### Additional information

This is roughly a duplicate of this issue [#581](https://github.com/ros2/ros2/issues/581)

But that issue was closed with it noted that the issue should be re-opened with a new ticket if the issue persisted.

I have already tried modifying the CMAKE files for Ogre to ensure that [per this issue] (https://github.com/OGRECave/ogre/issues/894) I have CMAKE_OSX_ARCHITECTURES=x86_64 set. But this was already configured in the cmake cache. 

See below for output of streams.log from the build failure:

```
Invoking command in '/Users/zackbright/ros2_ws/build/rviz_ogre_vendor': AMENT_PREFIX_PATH=/Users/zackbright/ros2_ws/install/ament_cmake:/Users/zackbright/ros2_ws/install/ament_cmake_target_dependencies:/Users/zackbright/ros2_ws/install/ament_cmake_export_interfaces:/Users/zackbright/ros2_ws/install/ament_cmake_export_dependencies:/Users/zackbright/ros2_ws/install/ament_cmake_test:/Users/zackbright/ros2_ws/install/ament_cmake_python:/Users/zackbright/ros2_ws/install/ament_cmake_libraries:/Users/zackbright/ros2_ws/install/ament_cmake_include_directories:/Users/zackbright/ros2_ws/install/ament_cmake_export_link_flags:/Users/zackbright/ros2_ws/install/ament_cmake_export_libraries:/Users/zackbright/ros2_ws/install/ament_cmake_export_include_directories:/Users/zackbright/ros2_ws/install/ament_cmake_export_definitions:/Users/zackbright/ros2_ws/install/ament_cmake_core:/Users/zackbright/ros2_ws/install/ament_package CMAKE_PREFIX_PATH=/Users/zackbright/ros2_ws/install/ament_cmake:/Users/zackbright/ros2_ws/install/ament_cmake_target_dependencies:/Users/zackbright/ros2_ws/install/ament_cmake_export_interfaces:/Users/zackbright/ros2_ws/install/ament_cmake_export_dependencies:/Users/zackbright/ros2_ws/install/ament_cmake_test:/Users/zackbright/ros2_ws/install/ament_cmake_python:/Users/zackbright/ros2_ws/install/ament_cmake_libraries:/Users/zackbright/ros2_ws/install/ament_cmake_include_directories:/Users/zackbright/ros2_ws/install/ament_cmake_export_link_flags:/Users/zackbright/ros2_ws/install/ament_cmake_export_libraries:/Users/zackbright/ros2_ws/install/ament_cmake_export_include_directories:/Users/zackbright/ros2_ws/install/ament_cmake_export_definitions:/Users/zackbright/ros2_ws/install/ament_cmake_core PATH=/Users/zackbright/ros2_ws/install/ament_cmake/bin:/Users/zackbright/ros2_ws/install/ament_cmake_target_dependencies/bin:/Users/zackbright/ros2_ws/install/ament_cmake_export_interfaces/bin:/Users/zackbright/ros2_ws/install/ament_cmake_export_dependencies/bin:/Users/zackbright/ros2_ws/install/ament_cmake_test/bin:/Users/zackbright/ros2_ws/install/ament_cmake_python/bin:/Users/zackbright/ros2_ws/install/ament_cmake_libraries/bin:/Users/zackbright/ros2_ws/install/ament_cmake_include_directories/bin:/Users/zackbright/ros2_ws/install/ament_cmake_export_link_flags/bin:/Users/zackbright/ros2_ws/install/ament_cmake_export_libraries/bin:/Users/zackbright/ros2_ws/install/ament_cmake_export_include_directories/bin:/Users/zackbright/ros2_ws/install/ament_cmake_export_definitions/bin:/Users/zackbright/ros2_ws/install/ament_cmake_core/bin:${PATH} PYTHONPATH=/Users/zackbright/ros2_ws/install/ament_package/lib/python3.6/site-packages${PYTHONPATH} SHLVL=2 _=/usr/bin/env __CF_USER_TEXT_ENCODING=0x1F5:0x0:0x0 /usr/local/bin/cmake --build /Users/zackbright/ros2_ws/build/rviz_ogre_vendor -- -j8 -l8
[ 12%] Performing build step for 'ogre-master-ca665a6'
[  0%] Built target OgreRTShader_swig_compilation
[  0%] Swig source
[  0%] Swig source
[  0%] Built target OgreOverlay_swig_compilation
[  0%] Built target OgreBites_swig_compilation
[  0%] Swig source
/Users/zackbright/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgreAny.h:414: Warning 302: Identifier 'any_cast' redefined (ignored),
/Users/zackbright/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgreAny.h:49: Warning 302: previous definition of 'any_cast'.
/Users/zackbright/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgreAny.h:414: Warning 302: Identifier 'any_cast' redefined (ignored),
/Users/zackbright/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgreAny.h:49: Warning 302: previous definition of 'any_cast'.
[... 1032 lines cut ...]
[ 99%] Linking CXX shared library ../../lib/macosx/RenderSystem_GL3Plus.dylib
[ 99%] Built target RenderSystem_GL
[ 99%] Built target RenderSystem_GL3Plus
[ 99%] Linking CXX shared module ../../lib/_Ogre.so
Undefined symbols for architecture x86_64:
  "Ogre::ZipArchive::ZipArchive(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, _zzip_plugin_io*)", referenced from:
      _wrap_new_ZipArchive(_object*, _object*) in OgrePYTHON_wrap.cxx.o
  "Ogre::EmbeddedZipArchiveFactory::addEmbbeddedFile(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, unsigned char const*, unsigned long, bool (*)(unsigned long, void*, unsigned long))", referenced from:
      _wrap_EmbeddedZipArchiveFactory_addEmbbeddedFile(_object*, _object*) in OgrePYTHON_wrap.cxx.o
  "Ogre::EmbeddedZipArchiveFactory::removeEmbbeddedFile(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      _wrap_EmbeddedZipArchiveFactory_removeEmbbeddedFile(_object*, _object*) in OgrePYTHON_wrap.cxx.o
  "Ogre::EmbeddedZipArchiveFactory::EmbeddedZipArchiveFactory()", referenced from:
      _wrap_new_EmbeddedZipArchiveFactory(_object*, _object*) in OgrePYTHON_wrap.cxx.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[5]: *** [lib/_Ogre.so] Error 1
make[4]: *** [Components/Python/CMakeFiles/_Ogre.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [ogre-master-ca665a6-prefix/src/ogre-master-ca665a6-stamp/ogre-master-ca665a6-build] Error 2
make[1]: *** [CMakeFiles/ogre-master-ca665a6.dir/all] Error 2
make: *** [all] Error 2
make: INTERNAL: Exiting with 9 jobserver tokens available; should be 8!
Invoked command in '/Users/zackbright/ros2_ws/build/rviz_ogre_vendor' returned '2': AMENT_PREFIX_PATH=/Users/zackbright/ros2_ws/install/ament_cmake:/Users/zackbright/ros2_ws/install/ament_cmake_target_dependencies:/Users/zackbright/ros2_ws/install/ament_cmake_export_interfaces:/Users/zackbright/ros2_ws/install/ament_cmake_export_dependencies:/Users/zackbright/ros2_ws/install/ament_cmake_test:/Users/zackbright/ros2_ws/install/ament_cmake_python:/Users/zackbright/ros2_ws/install/ament_cmake_libraries:/Users/zackbright/ros2_ws/install/ament_cmake_include_directories:/Users/zackbright/ros2_ws/install/ament_cmake_export_link_flags:/Users/zackbright/ros2_ws/install/amen [... cut ...] all/ament_cmake_export_libraries/bin:/Users/zackbright/ros2_ws/install/ament_cmake_export_include_directories/bin:/Users/zackbright/ros2_ws/install/ament_cmake_export_definitions/bin:/Users/zackbright/ros2_ws/install/ament_cmake_core/bin:${PATH} PYTHONPATH=/Users/zackbright/ros2_ws/install/ament_package/lib/python3.6/site-packages${PYTHONPATH} SHLVL=2 _=/usr/bin/env __CF_USER_TEXT_ENCODING=0x1F5:0x0:0x0 /usr/local/bin/cmake --build /Users/zackbright/ros2_ws/build/rviz_ogre_vendor -- -j8 -l8

```

clalancette, 2018-11-12:
Yeah, I also ran into this in Mojave, along with other issues.  As it says at the top of https://index.ros.org/doc/ros2/OSX-Development-Setup/, we currently only officially support Sierra with ROS 2.  I've gotten High Sierra to work, but Mojave is going to need some dedicated effort to make work.  However, this is a low-priority work item for the core ROS 2 team, so any help moving forward is appreciated.

zmk5, 2018-12-19:
I am having the same problem. Could it be due to Mojave making OpenGL deprecated?

francesco-romano, 2019-01-07:
I also have this problem in HighSierra. @clalancette Did you try something in particular?

clalancette, 2019-01-07:
No, with fixes as of November, it just worked for me on High Sierra.

That being said, I've since upgraded my Mac Mini to Mojave, so I can't test on the latest anymore.

clalancette, 2019-01-09:
@bhatsach Not as far as I know.  It's a rather low priority for us at the moment; if you have some time to debug it and provide some PRs, that would be great.

bhatsach, 2019-01-09:
Thanks @clalancette, do you have any pointers for debugging/fixing the issue? I am very new to ROS code and will give this a shot if possible

zackbright, 2019-01-09:
I'm personally not sure and have committed minimal time so far to figuring out what is up. My initial guess is to look deeper into the Ogre codebase and [this issue](https://github.com/OGRECave/ogre/issues/894) I referenced above. It's still open (and now pinned).

I could totally be wrong on that though, this is just a guess. I might consider playing around with this over the weekend, but I'm pretty swamped.

I am able to get the binaries to run, but for now I'm just using a Ubuntu machine and/or VM for everything I need.

francesco-romano, 2019-01-10:
@bhatsach no. But as I needed to test "only" some parts of ROS2 (without the GUI for now), I disable the compilation of rviz (and rqt btw, as I had other issues with qt)

bhatsach, 2019-01-10:
Thanks for the replies. I got the build to work on High Sierra using the following steps:
1. I upgraded my XCode version from 8.x to 9.x and this fixed the above error related to "Undefined symbols for architecture x86_64"
2. I ran into another issue while building "image_tools" package wherein certain variables "CvMat" and "CV_WINDOW_AUTOSIZE" were not found. Downgrading opencv version from 4.0.1 to 3.4.5 fixed this as these variables are not present in the versions >= 4.x

I am guessing the above error for Mojave is also related to XCode as ogre depends on the same. Downgrading to XCode 9.x might be a good way to unblock oneself for now

francesco-romano, 2019-01-10:
Good catch. I have Xcode 10 on High Sierra. This might be the reason. Unfortunately I cannot downgrade it. I think also Mojave needs Xcode 10.
I had also the problems on `image_tools` in `ros2/demo`. I will downgrade opencv

asimonov, 2019-05-26:
may be related to #700

asimonov, 2019-06-15:
I could build ROS2 Dashing on fresh Mojave install.
In addition to standard instructions had to do:
* install java
* brew install pyqt5 - for sip to work. but simply installing sip did not work
* pip3 install lxml

Karsten1987, 2019-06-15:
I've opened a PR which avoids building the java & python components of ogre. That helped me a lot to compile it correctly on mojave.
You might want to give it a shot: https://github.com/ros2/rviz/pull/414

DanielLuu, 2019-07-16:
> I've opened a PR which avoids building the java & python components of ogre. That helped me a lot to compile it correctly on mojave.
> You might want to give it a shot: [ros2/rviz#414](https://github.com/ros2/rviz/pull/414)

How do I actually try this? Is it a command line option or do I have to actually build rviz before I try the install?

Karsten1987, 2019-07-16:
Not sure I completely understand, but given that the referenced PR got merged, you should just be able to compile the master branch of RViz.

clalancette, 2019-09-24:
I believe this is solved by ros2/rviz#414, so I'm going to close this out.  Feel free to reopen if this is still a problem on the latest.

closed 2019-09-24 by clalancette