ros2/ros2 #601: Request shutdown of a node from another node in client libraries

brawner, 2018-11-13:
This issue was also discovered while porting rqt_top and probably related to issue #600. In ros1, there is functionality to request shutdown of nodes through a call to rosnode. Upon obtaining a reference to another node, it would be useful in Ros2 client libraries to have the ability to request shutdown of peer nodes.

dirk-thomas, 2018-11-13:
I am not sure this functionality is generally desired. I would rather see this as a service a node could decide to offer is desired.

gavanderhoorn, 2018-11-13:
With the proper authentication in place this could be something to consider for ROS 2, but in general the fact that *anyone* could XMLRPC a request to a node to shut itself down was considered a major security issue in ROS 1 (albeit a bit unfairly, as it did not consider bad actors at all).

brawner, 2018-11-13:
It certainly makes sense for restrictions to exist on this functionality. Maybe a request_shutdown would only come from a node that's been given elevated privileges, and only on the local machine

gavanderhoorn, 2018-11-13:
For a managed node it could make more sense to request it nicely to transition to its `shutdown` state.

That would not require any special functionality to be added.

dirk-thomas, 2018-11-13:
> For a managed node it could make more sense to request it nicely to transition to its `shutdown` state.

That is already available through the lifecycle service requesting a transition to that state.

gavanderhoorn, 2018-11-13:
> > For a managed node it could make more sense to request it nicely to transition to its `shutdown` state.
> 
> That is already available through the lifecycle service requesting a transition to that state.

yes. That was the suggestion.

tfoote, 2018-11-14:
> It certainly makes sense for restrictions to exist on this functionality. Maybe a request_shutdown would only come from a node that's been given elevated privileges, and only on the local machine

I think that as we start getting into this level of complexity it should be provided as a layer on top and go through standard topics or services that can have security applied to them. As with lifecycle nodes requesting the shutdown is already available. And other nodes could expose that in the same or similar ways if they choose. And if a node is managed through another launching mechanism the launching mechanism could expose that through an API as well. But I'd advocate that we not add new primitives for this specific use case that can be handled using conventions on top of the current primitives. I don't think that it's a good assumption that any node should by default have access to shutdown a node if you just have it's name.