ros2/ros2 #604: rosbag2 build getting fail

Rohita83, 2018-11-14:
We have pull rosbag2 pkg from https://github.com/ros2/rosbag2 into ~/ros2_ws/src path. Then We tried to build (under ros2_ws workspace) in bouncy and getting below error. Anyone can suggest us this issue.

colcon build --packages-select rosbag2

Starting >>> rosbag2 
[0.971s] ERROR:colcon.colcon_cmake.task.cmake.build:Failed to find the following files:
- /home/user/ros2_ws/install/shared_queues_vendor/share/shared_queues_vendor/package.sh
- /home/user/ros2_ws/install/rosbag2_test_common/share/rosbag2_test_common/package.sh
- /home/user/ros2_ws/install/rosbag2_storage/share/rosbag2_storage/package.sh
Check that the following packages have been built:
- shared_queues_vendor
- rosbag2_test_common
- rosbag2_storage
Failed   <<< rosbag2	[ Exited with code 1 ]

Summary: 0 packages finished [0.86s]
  1 package failed: rosbag2
--------------------------------------------

Rohita83, 2018-11-16:
Currently "release-latest/ros2.repos" source workspace is working fine means build successfully but when we are fetching source from "master/ros2.repos" to get all of the latest fixes and make build is getting failed in bouncy. We have followed "https://index.ros.org/doc/ros2/Installation/" instructions and tried on multiple system as well.

Error snapshot:

CMakeFiles/rosidl_generator_c_interfaces__rosidl_generator_c.dir/build.make:662: recipe for target 'CMakeFiles/rosidl_generator_c_interfaces__rosidl_generator_c.dir/rosidl_generator_c/rosidl_generator_c/msg/bounded_array_primitives__functions.c.o' failed
make[2]: *** [CMakeFiles/rosidl_generator_c_interfaces__rosidl_generator_c.dir/rosidl_generator_c/rosidl_generator_c/msg/bounded_array_primitives__functions.c.o] Error 1
CMakeFiles/Makefile2:794: recipe for target 'CMakeFiles/rosidl_generator_c_interfaces__rosidl_generator_c.dir/all' failed
make[1]: *** [CMakeFiles/rosidl_generator_c_interfaces__rosidl_generator_c.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

<== Command '. /home/user/ros2_master_ws/build/rosidl_generator_c/cmake__build.sh && /usr/bin/make -j4 -l4' failed in '/home/user/ros2_master_ws/build/rosidl_generator_c' with exit code '2'
<== Command '. /home/user/ros2_master_ws/build/rosidl_generator_c/cmake__build.sh && /usr/bin/make -j4 -l4' failed in '/home/user/ros2_master_ws/build/rosidl_generator_c' with exit code '2'

clalancette, 2018-11-16:
Sorry, you'll have to give a lot more information on what you are doing.  Please make sure to fill out all of the information from the original issue template, including what system you are on, the exact sequence of steps you followed, etc.  Since this works fine for us and on our [Continuous Integration](https://ci.ros2.org) server, there is likely a missing step in what you did.

Rohita83, 2018-11-16:
*** Failed to build rosidl_generator_c with latest ros2 master branch.***

Bug report:

Required Info:

    Operating System: Ubuntu 18.04
    Installation type: source
    Version or commit hash: https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos
    DDS implementation: Fast-RTPS
    Client library (if applicable): N/A

Steps to reproduce issue:

$ mkdir ros2master_ws
$ cd ros2master_ws/
$ env | grep ROS
ROS_VERSION=2
ROS_PYTHON_VERSION=3
ROS_DISTRO=bouncy

$ mkdir src
$ wget https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos
$ vcs import src < ros2.repos
$ colcon build --symlink-install

Error: (Attached full error log for reference)

make[2]: *** [CMakeFiles/rosidl_generator_c_interfaces__rosidl_generator_c.dir/rosidl_generator_c/rosidl_generator_c/msg/dynamic_array_primitives__functions.c.o] Error 1
make[1]: *** [CMakeFiles/rosidl_generator_c_interfaces__rosidl_generator_c.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< rosidl_generator_c	[ Exited with code 2 ]
Aborted  <<< rcutils                                                                                                                
Aborted  <<< fastrtps                                                                                                             
Aborted  <<< rviz_ogre_vendor                                                       
                                         
Summary: 86 packages finished [7min 31s]
  1 package failed: rosidl_generator_c
  3 packages aborted: fastrtps rcutils rviz_ogre_vendor
  2 packages had stderr output: connext_cmake_module rosidl_generator_c
  114 packages not processed

[Ros2master_build_failed_log.txt](https://waffleio-direct-uploads-production.s3.amazonaws.com/uploads/5bcf34650b58a20011d3f933/125516c66e82c728ace21e0d46da9b89718f82ba9becad7ca648f9982d7921722008a62f4f89ec2ffd0e39111d0014f9121c4d4eeefc8d7ef5e06a778e090fbed53e5c75973d11afb18a4db857a83c990dd4e8e8d51199a26867f5819d4e3714041d8dfd5e32a99d43cb58f149f300.txt)

clalancette, 2018-11-20:
```
$ env | grep ROS
ROS_VERSION=2
ROS_PYTHON_VERSION=3
ROS_DISTRO=bouncy
```

That part is odd.  Before building ros2 from scratch, you shouldn't have anything ROS related in the environment.  Can you try again with a "clean" environment, where nothing from ROS is sourced?

mlautman, 2018-12-07:
I came across the same error. I had these lines in my .bashrc for ROS1 development. 
```
    if [ "$UBUNTU_VERSION" = "bionic" ]; then
      export ROS_DISTRO="bouncy"
    elif [ "$UBUNTU_VERSION" = "xenial" ]; then
      export ROS_DISTRO="kinetic"
    fi
    source /opt/ros/$ROS_DISTRO/setup.bash
```

Changing the logic to this fixed the issue:
```
    if [ "$UBUNTU_VERSION" = "bionic" ]; then
      export ROS_DISTRO="bouncy"
    elif [ "$UBUNTU_VERSION" = "xenial" ]; then
      export ROS_DISTRO="kinetic"
      source /opt/ros/$ROS_DISTRO/setup.bash
    fi
```

It seems that the system install was causing the source build to fail. This should probably be documented in the build from source instructions.