ros2/ros2 #605: Can not use client at timer_callback function

IcanChen, 2018-11-14:
## Bug report

**Required Info:**

- Operating System:
  - ubunut 18.04 bouncy
- Installation type:
  - binaries
- DDS implementation: 
  - Fast-RTPS
- Client library (if applicable):
  -  rclcpp

#### Steps to reproduce issue

1. modified examples/minimal_client code as follows:
```
#include "example_interfaces/srv/add_two_ints.hpp"
#include "rclcpp/rclcpp.hpp"
#include <chrono>
#include <cinttypes>
#include <memory>

using namespace std::chrono_literals;
using AddTwoInts = example_interfaces::srv::AddTwoInts;

class Client {
public:
  Client(std::shared_ptr<rclcpp::Node> node) : node_(node) {
    client_ = node_->create_client<AddTwoInts>("add_two_ints");
    timer_ =
        node_->create_wall_timer(1s, std::bind(&Client::timer_callback, this));
  }
  ~Client() {}

private:
  std::shared_ptr<rclcpp::Node> node_ = nullptr;
  rclcpp::TimerBase::SharedPtr timer_ = nullptr;
  ::rclcpp::Client<AddTwoInts>::SharedPtr client_ = nullptr;

  void timer_callback() {
    while (!client_->wait_for_service(std::chrono::seconds(1))) {
      if (!rclcpp::ok()) {
        RCLCPP_ERROR(node_->get_logger(),
                     "client interrupted while waiting for service to appear.")
        return;
      }
      RCLCPP_INFO(node_->get_logger(), "waiting for service to appear...")
    }
    auto request = std::make_shared<AddTwoInts::Request>();
    request->a = 41;
    request->b = 1;
    auto result_future = client_->async_send_request(request);
    if (rclcpp::spin_until_future_complete(node_, result_future) !=
        rclcpp::executor::FutureReturnCode::SUCCESS) {
      RCLCPP_ERROR(node_->get_logger(), "service call failed :(")
      return;
    }
    auto result = result_future.get();
    RCLCPP_INFO(node_->get_logger(),
                "result of %" PRId64 " + %" PRId64 " = %" PRId64, request->a,
                request->b, result->sum);
  }
};
int main(int argc, char *argv[]) {
  rclcpp::init(argc, argv);
  auto node = rclcpp::Node::make_shared("minimal_client");
  Client client(node);
  rclcpp::spin(node);
  rclcpp::shutdown();
  return 0;
}
```
2.
 terminal1 run: ros2 run examples_rclcpp_minimal_service service_main   
 terminal2 run:  ros2 run examples_rclcpp_minimal_client client_main

#### Expected behavior
terminal1: 
```
droid@bouncy:~/Development/security_ws$ ros2 run examples_rclcpp_minimal_service service_main
[INFO] [minimal_service]: request: 41 + 1
[INFO] [minimal_service]: request: 41 + 1
[INFO] [minimal_service]: request: 41 + 1
[INFO] [minimal_service]: request: 41 + 1
```
terminal2:

```
droid@bouncy:~/Development/test$ ros2 run examples_rclcpp_minimal_client  client_main
[INFO] [minimal_client]: result of 41 + 1 = 42
[INFO] [minimal_client]: result of 41 + 1 = 42
[INFO] [minimal_client]: result of 41 + 1 = 42
[INFO] [minimal_client]: result of 41 + 1 = 42
[INFO] [minimal_client]: result of 41 + 1 = 42
```
#### Actual behavior
terminal1: 
```
droid@bouncy:~/Development/security_ws$ ros2 run examples_rclcpp_minimal_service service_main
[INFO] [minimal_service]: request: 41 + 1
```
terminal2:
```
droid@bouncy:~/Development/test$ ros2 run examples_rclcpp_minimal_client  client_main
terminate called after throwing an instance of 'std::runtime_error'
  what():  Node has already been added to an executor. 
```

clalancette, 2018-11-14:
We don't really recommend this style of writing nodes anymore.  If you look at https://github.com/ros2/demos/blob/master/demo_nodes_cpp/src/services/add_two_ints_client_async.cpp, for instance, you'll see that the style there is to have the class derive from `Node`, and then use a callback during the call to `async_send_request`.  That being said, I'm not sure if the style you've written is expected to work, so I'll defer to someone else to comment about that.

IcanChen, 2018-11-15:
Thanks
I fixed it.
```
// Copyright 2014 Open Source Robotics Foundation, Inc.
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at
//
//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

#include <chrono>
#include <cinttypes>
#include <iostream>
#include <memory>
#include <string>

#include "rclcpp/rclcpp.hpp"
#include "rcutils/cmdline_parser.h"

#include "example_interfaces/srv/add_two_ints.hpp"

using namespace std::chrono_literals;

void print_usage() {
  printf("Usage for add_two_ints_client app:\n");
  printf("add_two_ints_client [-s service_name] [-h]\n");
  printf("options:\n");
  printf("-h : Print this help function.\n");
  printf("-s service_name : Specify the service name for this client. Defaults "
         "to add_two_ints.\n");
}

class ClientNode : public rclcpp::Node {
public:
  explicit ClientNode(const std::string &service_name)
      : Node("add_two_ints_client") {
    client_ = create_client<example_interfaces::srv::AddTwoInts>(service_name);
    timer_ =
        create_wall_timer(1s, std::bind(&ClientNode::timer_callback, this));

    // Queue an asynchronous service request that will be sent once `spin` is
    // called on the node.
    // queue_async_request();
  }
  void queue_async_request() {
    while (!client_->wait_for_service(1s)) {
      if (!rclcpp::ok()) {
        RCLCPP_ERROR(this->get_logger(),
                     "Interrupted while waiting for the service. Exiting.")
        return;
      }
      RCLCPP_INFO(this->get_logger(), "service not available, waiting again...")
    }
    auto request =
        std::make_shared<example_interfaces::srv::AddTwoInts::Request>();
    request->a = 2;
    request->b = 3;

    // We give the async_send_request() method a callback that will get executed
    // once the response is received. This way we can return immediately from
    // this method and allow other work to be done by the executor in `spin`
    // while waiting for the response.
    using ServiceResponseFuture =
        rclcpp::Client<example_interfaces::srv::AddTwoInts>::SharedFuture;
    auto response_received_callback = [this](ServiceResponseFuture future) {
      auto result = future.get();
      RCLCPP_INFO(this->get_logger(), "Result of add_two_ints: %" PRId64,
                  result->sum)
      // rclcpp::shutdown();
    };
    auto future_result =
        client_->async_send_request(request, response_received_callback);
  }

private:
  rclcpp::Client<example_interfaces::srv::AddTwoInts>::SharedPtr client_;
  rclcpp::TimerBase::SharedPtr timer_ = nullptr;

  void timer_callback() {
    std::cout << "test" << std::endl;
    queue_async_request();
  }
};

int main(int argc, char **argv) {
  // Force flush of the stdout buffer.
  // This ensures a correct sync of all prints
  // even when executed simultaneously within the launch file.
  setvbuf(stdout, NULL, _IONBF, BUFSIZ);

  if (rcutils_cli_option_exist(argv, argv + argc, "-h")) {
    print_usage();
    return 0;
  }

  rclcpp::init(argc, argv);

  auto service_name = std::string("add_two_ints");
  char *cli_option = rcutils_cli_get_option(argv, argv + argc, "-s");
  if (nullptr != cli_option) {
    service_name = std::string(cli_option);
  }

  auto node = std::make_shared<ClientNode>(service_name);
  rclcpp::spin(node);

  rclcpp::shutdown();
  return 0;
}
```

sloretz, 2018-12-12:
@IcanChen glad you got it working.

The code builds and runs as expected using a `crystal` pre-release after adding `;` to the three `RCLCPP_*(...)` calls. I'll close this issue since you solved it.

closed 2018-12-12 by sloretz

cardboardcode, 2020-11-29:
Hi all, just reposting @IcanChen 's solution. 

Verified to work, albeit from a few minor syntax errors. 

The following copy below includes quick modifications so there are no compilation errors when you copy and paste the code to build.
```cpp
#include <chrono>
#include <cinttypes>
#include <iostream>
#include <memory>
#include <string>

#include "rclcpp/rclcpp.hpp"
#include "rcutils/cmdline_parser.h"

#include "example_interfaces/srv/add_two_ints.hpp"

using namespace std::chrono_literals;

void print_usage() {
  printf("Usage for add_two_ints_client app:\n");
  printf("add_two_ints_client [-s service_name] [-h]\n");
  printf("options:\n");
  printf("-h : Print this help function.\n");
  printf("-s service_name : Specify the service name for this client. Defaults "
         "to add_two_ints.\n");
}

class ClientNode : public rclcpp::Node {
public:
  explicit ClientNode(const std::string &service_name)
      : Node("add_two_ints_client") {
    client_ = create_client<example_interfaces::srv::AddTwoInts>(service_name);
    timer_ =
        create_wall_timer(1s, std::bind(&ClientNode::timer_callback, this));

    // Queue an asynchronous service request that will be sent once `spin` is
    // called on the node.
    // queue_async_request();
  }
  void queue_async_request() {
    while (!client_->wait_for_service(1s)) {
      if (!rclcpp::ok()) {
        RCLCPP_ERROR(this->get_logger(),
                     "Interrupted while waiting for the service. Exiting.");
        return;
      }
      RCLCPP_INFO(this->get_logger(), "service not available, waiting again...");
    }
    auto request =
        std::make_shared<example_interfaces::srv::AddTwoInts::Request>();
    request->a = 2;
    request->b = 3;

    // We give the async_send_request() method a callback that will get executed
    // once the response is received. This way we can return immediately from
    // this method and allow other work to be done by the executor in `spin`
    // while waiting for the response.
    using ServiceResponseFuture =
        rclcpp::Client<example_interfaces::srv::AddTwoInts>::SharedFuture;
    auto response_received_callback = [this](ServiceResponseFuture future) {
      auto result = future.get();
      RCLCPP_INFO(this->get_logger(), "Result of add_two_ints: %" PRId64,
                  result->sum);
    };
    auto future_result =
        client_->async_send_request(request, response_received_callback);
  }

private:
  rclcpp::Client<example_interfaces::srv::AddTwoInts>::SharedPtr client_;
  rclcpp::TimerBase::SharedPtr timer_ = nullptr;

  void timer_callback() {
    std::cout << "test" << std::endl;
    queue_async_request();
  }
};

int main(int argc, char **argv) {
  // Force flush of the stdout buffer.
  // This ensures a correct sync of all prints
  // even when executed simultaneously within the launch file.
  setvbuf(stdout, NULL, _IONBF, BUFSIZ);

  if (rcutils_cli_option_exist(argv, argv + argc, "-h")) {
    print_usage();
    return 0;
  }

  rclcpp::init(argc, argv);

  auto service_name = std::string("add_two_ints");
  char *cli_option = rcutils_cli_get_option(argv, argv + argc, "-s");
  if (nullptr != cli_option) {
    service_name = std::string(cli_option);
  }

  auto node = std::make_shared<ClientNode>(service_name);
  rclcpp::spin(node);

  rclcpp::shutdown();
  return 0;
}
```
CMakeLists.txt
```
cmake_minimum_required(VERSION 3.5)
project(your_ros2_package_name)

# Default to C99
if(NOT CMAKE_C_STANDARD)
  set(CMAKE_C_STANDARD 99)
endif()

# Default to C++14
if(NOT CMAKE_CXX_STANDARD)
  set(CMAKE_CXX_STANDARD 14)
endif()

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
endif()

# find dependencies
find_package(ament_cmake REQUIRED)
find_package(rclcpp REQUIRED)
find_package(example_interfaces REQUIRED)
find_package(rcutils REQUIRED)

add_executable(your_ros2_node_name src/main.cpp)
ament_target_dependencies(your_ros2_node_name
  rclcpp
  example_interfaces
  rcutils
)

install(TARGETS your_ros2_package_name
  DESTINATION lib/${PROJECT_NAME})

ament_package()

```

Verified on **ROS2 Foxy**.

To run the complimentary service node, use the following command:
```bash
ros2 run examples_rclcpp_minimal_service service_main
```