ros2/ros2 #606: Some ROS2 packages are not relocatable

cottsay, 2018-11-14:
## Bug report

**Required Info:**

- Operating System:
  - Linux (probably all)
- Installation type:
  - Fat Archive
- Version or commit hash:
  - Latest Bouncy
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
```
$ grep /home/rosbuild/ci_scripts -IR --exclude=*.{bash,ps1,sh,zsh} ros2-linux
```

#### Expected behavior
Build and packaging infrastructure does not contain paths from the original install space.

#### Actual behavior
Many cmake and pkg-config files contain original paths. When there isn't an install space located at that path, builds referencing the relocated install space fail.

#### Additional information
List of files: (`grep /home/ros -IRl --exclude=*.{bash,ps1,sh,zsh} ros2-linux | sort | uniq`)
```
lib/pkgconfig/console_bridge.pc
lib/pkgconfig/orocos-kdl.pc
lib/pkgconfig/orocos_kdl.pc
lib/pkgconfig/urdfdom_headers.pc
lib/pkgconfig/urdfdom.pc
lib/urdfdom/cmake/urdfdom-config.cmake
lib/urdfdom_headers/cmake/urdfdom_headers-config.cmake
opt/rviz_ogre_vendor/lib/pkgconfig/OGRE-Bites.pc
opt/rviz_ogre_vendor/lib/pkgconfig/OGRE-HLMS.pc
opt/rviz_ogre_vendor/lib/pkgconfig/OGRE-MeshLodGenerator.pc
opt/rviz_ogre_vendor/lib/pkgconfig/OGRE-Overlay.pc
opt/rviz_ogre_vendor/lib/pkgconfig/OGRE-Paging.pc
opt/rviz_ogre_vendor/lib/pkgconfig/OGRE.pc
opt/rviz_ogre_vendor/lib/pkgconfig/OGRE-PCZ.pc
opt/rviz_ogre_vendor/lib/pkgconfig/OGRE-Property.pc
opt/rviz_ogre_vendor/lib/pkgconfig/OGRE-RTShaderSystem.pc
opt/rviz_ogre_vendor/lib/pkgconfig/OGRE-Terrain.pc
opt/rviz_ogre_vendor/lib/pkgconfig/OGRE-Volume.pc
opt/rviz_ogre_vendor/share/OGRE/plugins.cfg
opt/rviz_ogre_vendor/share/OGRE/quakemap.cfg
opt/rviz_ogre_vendor/share/OGRE/resources.cfg
opt/rviz_ogre_vendor/share/OGRE/samples.cfg
opt/rviz_ogre_vendor/share/OGRE/tests.cfg
share/laser_geometry/cmake/laser_geometryExport.cmake
share/resource_retriever/cmake/resource_retrieverExport.cmake
share/rviz_common/cmake/rviz_commonExport.cmake
share/rviz_default_plugins/cmake/rviz_default_pluginsExport.cmake
share/rviz_rendering/cmake/rviz_renderingExport.cmake
share/rviz_visual_testing_framework/cmake/rviz_visual_testing_frameworkExport.cmake
```

Blocks ros-infrastructure/ros_buildfarm#590

dirk-thomas, 2018-11-14:
Afaik `.pc` don't support the case of being relocatable.

wjwwood, 2018-11-14:
The rviz ones are from the exported targets which I believe is generated by CMake. So I’m not sure we’ll have any control over those.

cottsay, 2019-08-14:
Latest list as of Dashing patch 2

CMake:
```
lib/urdfdom/cmake/urdfdom-config.cmake
lib/urdfdom_headers/cmake/urdfdom_headers-config.cmake
share/laser_geometry/cmake/laser_geometryExport.cmake
share/resource_retriever/cmake/resource_retrieverExport.cmake
share/rosbag2_transport/cmake/export_rosbag2_transportExport.cmake
share/rviz_common/cmake/rviz_commonExport.cmake
share/rviz_default_plugins/cmake/rviz_default_pluginsExport.cmake
share/rviz_rendering/cmake/rviz_renderingExport.cmake
share/rviz_visual_testing_framework/cmake/rviz_visual_testing_frameworkExport.cmake
share/test_interface_files/cmake/test_interface_files-extras.cmake
```

Non-CMake:
```
lib/pkgconfig/console_bridge.pc
lib/pkgconfig/orocos-kdl.pc
lib/pkgconfig/orocos_kdl.pc
lib/pkgconfig/urdfdom_headers.pc
lib/pkgconfig/urdfdom.pc
opt/rviz_ogre_vendor/lib/pkgconfig/OGRE-Bites.pc
opt/rviz_ogre_vendor/lib/pkgconfig/OGRE-HLMS.pc
opt/rviz_ogre_vendor/lib/pkgconfig/OGRE-MeshLodGenerator.pc
opt/rviz_ogre_vendor/lib/pkgconfig/OGRE-Overlay.pc
opt/rviz_ogre_vendor/lib/pkgconfig/OGRE-Paging.pc
opt/rviz_ogre_vendor/lib/pkgconfig/OGRE.pc
opt/rviz_ogre_vendor/lib/pkgconfig/OGRE-PCZ.pc
opt/rviz_ogre_vendor/lib/pkgconfig/OGRE-Property.pc
opt/rviz_ogre_vendor/lib/pkgconfig/OGRE-RTShaderSystem.pc
opt/rviz_ogre_vendor/lib/pkgconfig/OGRE-Terrain.pc
opt/rviz_ogre_vendor/lib/pkgconfig/OGRE-Volume.pc
opt/rviz_ogre_vendor/share/OGRE/plugins.cfg
opt/rviz_ogre_vendor/share/OGRE/quakemap.cfg
opt/rviz_ogre_vendor/share/OGRE/resources.cfg
opt/rviz_ogre_vendor/share/OGRE/samples.cfg
opt/rviz_ogre_vendor/share/OGRE/tests.cfg
opt/yaml_cpp_vendor/lib/pkgconfig/yaml-cpp.pc
```

dirk-thomas, 2019-08-14:
> CMake:

`share/test_interface_files/cmake/test_interface_files-extras.cmake` is a good example for a new case being introduced. That one should be addressed right away.

> Non-CMake:

As mentioned above: the `.pc` files are not relevant since they don't support relocation.

traversaro, 2020-04-22:
> > Non-CMake:
> 
> As mentioned above: the `.pc` files are not relevant since they don't support relocation.

I recently discovered that `.pc` file support relocation by using the `${pcfiledir}` variable, see: 
* https://web.archive.org/web/20190927173057/https://wp.colliertech.org/cj/?p=164
* https://bugs.freedesktop.org/show_bug.cgi?id=62018
* https://github.com/google/googletest/pull/1905

dirk-thomas, 2020-06-02:
> I recently discovered that .pc file support relocation by using the ${pcfiledir} variable, see:

Thanks for the reference. I create an example PR using this approach: ros2/urdfdom#20.

For various other packages it would be great if someone would contribute patches to do the same. I create tickets for some of them which are referenced above.

clalancette, 2024-01-12:
We've largely fixed this in the past few years.  There may still be some things that pop up from time-to-time along these lines, but we can just track those on an individual basis.  So I'm going to close this out.

closed 2024-01-12 by clalancette

oysstu, 2024-08-10:
I've opened a PR in urdfdom_headers which applies the same patch as the one dirk-thomas did for urdfdom.
https://github.com/ros/urdfdom_headers/pull/79

sdmcgrath, 2024-11-20:
Opened a PR in console_bridge that applies dirk-thomas' urdfdom patch there as well.
ros/console_bridge#98