ros2/ros2 #607: ROS 2 D-turtle

dirk-thomas, 2018-11-15:
This meta ticket enumerates the tasks which are being worked on for the ROS 2 release scheduled for June 2019. The list will be updated as the tasks move forward:

vmayoral, 2019-02-04:
Hello @dirk-thomas,

Here's a draft of what to expect from @AcutronicRobotics's for Dashing:

- Real-Time
  - [ ] ROS 2 Real-Time Quality Assurance Farm (inspired by https://www.osadl.org/QA-Farm-Realtime.qa-farm-about.0.html) 
    - [ ] **x86**
      - [ ] Intel i7 @ 3.70 GHz
        - [ ] DDS vendor 1
        - [ ] DDS vendor 2
    - [ ] **Arm**
      - [ ] Xilinx Zynq 70x0 SoC @ 666 MHz
        - [ ] DDS vendor 1
        - [ ] DDS vendor 2

- Information model
  - [ ] Abstract sensors_msgs interfaces
  - [ ] Port geometry_msgs interfaces
  - [ ] Port trajectory_msgs interfaces
  - [ ] Port control_msgs interfaces
  - [ ] Port moveit interfaces
  - [ ] quality assurance implementation
  - [ ] basic tooling for usage

- Motion planning
  - [ ] Update/setup infrastructure for development
  - [ ] Convert moveit packages to ROS 2.0
    - [ ] Adapt interfaces to common information model
    - [ ] Convert moveit_core
    - [ ] Convert moveit_kinematics
    - [ ] Convert moveit_planners_ompl
    - [ ] Convert moveit_ros_planning
    - [ ] ...
  - [ ] Provide a first demonstrator of the status of the work with a real robot

gavanderhoorn, 2019-02-04:
@vmayoral wrote:
> - [ ] Convert moveit packages to ROS 2.0 

Hi @vmayoral: is this coordinated / synced with @ros-planning/moveit-maintainers or is this a stand-alone effort?

dirk-thomas, 2019-02-04:
> * Port geometry_msgs interfaces
> * Port trajectory_msgs interfaces
> * Port control_msgs interfaces

All of these packages are already available as released ROS 2 packages.

> * Abstract sensors_msgs interfaces

Can you clarify that "abstract" means in this context? Btw `sensor_msgs` (without a trailing `s`) is also an already released ROS 2 package.

vmayoral, 2019-02-04:
@gavanderhoorn, for now it's a standalone effort. 

We're trying to coordinate with @davetcoleman and [his team](https://discourse.ros.org/t/versioning-roadmap-moveit-1-0-release-plan/7437/2?u=vmayoral). Fund them to ensure we're supported and aligned with what the community expects.

vmayoral, 2019-02-04:
> All of these packages are already available as released ROS 2 packages.

They are all released indeed. We're just abstracting them to XML and in a framework-agnostic manner to facilitate interoperability (even across different robotic frameworks) and by following a Model-driven engineering (MDE) methodology.

 This aligns with the [HRIM project](https://github.com/erlerobot/HRIM) and the objective is to have a common information model (ROS 2 focused mainly but we're also looking at OPC UA models and try to find "common ground" while respecting ROS 2 interfaces). Eventually, we're hoping that HRIM would allow to "generate" the communication artifacts for ROS 2 interactions by reading the dependencies across packages in a ROS 2 workspace.

Some preliminar documentation is available at https://acutronicrobotics.com/docs/technology/hrim. This is by no means finished, it's an ongoing effort.

vmayoral, 2019-02-12:
An update from our side:

After having received community input regarding the information model (mostly related to the HRIM project), we've decided to take a step back in our plans and focus solely on hardware-related interfaces. We were being overly ambitious with HRIM (hoping it would govern all communications across ROS 2 nodes interacting and facilitating interoperability with other frameworks, e.g. OPC UA). 

We've decided to stick to purely hardware-related interfaces (which was the origin of the project) for what concerns HRIM.

_I've updated our plans above accordingly. More updates may happen based on the input received and our progress._

davetcoleman, 2019-02-20:
@dirk-thomas please add to your roadmap, per @vmayoral, basic port of moveit to ros 2.0 for Dashing!

Also, the sub-milestones have been moved to https://github.com/ros-planning/moveit2

dirk-thomas, 2019-02-20:
@davetcoleman This ticket represents the roadmap for Dashing. Do you want to edit you comment to change the port of MoveIt! into an unchecked bullet and remove it from Victors comment above?

dejanpan, 2019-02-21:
@dirk-thomas regarding this point: "Launch improvements for launch-based testing ([ros2/launch#161](https://github.com/ros2/launch/issues/161))":
1. @wjwwood @hidmic @pbaughman decided that https://github.com/ros2/launch/issues/161 will get consolidated with the https://github.com/ApexAI/apex_rostest framework. The description of that project is available here: https://github.com/ApexAI/apex_rostest/blob/99e5de6e90942f28ec9ece7292c546593712d774/README.md.
2. This would be Apex.AI feature contribution for ROS 2 Dashing Diademata. Apex.AI will maintain it and keep releasing it.
3. After these PRs will be merged https://github.com/ApexAI/apex_rostest/pulls, the framework will be largely feature complete. The only thing left TBD will be to shift the things around.

dejanpan, 2019-02-21:
@vmayoral could you briefly comment which RTOSs you plan to support in https://github.com/ros2/ros2/issues/607#issuecomment-460319513 "ROS 2 Real-Time Quality Assurance Farm"?

@anup-pem FYI

dejanpan, 2019-02-21:
@dirk-thomas another thing: I would like that this `a test` implemented with this tool: https://github.com/ApexAI/performance_test would run nightly or weekly or monthly on OSRF farm. The tool is complete, Apex can help integrating it.

rutvih, 2019-02-21:
@dejanpan +1 on running these tests nightly or weekly on OSRF build farm. Our team has been trying to enhance tool created by Apex and we are planning to contribute our changes back to that repo as well.

@dirk-thomas These are the changes Amazon is working on.
- Support lifespan, liveliness & deadline QoS policies (add support in both ROS2 & FastRTPS).
- ROS2 quality testing.
  - [ ] Enable Asan, Tsan nightly jobs to ROS2 build-farm and produce reports.
  - [ ] Add thread safety annotation macro (clang only).
  - [ ] Enable symbol visibility for all RMW & RCL packages.
  - [ ] Add Linux Clang Nightly Job to build-farm
-  ROS2 performance benchmarking.
   - [ ] Enhance Apex's performance test suit.
   - [ ] Add more metrics, generate report.
-  Robot security threat model. 
- Enhance ROS2 contribution doc with our learnings.

JaimeMartin, 2019-02-22:
@rutvih here at eprosima we are working already on lifespan and deadline actively to support your efforts in ROS2 layer. We need to sync to understand your plan to expose those QoS in ROS2, and align our design.

rutvih, 2019-02-22:
@JaimeMartin That's great to hear! We have been looking in to work needed on FastRTPS and would be great to see if we can collaborate on this going forward.  My team has created https://github.com/eProsima/Fast-RTPS/issues/429 to follow up.

mkhansen-intel, 2019-02-25:
@dirk-thomas - Regarding this line item: 
* CI for simulation-based testing of navigation

Is there an issue for that? We have some simulation testing running in our CI already, and may be able to help on this. Currently those tests aren't running in the build farm, but we could help enable them.

vmayoral, 2019-02-26:
> @vmayoral could you briefly comment which RTOSs you plan to support in [#607 (comment)](https://github.com/ros2/ros2/issues/607#issuecomment-460319513) "ROS 2 Real-Time Quality Assurance Farm"?

@dejanpan we'll be using Linux with the PREEMPT_RT patches in a similar fashion to what OSADL has been doing. Probably re-using as much of their work as possible. As discussed, target is to have hardware benchmarks in a somewhat common platform supported across different devices. Let me know if you have further input.

dirk-thomas, 2019-02-26:
> CI for simulation-based testing of navigation
> Is there an issue for that?

Not that I know of. The current work is focus on enabling GPU-based test on the CI infrastructure. Before anyone starts working on some navigation tests I will make sure they reach out to you / here.

mkhansen-intel, 2019-02-26:
@dirk-thomas - thanks, sounds like a plan

gbiggs, 2019-02-27:
> @dejanpan we'll be using Linux with the PREEMPT_RT patches in a similar fashion to what OSADL has been doing. Probably re-using as much of their work as possible. As discussed, target is to have [...]

What hardware platform(s) are you considering?

thomas-moulard, 2019-02-27:
>   * [ ]  Enable Asan, Tsan nightly jobs to ROS2 build-farm and produce reports.
>   * [ ]  Add thread safety annotation macro (clang only) - https://discourse.ros.org/t/adding-clang-thread-safety-analysis-for-ros2-core-packages/7930
>   * [ ]  Enable symbol visibility for all RMW & RCL packages.
>   * [x]  Add Linux Clang Nightly Job to build-farm - https://discourse.ros.org/t/ci-ros2-org-clang-nightly-job-now-available/7857

@rutvih Could you link those to https://github.com/ros2/ros2/issues/662 ?

rutvih, 2019-02-27:
> @rutvih Could you link those to #662 ?

done!

vmayoral, 2019-02-27:
> What hardware platform(s) are you considering?

@gbiggs we'll initially be focused on:
-  Intel i7 @ 3.70 GHz and
- Xilinx Zynq 70x0 SoC @ 666 MHz

Ideally we should be able to open up the possibility of having interested parties to contribute with hardware to the QA farm so that it can get added to the results produced.  No big promises beyond these two platforms for now but feel free to send input and we'll try our best.

mkhansen-intel, 2019-03-05:
@dirk-thomas - regarding these items
> 
 Improve memory management
 Improve performance and reliability
>
Are there open issues for these yet? We would like to potentially contribute to these also.

bshin-lge, 2019-03-05:
Here is what we at LG Electronics are working on and will add for Dashing:
OpenEmbedded support: automatically generated Bitbake recipes of ROS2 index packages

dirk-thomas, 2019-03-08:
> regarding these items
> Are there open issues for these yet?

> Improve memory management

@mkhansen-intel Please see the updated description with a reference to another ticket.

> Improve performance and reliability

There is no ticket tracking this effort yet since it hasn't been started yet.

Karsten1987, 2019-03-18:
Bosch has committed to migrate the diagnostic packages to ROS2. Meta ticket for tracking progress: https://github.com/boschresearch/ros2_diagnostics/issues/1

vmayoral, 2019-03-19:
@AcutronicRobotics has just announced two open source releases meant to facilitate the interface of machine learning algorithms with ROS 2:
- [gym-gazebo2](https://github.com/AcutronicRobotics/gym-gazebo2) ([short announcement](https://acutronicrobotics.com/news/acutronic-robotics-launches-gym-gazebo2-a-toolkit-for-reinforcement-learning-using-ros-2-and-gazebo/), [paper](https://arxiv.org/pdf/1903.06278.pdf))
- [ros2learn](https://github.com/AcutronicRobotics/ros2learn) ([short announcement](https://acutronicrobotics.com/news/acutronic-robotics-presents-ros2learn-a-reinforcement-learning-framework-for-ros-2/), [article](https://arxiv.org/pdf/1903.06282.pdf))

I'm not sure if this qualifies as contributions towards Dashing so I'll bring up this topic in the next meeting and request input about what kind of additional features might be interesting for other parties.

thomas-moulard, 2019-03-29:
@rutvih We now have a ticket for QoS we can add to the roadmap for D: https://github.com/ros2/ros2/issues/680

rutvih, 2019-03-29:
> @rutvih We now have a ticket for QoS we can add to the roadmap for D: #680

Done!

vmayoral, 2019-04-18:
Updated @AcutronicRobotics  progress for Dashing.

iluetkeb, 2019-04-29:
What's the relationship of this ticket with https://index.ros.org/doc/ros2/Releases/Release-Dashing-Diademata/ ?
It seems it might be useful to either a) collate this discussion and put it on the page above, or b) link to this issue from that page.

robotpilot, 2019-04-29:
@dirk-thomas, the following are some of the things that ROBOTIS wants to contribute to ROS 2 Dashing version.

- Fetch cartographer to release-1.0 tag (https://github.com/ros2/cartographer/pull/7)
- Fetch cartographer_ros to release-1.0 tag (https://github.com/ros2/cartographer_ros/pull/25)
- Migrate the turtlesim package to ROS2

dirk-thomas, 2019-04-29:
@iluetkeb

> What's the relationship of this ticket with https://index.ros.org/doc/ros2/Releases/Release-Dashing-Diademata/ ?

The reference page is the Dashing release page which will contain links to install instructions, information about included features as well as migration steps.

> collate this discussion and put it on the page above

On this ticket every contributor can comment and describe what they are working on and planning to contribute. The referenced page requires PRs to change and should only contain what has actually landed for Dashing which will likely only be a subset of the items mentioned in this thread.

> link to this issue from that page.

That is already the case: https://index.ros.org/doc/ros2/Releases/Release-Dashing-Diademata/#new-features-in-this-ros-2-release

    > During the development the *Dashing meta ticket* on GitHub contains an up-to-date state of the ongoing high level tasks as well as references specific tickets with more details.

jhdcs, 2019-05-02:
@dirk-thomas I just wanted to mention that there has been a slight change as to how `NodeGraph::get_node_names()` works - it now returns qualified names of all nodes, rather than just the bare names without namespaces. See ros2/rclcpp#698

o-olalde, 2019-05-20:
From Alias Robotics’ offensive security perspective we are ready to contribute with:
	

- ROS 2 [Threat Model](http://design.ros2.org/articles/ros2_threat_model.html#threat-analysis-for-the-mara-robotic-platform), created in collaboration with @AcutronicRobotics to test the security of a ROS2 industrial collaborative robot

    - [x] Mara Threat model draft

    - [x] Merged version 

    - [x] Threat model for sharing

- ROS 2 Static Code analysis funded by @rosin-project

    - [x] Test for ROS-Industrial project

    - [ ] Result release

	

- [Aztarna](https://github.com/aliasrobotics/aztarna)

    - [x] [Footprinting tool](https://arxiv.org/pdf/1812.09490.pdf) for ROS/ROS 2

    - [ ] Module for ROS-Industrial project (vendor specific drivers) funded by @rosin-project

	

- Fuzzing hacking tool for ROS 2

    - [x] Flexible Fuzzer for ROS 2 (topic, protocol, etc)

    - [ ] Release as part of the @rosin-project

dirk-thomas, 2019-05-31:
:boat: https://discourse.ros.org/t/ros-2-dashing-diademata-released/9365

closed 2019-05-31 by dirk-thomas

parambeernegi, 2021-05-27:
[state_publisher-2]   File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
[state_publisher-2]   File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
[state_publisher-2]   File "<frozen importlib._bootstrap_external>", line 678, in exec_module
[state_publisher-2]   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
[state_publisher-2]   File "/home/$$$$$$/dev_ws/build/urdf_tutorial/urdf_tutorial/state_publisher.py", line 10, in <module>
[state_publisher-2]     from tf2_ros import TransformBroadcaster, TransformStamped
[state_publisher-2]   File "/opt/ros/melodic/lib/python2.7/dist-packages/tf2_ros/__init__.py", line 38, in <module>
[state_publisher-2]     from tf2_py import *
[state_publisher-2]   File "/opt/ros/melodic/lib/python2.7/dist-packages/tf2_py/__init__.py", line 38, in <module>
[state_publisher-2]     from ._tf2 import *
[state_publisher-2] ImportError: /opt/ros/melodic/lib/python2.7/dist-packages/tf2_py/_tf2.so: undefined symbol: _ZNK3tf210BufferCore15_allFramesAsDotB5cxx11Ed
[ERROR] [state_publisher-2]: process has died [pid 29217, exit code 1, cmd '/home/$$$$$$$$$/dev_ws/install/urdf_tutorial/lib/urdf_tutorial/state_publisher __node:=state_publisher'].

Facing issue while working on urdf tutorial ros dashing
can someone tell the solution of this

jhdcs, 2021-05-27:
So, this GitHub issue has been closed for some time. Your best bet for getting your questions answered is to try asking on ROS Answers (https://answers.ros.org/questions/).