ros2/ros2 #608: Errors while building packages on MacOS

abylikhsanov, 2018-11-15:
## Bug report

**Required Info:**

- Operating System:
  - macOS High Sierra 10.13.6
- Installation type:
  - colcon-tutorial (https://index.ros.org/doc/ros2/Colcon-Tutorial/)
- Version or commit hash:
  - ROS 2 Bouncy Bolson - 2018-08-24
- DDS implementation:
  - Standard ROS2
>

#### Steps to reproduce issue

Just followed the standard installation here [LINK]( https://index.ros.org/doc/ros2/OSX-Install-Binary/)
After that, I have sourced the setup.zsh and listener + talker worked well. Then, I have followed this tutorial on colcon:
[LINK-COLCON](https://index.ros.org/doc/ros2/Colcon-Tutorial/), followed all the instructions step by step and when running this command from the tutorial:
```
colcon build --symlink-install
```
I have waited for 20 minutes and at the end, I got an error:

``` Summary: 90 packages finished [3min 32s]
  1 package failed: rviz_rendering
  2 packages aborted: rmw_connext_shared_cpp rosidl_generator_cpp
  29 packages had stderr output: ament_cmake ament_cmake_core ament_cmake_export_definitions ament_cmake_export_dependencies ament_cmake_export_include_directories ament_cmake_export_interfaces ament_cmake_export_libraries ament_cmake_export_link_flags ament_cmake_gmock ament_cmake_gtest ament_cmake_include_directories ament_cmake_libraries ament_cmake_nose ament_cmake_pytest ament_cmake_python ament_cmake_target_dependencies ament_cmake_test ament_lint_auto ament_lint_common connext_cmake_module gmock_vendor gtest_vendor orocos_kdl rmw_connext_shared_cpp ros_environment rttest rviz_rendering tinyxml2_vendor tlsf
  96 packages not processed
The file /Users/ikhsanov/ros2_ws/None does not exist ```

#### Expected behavior
The compilation of the files using colcon build.
#### Actual behavior
Got an error described above
#### Additional information

Here are the other detailed errors during the compilation (I wonder why qt5 was not found even though 
I have installed qt5 using brew and 

``` brew --prefix qt5 ``` 

shows me:

``` /usr/local/opt/qt``` 

Anyway, other errors above the summary:

``` 
By not providing "Findrticonnextmsgcpp.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "rticonnextmsgcpp", but CMake did not find one.

  Could not find a package configuration file provided by "rticonnextmsgcpp"
  with any of the following names:

    rticonnextmsgcppConfig.cmake
    rticonnextmsgcpp-config.cmake

  Add the installation prefix of "rticonnextmsgcpp" to CMAKE_PREFIX_PATH or
  set "rticonnextmsgcpp_DIR" to a directory containing one of the above
  files.  If "rticonnextmsgcpp" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:16 (find_package)

By not providing "Findnddscpp.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "nddscpp", but
  CMake did not find one.

  Could not find a package configuration file provided by "nddscpp" with any
  of the following names:

    nddscppConfig.cmake
    nddscpp-config.cmake

  Add the installation prefix of "nddscpp" to CMAKE_PREFIX_PATH or set
  "nddscpp_DIR" to a directory containing one of the above files.  If
  "nddscpp" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:16 (find_package)

CMake Error at CMakeLists.txt:27 (find_package):
  By not providing "FindQt5.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Qt5", but
  CMake did not find one.

  Could not find a package configuration file provided by "Qt5" with any of
  the following names:

    Qt5Config.cmake
    qt5-config.cmake

  Add the installation prefix of "Qt5" to CMAKE_PREFIX_PATH or set "Qt5_DIR"
  to a directory containing one of the above files.  If "Qt5" provides a
  separate development package or SDK, be sure it has been installed.
``` 

My stupid assumption is due to the fact that I did not included the CMAKE_PATH which is also shown as warning (as above errors seems to be CMAKE related):
``` 
CMake Warning:
  Manually-specified variables were not used by the project:

    CMAKE_PREFIX_PATH

```

wjwwood, 2018-11-15:
Looks like you're building from source, but following the "binary" installation instructions?

This tutorial (the source install tutorial for macOS) instructs you on how to extend your `CMAKE_PREFIX_PATH` to include the Qt installed by Homebrew:

https://index.ros.org/doc/ros2/OSX-Development-Setup/#install-prerequisites

abylikhsanov, 2018-11-15:
Yes, I just realized... It was a bit confusing

wjwwood, 2018-11-15:
No, I'm not sure where `test-pkg` is coming from. Is that something you made?

Either way, the right thing to do at this point (and honestly initially) is to ask new questions on https://answers.ros.org.

closed 2018-11-15 by wjwwood

deniscalin, 2024-10-26:
> Looks like you're building from source, but following the "binary" installation instructions?
>
> This tutorial (the source install tutorial for macOS) instructs you on **how to extend your [...]

@wjwwood Thank you for the pointer! That link is no longer accessible, but the concept helped me fix this issue on Mac M2 Sonoma 14.5.

For anyone facing errors with homebrew-installed Qt5 (and potentially others) when running `colcon build` to build your ROS2 packages, here is what I did to fix:

1. Run `brew --prefix qt@5` to find your Qt5 installation directory. Save that full path for later.
2. Find and open the CMakeLists.txt file of the package you are installing. For turtlesim, it was in the `src/ros_tutorials/turtlesim` path relative to my workspace directory.
3. Add the following line to the CMakeLists.txt file, and save:
`set(CMAKE_PREFIX_PATH "{your_qt5_prefix_from_step_1}/lib/cmake")`

Essentially you want to add the full path to the package's cmake directory to the CMAKE_PREFIX_PATH. For example, to resolve the error with Qt5, I added the following line to the CMakeLists.txt file and saved:
`set(CMAKE_PREFIX_PATH "/opt/homebrew/opt/qt@5/lib/cmake")`

And that's it! After this my package built successfully. This could potentially work for other CMakeLists.txt-related errors.