ros2/ros2 #610: Can't run python files

abylikhsanov, 2018-11-16:
## Bug report

**Required Info:**

- Operating System:
  - macOS High Sierra 10.13.6
- Installation type:
  - colcon-tutorial (https://index.ros.org/doc/ros2/Colcon-Tutorial/)
- Version or commit hash:
  - ROS 2 Bouncy Bolson - 2018-08-24
- DDS implementation:
  - Standard ROS2
>

#### Steps to reproduce issue

I have installed ROS 2 for macOS and sourced the setup.zsh (I am using zshell). All works great and I have decided to create my custom python package to see how everything works. I have downloaded the examples files from here:

```
https://github.com/ros2/examples
```
I have copied package.xml and setup.py from 
```
https://github.com/ros2/examples/tree/master/rclpy/topics/minimal_publisher
```

and changed the package names and other necessary fields. My package name is python_pkg_test and my package.xml looks like this:

```
<?xml version="1.0"?>
<?xml-model href="http://download.ros.org/schema/package_format2.xsd" schematypens="http://www.w3.org/2001/XMLSchema"?>
<package format="2">
  <name>python_pkg_test</name>
  <version>0.0.1</version>
  <description>Examples of minimal publishers using rclpy.</description>

  <maintainer email="mikael@osrfoundation.org">Mikael Arguedas</maintainer>
  <license>Apache License 2.0</license>

  <exec_depend>rclpy</exec_depend>
  <exec_depend>std_msgs</exec_depend>

  <!-- These test dependencies are optional
  Their purpose is to make sure that the code passes the linters -->
  <test_depend>ament_copyright</test_depend>
  <test_depend>ament_flake8</test_depend>
  <test_depend>ament_pep257</test_depend>
  <test_depend>python3-pytest</test_depend>

  <export>
    <build_type>ament_python</build_type>
  </export>
</package>

```

And my setup.py looks like this:

```
from setuptools import setup

package_name = 'python_pkg_test'
```
setup(
    name=package_name,
    version='0.0.0',
    packages=[],
    py_modules=[],
    install_requires=['setuptools'],
    zip_safe=True,
    author='Mikael Arguedas',
    author_email='mikael@osrfoundation.org',
    maintainer='Mikael Arguedas',
    maintainer_email='mikael@osrfoundation.org',
    keywords=['ROS'],
    classifiers=[
        'Intended Audience :: Developers',
        'License :: OSI Approved :: Apache Software License',
        'Programming Language :: Python',
        'Topic :: Software Development',
    ],
    description='Examples of minimal publishers using rclpy.',
    license='Apache License, Version 2.0',
    tests_require=[],
    entry_points={
        'console_scripts': [
            'talker = talker:main',
        ],
    },
)
```

```

from setuptools import setup

package_name = 'python_pkg_test'

setup(
    name=package_name,
    version='0.0.0',
    packages=[],
    py_modules=[],
    install_requires=['setuptools'],
    zip_safe=True,
    author='Mikael Arguedas',
    author_email='mikael@osrfoundation.org',
    maintainer='Mikael Arguedas',
    maintainer_email='mikael@osrfoundation.org',
    keywords=['ROS'],
    classifiers=[
        'Intended Audience :: Developers',
        'License :: OSI Approved :: Apache Software License',
        'Programming Language :: Python',
        'Topic :: Software Development',
    ],
    description='Examples of minimal publishers using rclpy.',
    license='Apache License, Version 2.0',
    tests_require=[],
    entry_points={
        'console_scripts': [
            'talker = talker:main',
        ],
    },
)

```
After running  ``` colcon build --symlink-install ``` I went to my exectuable file in ``` workspace/install/python_pkg_test/bin/talker ``` and tried to execute ``` python3 talker ```

After which I got this error:

```
Traceback (most recent call last):
  File "talker", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/usr/local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 3126, in <module>
    @_call_aside
  File "/usr/local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 3110, in _call_aside
    f(*args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 3139, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 581, in _build_master
    ws.require(__requires__)
  File "/usr/local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 898, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 784, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'python-pkg-test' distribution was not found and is required by the application
```
Other example executables work fine but whenever I create my own rclpy package with some similar executable I always get the error above. I did not find any similar problems on a web and could not find a guide on how to create rclpy packages, eventhough there are some third party tutorials but they are similar to what I did.

sloretz, 2018-11-16:
In another terminal where the `install` folder of the workspace has sourced, what's the output of

```ros2 run python_pkg_test talker```
?

abylikhsanov, 2018-11-16:
This is the output:
`Package 'python_pkg_test' not found`

abylikhsanov, 2018-11-16:
Yes, I did just now and the error persists

dirk-thomas, 2018-11-29:
Since it works for the example packages it is very likely that their is either a copy-n-paste error or you are missing a step. Please consider sharing a complete package (e.g. via a GitHub repo) which is ready to try as well as describe the exact sequence of commands you run.

dirk-thomas, 2019-01-11:
I will close this ticket for now due to no further response. Please feel free to comment with the requested information and the ticket can be reopened.

closed 2019-01-11 by dirk-thomas

SouLeo, 2019-09-06:
Hi, I have followed the steps provided for @abylikhsanov  and gotten the same error outputs. I was wondering if you could help me track down the error in this package: https://github.com/SouLeo/ROS2_AllenNLP_Parser/tree/master/temoto_parser

I'm running Ubuntu 18.04 (bionic) with ROS2 Bouncy

I've been trying to follow this tutorial for creating workspaces:
https://index.ros.org//doc/ros2/Tutorials/Colcon-Tutorial/

and this tutorial for creating python packages:
https://index.ros.org//doc/ros2/Tutorials/Colcon-Tutorial/

Any help would be appreciated, as I have rebuilt my workspace and package multiple times to try and get this right.

clalancette, 2019-09-06:
> I'm running Ubuntu 18.04 (bionic) with ROS2 Bouncy

Bouncy is out of support now and we've made quite a lot of changes since then.  I'll suggest that you try the newest release instead, [ROS 2 Dashing](https://index.ros.org/doc/ros2/Installation/Dashing/).  If you are still having problems, please open a question at https://answers.ros.org, where you may be able to get help.

3wnbr1, 2019-12-17:
Encountered the same issue on macOS Mojave with eloquent, upgrading colcon with pip solved the issue :

```pip3 install --upgrade colcon-common-extensions --upgrade-strategy=eager```