ros2/ros2 #611: Failed to build rosidl_generator_c with latest ros2 master branch

Rohita83, 2018-11-17:
***   Failed to build rosidl_generator_c with latest ros2 master branch  ***

We have tried below build steps on multiple system.

**Required Info:**

    Operating System: Ubuntu 18.04
    Installation type: source
    Version or commit hash: https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos
    DDS implementation: Fast-RTPS
    Client library (if applicable): N/A

**Steps to reproduce issue:**

cd ~/ros2master_ws
sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
sudo apt-key adv --keyserver hkp://pool.sks-keyservers.net --recv-key 0xB01FA116
sudo apt-get update
sudo apt-get install python-vcstool

sudo apt-get update
sudo apt-get install git wget
sudo apt-get install build-essential cppcheck cmake libopencv-dev libpoco-dev libpocofoundation9v5 libpocofoundation9v5-dbg python-empy python3-dev python3-empy python3-nose python3-pip python3-setuptools python3-vcstool

*dependencies for testing*
sudo apt-get install clang-format pydocstyle pyflakes python3-coverage python3-mock python3-pep8 uncrustify

*dependencies for fastrtps*
sudo apt-get install libboost-chrono-dev libboost-date-time-dev libboost-program-options-dev libboost-regex-dev libboost-system-dev libboost-thread-dev
sudo apt-get install libboost-all-dev libpcre3-dev zlib1g-dev python-empy python-pkg-resources

wget https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos
vcs import ~/ros2master_ws/src < ros2.repos

cd src/ament/
git clone git@github.com:ament/ament_tools.git
cd ~/ros2master_ws

src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install

**Error:** (Attached error log **Ros2MasterBuildFailed.txt**)
```
In file included from /home/user/ros2master_ws/build/rosidl_generator_c/rosidl_generator_c/rosidl_generator_c/msg/dynamic_array_primitives__functions.c:27:0:
/home/user/ros2master_ws/src/ros2/rosidl/rosidl_generator_c/include/rosidl_generator_c/string_functions.h:60:1: note: expected ‘rosidl_generator_c__String__Sequence * {aka struct rosidl_generator_c__String__Sequence *}’ but argument is of type ‘int *’
 rosidl_generator_c__String__Sequence__fini(
 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMakeFiles/rosidl_generator_c_interfaces__rosidl_generator_c.dir/build.make:782: recipe for target 'CMakeFiles/rosidl_generator_c_interfaces__rosidl_generator_c.dir/rosidl_generator_c/rosidl_generator_c/msg/dynamic_array_primitives__functions.c.o' failed
make[2]: *** [CMakeFiles/rosidl_generator_c_interfaces__rosidl_generator_c.dir/rosidl_generator_c/rosidl_generator_c/msg/dynamic_array_primitives__functions.c.o] Error 1
CMakeFiles/Makefile2:1145: recipe for target 'CMakeFiles/rosidl_generator_c_interfaces__rosidl_generator_c.dir/all' failed
make[1]: *** [CMakeFiles/rosidl_generator_c_interfaces__rosidl_generator_c.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

<== Command '. /home/user/ros2master_ws/build/rosidl_generator_c/cmake__build.sh && /usr/bin/make -j8 -l8' failed in '/home/user/ros2master_ws/build/rosidl_generator_c' with exit code '2'
<== Command '. /home/user/ros2master_ws/build/rosidl_generator_c/cmake__build.sh && /usr/bin/make -j8 -l8' failed in '/home/user/ros2master_ws/build/rosidl_generator_c' with exit code '2'
[Ros2MasterBuildFailed.txt](https://github.com/ros2/ros2/files/2592225/Ros2MasterBuildFailed.txt)

```

Rohita83, 2018-11-17:
[Ros2MasterBuildFailed.txt](https://github.com/ros2/ros2/files/2592229/Ros2MasterBuildFailed.txt)

Attached error log.

gavanderhoorn, 2018-11-18:
The error you show is not the actual error. That is printed all the way at the start almost of the log:

```
[ 29%] Building C object CMakeFiles/rosidl_generator_c_interfaces__rosidl_generator_c.dir/rosidl_generator_c/rosidl_generator_c/msg/dynamic_array_primitives__functions.c.o
In file included from /home/user/ros2master_ws/build/rosidl_generator_c/rosidl_generator_c/rosidl_generator_c/msg/bounded_array_primitives__functions.h:18:0,
                 from /home/user/ros2master_ws/build/rosidl_generator_c/rosidl_generator_c/rosidl_generator_c/msg/bounded_array_primitives__functions.c:4:
/home/user/ros2master_ws/build/rosidl_generator_c/rosidl_generator_c/rosidl_generator_c/msg/bounded_array_primitives__struct.h:109:3: error: unknown type name ‘rosidl_generator_c__bool__Array’
   rosidl_generator_c__bool__Array bool_values;
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/user/ros2master_ws/build/rosidl_generator_c/rosidl_generator_c/rosidl_generator_c/msg/bounded_array_primitives__struct.h:110:3: error: unknown type name ‘rosidl_generator_c__byte__Array’
   rosidl_generator_c__byte__Array byte_values;
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
...
```

For C/C++, always scroll all the way back to where the first error is printed. Subsequent error messages printed by the compiler are uninformative, as they are essentially 'caused' by the first error (the error you posted in #604 was also not the actual error).

gavanderhoorn, 2018-11-18:
My point was not that you don't provide enough information, but that *if* you quote an error message directly in your post(s), you might as well show the *actual* error message.

clalancette, 2018-11-20:
I'm going to close this as a dup of #604 for now.  We can continue iterating there, and we'll reopen that one if there is a problem that needs to be addressed.

closed 2018-11-20 by clalancette