ros2/ros2 #614: Can we add 'angles' back into ros2.repos since navigation2 has a dependency on it?

mkhansen-intel, 2018-11-26:
## Feature request
navigation2 has a dependency on the 'angles' package which is part of the distro but not included in the ros2.repos file (recently removed). Because of this we have to add the angles package and build from source separately in an overlay workspace. It seems to me that if it is part of the distro, it should be part of the ros2.repos file since the 'build from source' directions point to using that file.  Can we get this re-added?

SteveMacenski, 2018-11-26:
Its a rosdep key now (https://github.com/ros2/rosdistro/blob/ros2/crystal/distribution.yaml#L132) it doesn't need to be in the repos file

mkhansen-intel, 2018-11-26:
Ah, OK I didn't think about that. I'll close this ticket then.

closed 2018-11-26 by mkhansen-intel