ros2/ros2 #621: Error output from talker/listener with connext/fastrtps

wjwwood, 2018-12-13:
## Bug report

**Required Info:**

- Operating System:
  - macOS High Sierra 10.13.6
- Installation type:
  - source
- Version or commit hash:
  - master
- DDS implementation:
  - fastrtps and connext (both static)
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

In terminal A:

```
$ source ./install/setup.bash
$ RMW_IMPLEMENTATION=rmw_connext_cpp ./install/demo_nodes_cpp/lib/demo_nodes_cpp/talker
```

In terminal B:

```
$ source ./install/setup.bash
$ RMW_IMPLEMENTATION=rmw_fastrtps_cpp ./install/demo_nodes_cpp/lib/demo_nodes_cpp/listener
```

Terminate the listener (ctrl-c in terminal B).

#### Expected behavior

No error output.

#### Actual behavior

Talker and listener communicate, but there is error output on the talker side in these cases:

- if the talker is run before the listener and the listener is terminated first:

```
[INFO] [talker]: Publishing: 'Hello World: 1'
[INFO] [talker]: Publishing: 'Hello World: 2'
[INFO] [talker]: Publishing: 'Hello World: 3'
[INFO] [talker]: Publishing: 'Hello World: 4'
[INFO] [talker]: Publishing: 'Hello World: 5'
[INFO] [talker]: Publishing: 'Hello World: 6'
[INFO] [talker]: Publishing: 'Hello World: 7'
PRESPsService_removeMatchingRemoteEndpoints:!precondition: me == ((void*)0) || worker == ((void*)0) || remoteGuid == ((void*)0) || !( ((remoteGuid->prefix.hostId == (0x00000000)) && (remoteGuid->prefix.appId == (0x00000000)) && (remoteGuid->prefix.instanceId == (0x00000000)) && (remoteGuid->objectId != (0x00000000))) || ((remoteGuid->prefix.hostId != (0x00000000)) && (remoteGuid->prefix.appId != (0x00000000)) && (remoteGuid->prefix.instanceId != (0x00000000)) && (remoteGuid->objectId == (0x00000000))) )
DISCParticipantDiscoveryPlugin_removeRemoteParticipantI:!remove remote endpoints of remote participant: 0X10F3702,0X85910000,0,0X1C1
PRESInterParticipant_removeRemoteParticipant:[Local Participant: ad33702 9184 1] [Remote Participant: 1c1 10f3702 85910000 | Remote Endpoint: 0] Could not remove remote endpoint
PRESInterParticipant_removeRemoteParticipant:[Local Participant: ad33702 9184 1] [Remote Participant: 1c1 10f3702 85910000 | Remote Endpoint: 0] Could not remove remote endpoint
PRESParticipant_removeRemoteParticipantFromBuiltinChannels:Failed to remove remote participant 10f3702 85910000 0 to inter participant channel
PRESParticipant_removeRemoteParticipant:!remove remote participant from builtin channels
DISCParticipantDiscoveryPlugin_removeRemoteParticipantI:!remove remote participant: 0X10F3702,0X85910000,0,0X1C1
DISCParticipantDiscoveryPlugin_removeRemoteParticipant:!remove remote participant: 0X10F3702,0X85910000,0,0X1C1
DISCSimpleParticipantDiscoveryPluginReaderListener_onDataAvailable:!remove remote participant
[INFO] [talker]: Publishing: 'Hello World: 8'
[INFO] [talker]: Publishing: 'Hello World: 9'
^C[INFO] [rclcpp]: signal_handler(signal_value=2)
```

- if the talker is run first, and the talker is terminated first:

```
[INFO] [talker]: Publishing: 'Hello World: 1'
[INFO] [talker]: Publishing: 'Hello World: 2'
[INFO] [talker]: Publishing: 'Hello World: 3'
[INFO] [talker]: Publishing: 'Hello World: 4'
[INFO] [talker]: Publishing: 'Hello World: 5'
[INFO] [talker]: Publishing: 'Hello World: 6'
^C[INFO] [rclcpp]: signal_handler(signal_value=2)
PRESPsService_removeMatchingRemoteEndpoints:!precondition: me == ((void*)0) || worker == ((void*)0) || remoteGuid == ((void*)0) || !( ((remoteGuid->prefix.hostId == (0x00000000)) && (remoteGuid->prefix.appId == (0x00000000)) && (remoteGuid->prefix.instanceId == (0x00000000)) && (remoteGuid->objectId != (0x00000000))) || ((remoteGuid->prefix.hostId != (0x00000000)) && (remoteGuid->prefix.appId != (0x00000000)) && (remoteGuid->prefix.instanceId != (0x00000000)) && (remoteGuid->objectId == (0x00000000))) )
DISCParticipantDiscoveryPlugin_removeRemoteParticipantI:!remove remote endpoints of remote participant: 0X10F3702,0XB5910000,0,0X1C1
PRESInterParticipant_removeRemoteParticipant:[Local Participant: ad33702 91b4 1] [Remote Participant: 1c1 10f3702 b5910000 | Remote Endpoint: 0] Could not remove remote endpoint
PRESInterParticipant_removeRemoteParticipant:[Local Participant: ad33702 91b4 1] [Remote Participant: 1c1 10f3702 b5910000 | Remote Endpoint: 0] Could not remove remote endpoint
PRESParticipant_removeRemoteParticipantFromBuiltinChannels:Failed to remove remote participant 10f3702 b5910000 0 to inter participant channel
PRESParticipant_removeRemoteParticipant:!remove remote participant from builtin channels
DISCParticipantDiscoveryPlugin_removeRemoteParticipantI:!remove remote participant: 0X10F3702,0XB5910000,0,0X1C1
DISCParticipantDiscoveryPlugin_removeRemoteParticipantsByCookie:!remove remote participant: 0X10F3702,0XB5910000,0,0X1C1
DISCSimpleParticipantDiscoveryPluginPDFListener_onBeforeLocalParticipantDeleted:!remove remote participant
```

#### Additional information

If you switch them (run talker as fastrtps and listener as connext) you still get output.

- if talker (fastrtps) is run first, and listener is killed first:

```
[INFO] [listener]: I heard: [Hello World: 3]
[INFO] [listener]: I heard: [Hello World: 4]
[INFO] [listener]: I heard: [Hello World: 5]
[INFO] [listener]: I heard: [Hello World: 6]
^C[INFO] [rclcpp]: signal_handler(signal_value=2)
PRESPsService_removeMatchingRemoteEndpoints:!precondition: me == ((void*)0) || worker == ((void*)0) || remoteGuid == ((void*)0) || !( ((remoteGuid->prefix.hostId == (0x00000000)) && (remoteGuid->prefix.appId == (0x00000000)) && (remoteGuid->prefix.instanceId == (0x00000000)) && (remoteGuid->objectId != (0x00000000))) || ((remoteGuid->prefix.hostId != (0x00000000)) && (remoteGuid->prefix.appId != (0x00000000)) && (remoteGuid->prefix.instanceId != (0x00000000)) && (remoteGuid->objectId == (0x00000000))) )
DISCParticipantDiscoveryPlugin_removeRemoteParticipantI:!remove remote endpoints of remote participant: 0X10F3702,0XCB910000,0,0X1C1
PRESInterParticipant_removeRemoteParticipant:[Local Participant: ad33702 91cc 1] [Remote Participant: 1c1 10f3702 cb910000 | Remote Endpoint: 0] Could not remove remote endpoint
PRESInterParticipant_removeRemoteParticipant:[Local Participant: ad33702 91cc 1] [Remote Participant: 1c1 10f3702 cb910000 | Remote Endpoint: 0] Could not remove remote endpoint
PRESParticipant_removeRemoteParticipantFromBuiltinChannels:Failed to remove remote participant 10f3702 cb910000 0 to inter participant channel
PRESParticipant_removeRemoteParticipant:!remove remote participant from builtin channels
DISCParticipantDiscoveryPlugin_removeRemoteParticipantI:!remove remote participant: 0X10F3702,0XCB910000,0,0X1C1
DISCParticipantDiscoveryPlugin_removeRemoteParticipantsByCookie:!remove remote participant: 0X10F3702,0XCB910000,0,0X1C1
DISCSimpleParticipantDiscoveryPluginPDFListener_onBeforeLocalParticipantDeleted:!remove remote participant
```

- if the listener (connext) is run first, and the talker is terminated first:

```
[INFO] [listener]: I heard: [Hello World: 1]
[INFO] [listener]: I heard: [Hello World: 2]
[INFO] [listener]: I heard: [Hello World: 3]
PRESPsService_removeMatchingRemoteEndpoints:!precondition: me == ((void*)0) || worker == ((void*)0) || remoteGuid == ((void*)0) || !( ((remoteGuid->prefix.hostId == (0x00000000)) && (remoteGuid->prefix.appId == (0x00000000)) && (remoteGuid->prefix.instanceId == (0x00000000)) && (remoteGuid->objectId != (0x00000000))) || ((remoteGuid->prefix.hostId != (0x00000000)) && (remoteGuid->prefix.appId != (0x00000000)) && (remoteGuid->prefix.instanceId != (0x00000000)) && (remoteGuid->objectId == (0x00000000))) )
DISCParticipantDiscoveryPlugin_removeRemoteParticipantI:!remove remote endpoints of remote participant: 0X10F3702,0XE3910000,0,0X1C1
PRESInterParticipant_removeRemoteParticipant:[Local Participant: ad33702 91e2 1] [Remote Participant: 1c1 10f3702 e3910000 | Remote Endpoint: 0] Could not remove remote endpoint
PRESInterParticipant_removeRemoteParticipant:[Local Participant: ad33702 91e2 1] [Remote Participant: 1c1 10f3702 e3910000 | Remote Endpoint: 0] Could not remove remote endpoint
PRESParticipant_removeRemoteParticipantFromBuiltinChannels:Failed to remove remote participant 10f3702 e3910000 0 to inter participant channel
PRESParticipant_removeRemoteParticipant:!remove remote participant from builtin channels
DISCParticipantDiscoveryPlugin_removeRemoteParticipantI:!remove remote participant: 0X10F3702,0XE3910000,0,0X1C1
DISCParticipantDiscoveryPlugin_removeRemoteParticipant:!remove remote participant: 0X10F3702,0XE3910000,0,0X1C1
DISCSimpleParticipantDiscoveryPluginReaderListener_onDataAvailable:!remove remote participant
^C[INFO] [rclcpp]: signal_handler(signal_value=2)
```

I am using a unique domain id and I used the RTI admin console to verify that there was not any unexpected peers.

wjwwood, 2018-12-13:
This doesn't appear to affect the return code, nor does it seem to affect the `test_communication` tests.

MiguelCompany, 2018-12-13:
@wjwwood It seems that Connext is being more restrictive on the guidPrefix than required by the standard. Please check with eProsima/Fast-RTPS#353

clalancette, 2023-03-30:
This particular issue doesn't seem to be there anymore, so closing this one out.

closed 2023-03-30 by clalancette