ros2/ros2 #623: ROS2 Fails to build qt_gui_cpp

igorrecioh, 2018-12-13:
## Bug report

**Required Info:**

- Operating System:
  - MacOS High Sierra 10.13.6
- Installation type:
  - From source
- Version or commit hash:
  - Master (December 12, 2018) -> Crystal
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - qt_gui_cpp

#### Steps to reproduce issue

Followed installation instructions provided  [here](https://index.ros.org/doc/ros2/OSX-Development-Setup/). When 
```
colcon build --symlink-install
```
is done, I get errors on qt_gui_cpp package.

#### Expected behavior
  - Compiled without problems
#### Actual behavior
  - This error:
```
Starting >>> qt_gui_cpp
--- stderr: qt_gui_cpp                                                                                                                                                      
Traceback (most recent call last):
  File "/Users/igorrecio/ros2_ws/install/python_qt_binding/share/python_qt_binding/cmake/sip_configure.py", line 83, in <module>
    config = Configuration()
  File "/Users/igorrecio/ros2_ws/install/python_qt_binding/share/python_qt_binding/cmake/sip_configure.py", line 19, in __init__
    [qmake_exe, '-query'], env=env, universal_newlines=True)
  File "/usr/local/Cellar/python/3.7.1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/subprocess.py", line 389, in check_output
    **kwargs).stdout
  File "/usr/local/Cellar/python/3.7.1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/subprocess.py", line 466, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/local/Cellar/python/3.7.1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/subprocess.py", line 769, in __init__
    restore_signals, start_new_session)
  File "/usr/local/Cellar/python/3.7.1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/subprocess.py", line 1516, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'qmake': 'qmake'
make[2]: *** [sip/qt_gui_cpp_sip/Makefile] Error 1
make[1]: *** [src/qt_gui_cpp_sip/CMakeFiles/libqt_gui_cpp_sip.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< qt_gui_cpp	[ Exited with code 2 ]
```
#### Additional information

----
Anyone facing something similar? Thanks in advance!

nuclearsandwich, 2018-12-13:
> Anyone facing something similar? Thanks in advance!

What is the output of `brew link -n qt` for you? It looks like qmake is missing from your path.

igorrecioh, 2018-12-13:
Hi @nuclearsandwich! This is what I get:

```
MacBook-Pro-de-Igor:ros2_ws igorrecio$ brew link -n qt
Would link:
/usr/local/bin/canbusutil
/usr/local/bin/fixqt4headers.pl
/usr/local/bin/lconvert
/usr/local/bin/lrelease
/usr/local/bin/lupdate
/usr/local/bin/macchangeqt
/usr/local/bin/macdeployqt
/usr/local/bin/moc
/usr/local/bin/qdbus
/usr/local/bin/qdbuscpp2xml
/usr/local/bin/qdbusxml2cpp
/usr/local/bin/qdistancefieldgenerator
/usr/local/bin/qgltf
/usr/local/bin/qhelpgenerator
/usr/local/bin/qlalr
/usr/local/bin/qmake
/usr/local/bin/qmlcachegen
/usr/local/bin/qmleasing
/usr/local/bin/qmlimportscanner
/usr/local/bin/qmllint
/usr/local/bin/qmlmin
/usr/local/bin/qmlplugindump
/usr/local/bin/qmlpreview
/usr/local/bin/qmlprofiler
/usr/local/bin/qmlscene
/usr/local/bin/qmltestrunner
/usr/local/bin/qscxmlc
/usr/local/bin/qtattributionsscanner
/usr/local/bin/qtdiag
/usr/local/bin/qtpaths
/usr/local/bin/qtplugininfo
/usr/local/bin/qvkgen
/usr/local/bin/qwebengine_convert_dict
/usr/local/bin/rcc
/usr/local/bin/repc
/usr/local/bin/syncqt.pl
/usr/local/bin/uic
/usr/local/bin/xmlpatterns
/usr/local/bin/xmlpatternsvalidator
/usr/local/include/Qt3DAnimation
/usr/local/include/Qt3DCore
/usr/local/include/Qt3DExtras
/usr/local/include/Qt3DInput
/usr/local/include/Qt3DLogic
/usr/local/include/Qt3DQuick
/usr/local/include/Qt3DQuickAnimation
/usr/local/include/Qt3DQuickExtras
/usr/local/include/Qt3DQuickInput
/usr/local/include/Qt3DQuickRender
/usr/local/include/Qt3DQuickScene2D
/usr/local/include/Qt3DRender
/usr/local/include/QtAccessibilitySupport
/usr/local/include/QtBluetooth
/usr/local/include/QtCharts
/usr/local/include/QtClipboardSupport
/usr/local/include/QtConcurrent
/usr/local/include/QtCore
/usr/local/include/QtDBus
/usr/local/include/QtDataVisualization
/usr/local/include/QtDesigner
/usr/local/include/QtDesignerComponents
/usr/local/include/QtDeviceDiscoverySupport
/usr/local/include/QtEdidSupport
/usr/local/include/QtEventDispatcherSupport
/usr/local/include/QtFbSupport
[... 336 lines cut ...]
/usr/local/Frameworks/QtRepParser.framework
/usr/local/Frameworks/QtScript.framework
/usr/local/Frameworks/QtScriptTools.framework
/usr/local/Frameworks/QtScxml.framework
/usr/local/Frameworks/QtSensors.framework
/usr/local/Frameworks/QtSerialBus.framework
/usr/local/Frameworks/QtSerialPort.framework
/usr/local/Frameworks/QtSql.framework
/usr/local/Frameworks/QtSvg.framework
/usr/local/Frameworks/QtTest.framework
/usr/local/Frameworks/QtTextToSpeech.framework
/usr/local/Frameworks/QtUiPlugin.framework
/usr/local/Frameworks/QtVirtualKeyboard.framework
/usr/local/Frameworks/QtWebChannel.framework
/usr/local/Frameworks/QtWebEngine.framework
/usr/local/Frameworks/QtWebEngineCore.framework
/usr/local/Frameworks/QtWebEngineWidgets.framework
/usr/local/Frameworks/QtWebSockets.framework
/usr/local/Frameworks/QtWebView.framework
/usr/local/Frameworks/QtWidgets.framework
/usr/local/Frameworks/QtXml.framework
/usr/local/Frameworks/QtXmlPatterns.framework

If you need to have this software first in your PATH instead consider running:
  echo 'export PATH="/usr/local/opt/qt/bin:$PATH"' >> ~/.bash_profile

nuclearsandwich, 2018-12-13:
In the past I can recall instructions indicating that we should not link Qt but when I went and looked at all our CI nodes they have the latest version of Qt linked. So I would recommend matching that while testing and maybe we'll learn why it was recommended against in the first place.

Alternatively, setting up PATH as suggested at the bottom of the brew link output would, along with the existing CMAKE_PREFIX_PATH that is documented, likely resolve the issue as well.

Karsten1987, 2018-12-13:
I do have this in my bash_profile:
`export PATH="/usr/local/opt/qt5/bin:$PATH"`

This definitely helps to find Qt5

nuclearsandwich, 2018-12-13:
> I do have this in my bash_profile:
> `export PATH="/usr/local/opt/qt5/bin:$PATH"`

We should add one or the other of making sure qt has been `brew link`'d or setting PATH as well as CMAKE_PREFIX_PATH gets added to the documentation.

igorrecioh, 2018-12-13:
Thanks for your support. Seems like there is something else wrong in my system:

```
MacBook-Pro-de-Igor:ros2_ws igorrecio$ colcon build --symlink-install --packages-select qt_gui_cpp
Starting >>> qt_gui_cpp
--- stderr: qt_gui_cpp                             
Traceback (most recent call last):
  File "/Users/igorrecio/ros2_ws/install/python_qt_binding/share/python_qt_binding/cmake/sip_configure.py", line 50, in get_sip_dir_flags
    sip_dir = config.pyqt_sip_dir
  File "/usr/local/lib/python3.7/site-packages/sipconfig.py", line 191, in __getattr__
    raise AttributeError("\"%s\" is not a valid configuration value or user option" % name)
AttributeError: "pyqt_sip_dir" is not a valid configuration value or user option

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/igorrecio/ros2_ws/install/python_qt_binding/share/python_qt_binding/cmake/sip_configure.py", line 85, in <module>
    sip_dir, sip_flags = get_sip_dir_flags(config)
  File "/Users/igorrecio/ros2_ws/install/python_qt_binding/share/python_qt_binding/cmake/sip_configure.py", line 65, in get_sip_dir_flags
    raise FileNotFoundError('The sip directory for PyQt5 could not be located. Please ensure' +
FileNotFoundError: The sip directory for PyQt5 could not be located. Please ensure that PyQt5 is installed
make[2]: *** [sip/qt_gui_cpp_sip/Makefile] Error 1
make[1]: *** [src/qt_gui_cpp_sip/CMakeFiles/libqt_gui_cpp_sip.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< qt_gui_cpp	[ Exited with code 2 ]
```

nuclearsandwich, 2018-12-13:
@Karsten1987 is meeting the same issue.

Karsten1987, 2018-12-13:
I in the end came around solving this by compiling PyQt with SIP myself. I am sure there are better solutions for it, but `brew` and `pip` seemed to be incomplete for this problem.

I basically followed these steps here.
http://pyqt.sourceforge.net/Docs/PyQt5/installation.html

The important part is that `SIP` has to be installed first and the `sip`-executable has to be linked when configuring `PyQt`.
The benefit of compiling it myself is that I don't have to rely on `brew`s python and can use my virtualenv.

Roughly the steps i did:

``` bash
wget https://sourceforge.net/projects/pyqt/files/sip/sip-4.19.13/sip-4.19.13.tar.gz
wget https://sourceforge.net/projects/pyqt/files/PyQt5/PyQt-5.11.3/PyQt5_gpl-5.11.3.tar.gz
# install SIP first
tar xzf sip-4.19.13.tar.gz
cd sip-4.19.13
python ./configure.py
make
make install
# install pyqt
tar xzf PyQt5_gpl-5.11.3.tar.gz
cd PyQt5_gpl-5.11.3
python ./configure.py --sip /Users/karsten/.pyenv/versions/3.7.0/Python.framework/Versions/3.7/bin/sip  # <-- This has to point to your sip install location
make -j4
make install
```

after that ipython is happy
``` python
import sipconfig
```

wjwwood, 2018-12-13:
@Karsten1987 (before your example of what works) you're installing sip from brew, but you appear to be using python from pyenv. In which case it doesn't surprise me that you need to build sip yourself to make it work.

Karsten1987, 2018-12-13:
where do I see that I am using sip from brew?

igorrecioh, 2018-12-13:
> I in the end came around solving this by compiling PyQt with SIP myself. I am sure there are better solutions for it, but `brew` and `pip` seemed to be incomplete for this problem.
> 
> I basically followed these steps here.
> http://pyqt.sourceforge.net/Docs/PyQt5/installation.html
> 
> The important part is that `SIP` has to be installed first and the `sip`-executable has to be linked when configuring `PyQt`.
> The benefit of compiling it myself is that I don't have to rely on `brew`s python and can use my virtualenv.
> 
> Roughly the steps i did:
> 
> ```shell
> wget https://sourceforge.net/projects/pyqt/files/sip/sip-4.19.13/sip-4.19.13.tar.gz
> wget https://sourceforge.net/projects/pyqt/files/PyQt5/PyQt-5.11.3/PyQt5_gpl-5.11.3.tar.gz
> # install SIP first
> tar xzf sip-4.19.13.tar.gz
> cd sip-4.19.13
> python ./configure.py
> make
> make install
> # install pyqt
> tar xzf PyQt5_gpl-5.11.3.tar.gz
> cd PyQt5_gpl-5.11.3
> python ./configure.py --sip /Users/karsten/.pyenv/versions/3.7.0/Python.framework/Versions/3.7/bin/sip  # <-- This has to point to your sip install location
> make -j4
> make install
> ```
> 
> after that ipython is happy
> 
> ```python
> import sipconfig
> ```

Thanks for that, @Karsten1987, but I don't use python from pyenv... Will wait for another kind of solution if possible ;)

Karsten1987, 2018-12-13:
@igorrecioh Sorry, maybe I wasn't precise. The instructions will also use with whatever python installation you use. Only make sure that when calling `python ./configure.py --sip` the filepath to your sip module is correct.

wjwwood, 2018-12-13:
> where do I see that I am using sip from brew?

What I meant is that if you were previously following the official install instructions, then you'd be mixing sip from brew and python from pyenv, which would lead you to need to build sip manually.

wjwwood, 2018-12-13:
Whereas, you may not need to build sip manually or pyqt (specifying `--sip` to configure) if you installed everything from brew.

igorrecioh, 2018-12-13:
> @igorrecioh Sorry, maybe I wasn't precise. The instructions will also use with whatever python installation you use. Only make sure that when calling `python ./configure.py --sip` the filepath to your sip module is correct.

Understood now! Thanks!

wjwwood, 2018-12-13:
@Karsten1987 I'm going to test the brew python + brew sip/pyqt5 + fat archive setup as soon as I can. Currently blocked because I'm doing a source build on tinyxml2 7.0 and I need the other 6.something to use the fat archive.

wjwwood, 2018-12-14:
Ok, so using brew python and brew sip/pyqt and following exactly the install instructions allows me to build everything and also run `rqt`.

nuclearsandwich, 2018-12-14:
> Ok, so using brew python and brew sip/pyqt and following exactly the install instructions allows me to build everything and also run `rqt`.

/me throws hands up in anguish

igorrecioh, 2018-12-14:
Tried what @Karsten1987 said, compiled sip and pyqt but still ROS2 building failed...

brawner, 2018-12-15:
> Not working for me, but thanks for the answer, @brawner

That’s your issue then. You need sip and pyqt5 to build qt_gui_cpp. Install sip and pyqt5 with homebrew. If they are already installed then your pythonpath probably doesn’t match your homebrew install.

igorrecioh, 2018-12-16:
I installed both sip and pyqt5 with homebrew. BTW, I have uninstalled homebrew completely and reinstalled all the stuff following the instructions provided [here](https://index.ros.org/doc/ros2/OSX-Development-Setup/).

Now I get this:
```
Starting >>> test_osrf_testing_tools_cpp
--- stderr: qt_gui_cpp                                                                                                                                
CMake Warning at /Users/igorrecio/ros2_ws/install/python_qt_binding/share/python_qt_binding/cmake/sip_helper.cmake:17 (message):
  SIP binding generator NOT available.
Call Stack (most recent call first):
  src/qt_gui_cpp_sip/CMakeLists.txt:54 (include)

CMake Error at src/CMakeLists.txt:10 (message):
  No Python binding generator found.

---
Failed   <<< qt_gui_cpp	[ Exited with code 1 ]
```

brawner, 2018-12-18:
So this error message pops up when python_qt_binding's sip_helper.cmake cannot find the PyQt5 directory. Typically this means PyQt5 is not installed, but in this case it looks like it was installed in a different location. Do a search on your system for where the following are and post the results.

python3
```where -a python3```

The sip executable
```where -a sip```

It's version
```sip -V```

Where does sip thing PyQt5 files should be installed
```python3 -c "import sipconfig; print(sipconfig._pkg_config['default_sip_dir'])"```

Do a spotlight search for where PyQt5's sip files are located, specifically ```QtCoremod.sip```

igorrecioh, 2018-12-18:
> `QtCoremod`

Hi again @brawner. Thanks again for your answer. This is my output:
```
MacBook-Pro-de-Igor:ros2_ws igorrecio$ type -a python3
python3 is /usr/local/bin/python3
MacBook-Pro-de-Igor:ros2_ws igorrecio$ type -a sip
sip is /usr/local/bin/sip
MacBook-Pro-de-Igor:ros2_ws igorrecio$ sip -V
4.19.8
MacBook-Pro-de-Igor:ros2_ws igorrecio$  python3 -c "import sipconfig; print(sipconfig._pkg_config['default_sip_dir'])"
/usr/local/share/sip
```

When I search QtCoremod.sip on Spotlight, it is not found...

brawner, 2018-12-18:
That looks like a typical homebrew setup. You have sip installed correctly, but it doesn't look like PyQt5 is correctly installed. Check the contents of ```/usr/local/share/sip```, it should have a directory called ```Qt5```. If not, check where homebrew installed PyQt5. Post the output of the following:

```brew --prefix pyqt5```

Try running a reinstall of PyQt5
```brew install --force pyqt5```

igorrecioh, 2018-12-18:
> That looks like a typical homebrew setup. You have sip installed correctly, but it doesn't look like PyQt5 is correctly installed. Check the contents of `/usr/local/share/sip`, it should have a [...]

Empty directory /usr/local/share/sip/:
```
MacBook-Pro-de-Igor:ros2_ws igorrecio$ cd /usr/local/share/sip/
MacBook-Pro-de-Igor:sip igorrecio$ ls

MacBook-Pro-de-Igor:sip igorrecio$ brew --prefix pyqt5
/usr/local/opt/pyqt
```

More:

```
MacBook-Pro-de-Igor:sip igorrecio$  brew install --force pyqt5
Warning: pyqt 5.10.1_1 is already installed, it's just not linked
You can use `brew link pyqt` to link this version.
MacBook-Pro-de-Igor:sip igorrecio$ brew link pyqt
Linking /usr/local/Cellar/pyqt/5.10.1_1... 
Error: Could not symlink bin/pyrcc5
Target /usr/local/bin/pyrcc5
already exists. You may want to remove it:
  rm '/usr/local/bin/pyrcc5'

To force the link and overwrite all conflicting files:
  brew link --overwrite pyqt

To list all files that would be deleted:
  brew link --overwrite --dry-run pyqt
MacBook-Pro-de-Igor:sip igorrecio$ brew --prefix pyqt5
shell-init: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
Error: The current working directory doesn't exist, cannot proceed.
```

igorrecioh, 2018-12-18:
No success....
```
Starting >>> qt_gui_app
--- stderr: qt_gui_cpp                                                                                                                                                      
CMake Error at src/qt_gui_cpp/CMakeLists.txt:3 (find_package):
  By not providing "FindQt5.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Qt5", but
  CMake did not find one.

  Could not find a package configuration file provided by "Qt5" with any of
  the following names:

    Qt5Config.cmake
    qt5-config.cmake

  Add the installation prefix of "Qt5" to CMAKE_PREFIX_PATH or set "Qt5_DIR"
  to a directory containing one of the above files.  If "Qt5" provides a
  separate development package or SDK, be sure it has been installed.

---
Failed   <<< qt_gui_cpp	[ Exited with code 1 ]
```

brawner, 2018-12-18:
Is Qt in your cmake prefix path as suggested in the [osx-development-setup](https://github.com/ros2/ros2_documentation/blob/master/OSX-Development-Setup.rst)?

```
# Add the Qt directory to the PATH and CMAKE_PREFIX_PATH
export CMAKE_PREFIX_PATH=$CMAKE_PREFIX_PATH:/usr/local/opt/qt
export PATH=$PATH:/usr/local/opt/qt/bin
```

igorrecioh, 2018-12-18:
```
MacBook-Pro-de-Igor:ros2_ws igorrecio$ colcon build --packages-select qt_gui_cpp
Starting >>> qt_gui_cpp
--- stderr: qt_gui_cpp                               
In file included from /Users/igorrecio/ros2_ws/src/ros-visualization/qt_gui_core/qt_gui_cpp/src/qt_gui_cpp/recursive_plugin_provider.cpp:33:
In file included from /Users/igorrecio/ros2_ws/src/ros-visualization/qt_gui_core/qt_gui_cpp/include/qt_gui_cpp/recursive_plugin_provider.h:37:
In file included from /Users/igorrecio/ros2_ws/src/ros-visualization/qt_gui_core/qt_gui_cpp/include/qt_gui_cpp/ros_pluginlib_plugin_provider_for_plugin_providers.h:37:
/Users/igorrecio/ros2_ws/src/ros-visualization/qt_gui_core/qt_gui_cpp/include/qt_gui_cpp/ros_pluginlib_plugin_provider.h:301:126: error: no member named 'ErrorStr' in 'tinyxml2::XMLDocument'
      qWarning("RosPluginlibPluginProvider::parseManifest() could not load manifest \"%s\" (%s)", manifest_path.c_str(), doc.ErrorStr());
                                                                                                                         ~~~ ^
1 error generated.
make[2]: *** [src/qt_gui_cpp/CMakeFiles/qt_gui_cpp.dir/recursive_plugin_provider.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [src/qt_gui_cpp/CMakeFiles/qt_gui_cpp.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< qt_gui_cpp	[ Exited with code 2 ]
```
No luck today...

brawner, 2018-12-18:
You're hitting every issue possible. This error occurs because you don't have an up-to-date version of Tinyxml2

What does the following produce?
```brew list --versions tinyxml2```

The binary install requires Tinyxml2 v6.2, but I've been able to build from source with 2.7 as well.
```brew remove tinyxml2
brew install osrf/simulation/tinyxml2@6.2.0
brew link --force tinyxml2@6.2.0
```

igorrecioh, 2018-12-18:
```
MacBook-Pro-de-Igor:ros2_ws igorrecio$  brew list --versions tinyxml2
tinyxml2 5.0.1
```
I installed this version because of this issue #440. I will try what you recommend... Sorry for this and thank you again for your support, I really appreciate

igorrecioh, 2018-12-18:
No way:
```
--- stderr: qt_gui_cpp                                                                                                                                         
Traceback (most recent call last):
  File "/Users/igorrecio/ros2_ws/install/python_qt_binding/share/python_qt_binding/cmake/sip_configure.py", line 83, in <module>
    config = Configuration()
  File "/Users/igorrecio/ros2_ws/install/python_qt_binding/share/python_qt_binding/cmake/sip_configure.py", line 19, in __init__
    [qmake_exe, '-query'], env=env, universal_newlines=True)
  File "/usr/local/Cellar/python/3.7.1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/subprocess.py", line 389, in check_output
    **kwargs).stdout
  File "/usr/local/Cellar/python/3.7.1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/subprocess.py", line 466, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/local/Cellar/python/3.7.1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/subprocess.py", line 769, in __init__
    restore_signals, start_new_session)
  File "/usr/local/Cellar/python/3.7.1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/subprocess.py", line 1516, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'qmake': 'qmake'
make[2]: *** [sip/qt_gui_cpp_sip/Makefile] Error 1
make[1]: *** [src/qt_gui_cpp_sip/CMakeFiles/libqt_gui_cpp_sip.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< qt_gui_cpp	[ Exited with code 2 ]
```

I don't want to disturb you anymore. I will go back to ROS2 Bouncy until an official solution is given :) 
Thank you for your patience!

brawner, 2018-12-18:
It sounds like qmake is not in  ```/usr/local/opt/qt/bin```, I would update that part of your path variable

igorrecioh, 2018-12-18:
Success!!! This is what I made:
```
MacBook-Pro-de-Igor:ros2_ws igorrecio$ echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/X11/bin
MacBook-Pro-de-Igor:ros2_ws igorrecio$ brew link qt5 --force
Linking /usr/local/Cellar/qt/5.12.0... 423 symlinks created

If you need to have this software first in your PATH instead consider running:
  echo 'export PATH="/usr/local/opt/qt/bin:$PATH"' >> ~/.bash_profile
MacBook-Pro-de-Igor:ros2_ws igorrecio$ export PATH="/usr/local/opt/qt/bin:$PATH"
MacBook-Pro-de-Igor:ros2_ws igorrecio$ echo $PATH
/usr/local/opt/qt/bin:/usr/local/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/X11/bin
```
Then,
```
colcon build --packages-select qt_gui_cpp
```

Now yes, I will build ROS2 Crystal! Keep you informed. Hope this long issue helps anyone else. Thanks a lot, @brawner

nuclearsandwich, 2018-12-19:
I'm glad you were able to get things working @igorrecioh. The current version of the MacOS development setup sets up Qt with the CMAKE_PREFIX_PATH and PATH without forcing brew to link it. https://index.ros.org/doc/ros2/OSX-Development-Setup/ and I would advise most building ROS 2 for MacOS to try that first before `brew link qt5 --force`.

closed 2018-12-19 by nuclearsandwich

mattsara24, 2020-05-08:
What was the solution to this issue? I am still encountering it

mattsara24, 2020-05-08:
Is there a definitive solution to this problem? When I use ``brew link qt5 --force``  I get the following issue on MacOS trying to build qt_gui_cpp

``Traceback (most recent call last):
  File "/Users/mattsaraceno/ROS2/install/python_qt_binding/share/python_qt_binding/cmake/sip_configure.py", line 83, in <module>
    config = Configuration()
  File "/Users/mattsaraceno/ROS2/install/python_qt_binding/share/python_qt_binding/cmake/sip_configure.py", line 19, in __init__
    [qmake_exe, '-query'], env=env, universal_newlines=True)
  File "/usr/local/Cellar/python/3.7.7/Frameworks/Python.framework/Versions/3.7/lib/python3.7/subprocess.py", line 411, in check_output
    **kwargs).stdout
  File "/usr/local/Cellar/python/3.7.7/Frameworks/Python.framework/Versions/3.7/lib/python3.7/subprocess.py", line 488, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/local/Cellar/python/3.7.7/Frameworks/Python.framework/Versions/3.7/lib/python3.7/subprocess.py", line 800, in __init__
    restore_signals, start_new_session)
  File "/usr/local/Cellar/python/3.7.7/Frameworks/Python.framework/Versions/3.7/lib/python3.7/subprocess.py", line 1551, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
PermissionError: [Errno 13] Permission denied: 'qmake'
``

mdx8, 2022-01-09:
colcon build --symlink-install --packages-ignore qt_gui_cpp rqt_gui_cpp