ros2/ros2 #627: Build reult on win10, succeed but have some problems

tty2099, 2018-12-15:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10 x64, Microsoft visual studio 2017 community x64
- Installation type:
  - build from source
- Version or commit hash:
  - lastest crystall release source
- DDS implementation:
  - default
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
just follow the official steps

## Problems
1.  patch.exe, the version installed by the choco is not work, I use the patch.exe been installed from the MSYS2 (need to modify the path_exe variable in the corresponding CMakeLists.txt) 
2. final warning after my build the latest crystal release source
    Summary: 228 packages finished [39min 58s]
  10 packages had stderr output: rmw_connext_cpp rmw_connext_shared_cpp rmw_opensplice_cpp ros1_bridge rosidl_typesupport_connext_c rosidl_typesupport_connext_cpp rosidl_typesupport_opensplice_c rosidl_typesupport_opensplice_cpp rqt test_launch_ros

wjwwood, 2018-12-17:
> patch.exe, the version installed by the choco is not work, I use the patch.exe from the MSYS2 (need to modify the path_exe variable in the corresponding CMakeLists.txt)

You need to run your shell as administrator for patch from choco to work. It is noted in the install instructions:

> To build ROS 2 you will need a Visual Studio Command Prompt (usually titled “x64 Native Tools Command Prompt for VS 2017” for bouncy and later or “x64 Native Tools Command Prompt for VS 2015” for ardent and earlier) running as Administrator.
-- https://index.ros.org/doc/ros2/Windows-Development-Setup/#building-the-ros-2-code

Also:

https://index.ros.org/doc/ros2/Windows-Development-Setup/#patch-exe-opens-a-new-command-window-and-asks-for-administrator

wjwwood, 2018-12-17:
> final warning after my build the latest crystal release source
Summary: 228 packages finished [39min 58s]
10 packages had stderr output: rmw_connext_cpp rmw_connext_shared_cpp rmw_opensplice_cpp ros1_bridge rosidl_typesupport_connext_c rosidl_typesupport_connext_cpp rosidl_typesupport_opensplice_c rosidl_typesupport_opensplice_cpp rqt test_launch_ros

Why is this a problem? In each of those cases, I think it's simply warnings that they're not supported on Windows and so they gracefully decline to build.

tty2099, 2018-12-17:
yes，I certainly know, and I have follow the same steps

发自我的 iPhone

wjwwood, 2018-12-18:
> call install\local_steup.bat is very slow(about 30sec)

This is a known issue, though I cannot find a single issue to capture this problem. I would suggest opening an issue on the `colcon/colcon-core` repository (it may get redirected to `ament_cmake` or `ament_package`, but I'm not sure).

> and throw the following warnings:
not found: "C:\Development\ros2\install\share/rttest/local_setup.bat"
not found: "C:\Development\ros2\install\share/tlsf/local_setup.bat"
not found: "C:\Development\ros2\install\share/tlsf_cpp/local_setup.bat"
not found: "C:\Development\ros2\install\share/pendulum_control/local_setup.bat"

This is a known issue, and was reported here:

https://github.com/colcon/colcon-core/issues/71

But it was supposed to be fixed already, make sure you have the latest colcon version and then if it still occurs consider opening an issue on the colcon repository or commenting on that closed issue.

wjwwood, 2018-12-18:
> rviz2 starting message:

For sure this is an issue. Can you open an issue on `ros2/rviz`?

wjwwood, 2018-12-18:
I'm going to close this for now, because each of these issues can be reopened on different repositories (hopefully I've given you enough information to move them).

Feel free to continue discussion here, and if need I can reopen this.

closed 2018-12-18 by wjwwood