ros2/ros2 #630: Documentation on install ros2-base from source

mturnshek, 2018-12-18:
## Feature request
Instructions for installing ros2-base from source.

#### Feature description
Documentation with a list of the essential packages for each distribution. This would allow a user to install precisely the packages they want, from source, by ignoring all auxiliary packages. This is important, because currently installing from source requires many gigabytes, but not all ros2 packages are used by everyone.

#### Implementation considerations
Alternatively, instructions for how to compile additional packages with new or modified code on top of an existing debian package installation.

clalancette, 2018-12-18:
> Alternatively, instructions for how to compile additional packages with new or modified code on top of an existing debian package installation.

FWIW, this is possible today by installing from binaries, cloning the set of repositories you want to modify, modifying them, building, and sourcing the resulting `install/setup.bash`.  It's almost identical to how it would work in ROS 1, but I agree that more documentation about overlaying like this could be helpful for ROS 2.

nuclearsandwich, 2018-12-18:
[rosinstall_generator](https://github.com/ros-infrastructure/rosinstall_generator/) can be used with the `--upstream-development` flag to create a rosinstall file that includes only ros-base. As packages start to diverge for Dashing, this method will be out-of-sync using crystal as the target rosdistro but it's suitable for the time being.

goekce, 2020-08-31:
As a newbie I was also searching for information to build only `ros-base` from scratch for Raspberry Pi. The ROS2 documentation only mentions downloading https://github.com/ros2/ros2/blob/master/ros2.repos, which probably contains the `desktop` (variant)[https://github.com/ros2/variants]. It would be great if the instructions for compiling the packages for an embedded device could be included. For example, compiling `desktop` variant on a Raspi 3 with 1GB RAM fails when compiling the `rviz` tools.

I generated and cloned the dependencies for `ros-base` using [rosinstall_generator](http://wiki.ros.org/rosinstall_generator):
```
rosinstall_generator ros_base --deps --rosdistro foxy | vcs import src/
```
Note that [some users had problems with rosinstall_generator](https://answers.ros.org/question/319587/ros2-crystal-minimalistic-built-from-sources/).

clalancette, 2020-09-17:
I'm actually going to move this over to the ros2_documentation repository, which is where any documentation for this would reside.  I'll also say that any help with doing this documentation would be appreciated.  It's on our backlog, but we may not get to it anytime soon.