ros2/ros2 #632: ros2.repos in master is using bouncy for ros/ros_environment

juanrh, 2018-12-20:
## Bug report

**Required Info:**

- Operating System:
Ubuntu 18: running a Docker container for ubuntu:bionic-20181112 on an Ubuntu 16 host 

- Installation type:
source 

- Version or commit hash:
 wget https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos 

- DDS implementation:
N/A
- Client library (if applicable):
N/A

#### Steps to reproduce issue
Follow the instructions on https://index.ros.org/doc/ros2/Linux-Development-Setup/

#### Expected behavior
```
root@ub80c134adc2254:/# export ROS_DISTRO=crystal
root@ub80c134adc2254:/# source /opt/ros/crystal/setup.bash 
root@ub80c134adc2254:/# echo $ROS_DISTRO 
crystal
root@ub80c134adc2254:/# 
```

#### Actual behavior
```
root@ub80c134adc2254:/# export ROS_DISTRO=crystal
root@ub80c134adc2254:/# source /opt/ros/crystal/setup.bash 
ROS_DISTRO was set to 'crystal' before. Please make sure that the environment does not mix paths from different distributions.
root@ub80c134adc2254:/# echo $ROS_DISTRO 
bouncy
root@ub80c134adc2254:/# 
```

#### Additional information
Also note 

```
root@ub80c134adc2254:~/ros2_ws/src# grep -rni bouncy .
./ros2/rosidl_python/.git/packed-refs:3:c2a54d1a052eab392c060ad0ed63e96f94acf4a1 refs/remotes/origin/bouncy
./ros2/common_interfaces/.git/packed-refs:3:de2b1335a52a8301f3a05606736eb6a2b9ee8580 refs/remotes/origin/bouncy
./ros2/rcutils/.git/packed-refs:4:244fa98ad6421ed3d6295bd157ec4b73400fab52 refs/remotes/origin/bouncy
...
root@ub80c134adc2254:~/ros2_ws/src# grep -rni crystal .
./ros2/rviz/.git/packed-refs:4:31b35f99b870421f0780480276a0a71e67aaa7e0 refs/remotes/origin/crystal
./ros/ros_environment/.git/packed-refs:4:2333be6972e22e5fdcbed78049bf7c5c40186463 refs/remotes/origin/crystal
...
```

so there is a mixture of bouncy and crystal packages

I think the problem is that  https://github.com/ros2/ros2/commits/master is missing commit 5e812b7 from https://github.com/ros2/ros2/commits/crystal that sets ros/ros_environment to version 2.2.0, which includes the commit https://github.com/ros/ros_environment/commit/e4d31cc683385fa2bbd4fe9de2645d1222fb9f77 that does set(ROS_DISTRO "crystal") instead of set(ROS_DISTRO "bouncy"). Without that commit ENV ROS_DISTRO crystal is overridden in the middle of the build, as far as I understand

juanrh, 2018-12-20:
This is probably fixed by https://github.com/ros2/ros2/pull/631

wjwwood, 2018-12-20:
Fixed by #631

closed 2018-12-20 by wjwwood