ros2/ros2 #633: Is there any way to run some custom code before "colcon build" runs setuptools?

losemyheaven, 2018-12-25:
I hope to colcon build run some custom code, before it runs setup.py.

Is there any hook or some magic ways to make it?

Thanks in advance.

lyicecream1012, 2018-12-25:
I have the same problem, anyone can help??

dirk-thomas, 2018-12-26:
There is no hook to be run before the `setup.py` file is being invoked. Conceptionally I would consider such a hook within `colcon` undesired since you want to be able to build a package in the same way when not using `colcon` as the build tool, e.g. by just invoking `python setup.py install|develop`.

I am not sure what additional behavior you would like to see but in general I would suggest to add it to your `setup.py` file. There are a few packages which already do something custom in their `setup.py` file:

* applying a patch file in [colcon-core](https://github.com/colcon/colcon-core/blob/832799c182bc6ac4f4b032fe2cdb953ef8486450/setup.py) and [colcon-ros](https://github.com/colcon/colcon-ros/blob/08ea63055c02f92ee3ac18a738f63ee39dd0a5e0/setup.py)
* installing additional files in custom locations in [colcon-argcomplete](https://github.com/colcon/colcon-argcomplete/blob/0d7f45b2bf61dbc8d90d8083043e7be7108fca17/setup.py) and [colcon-notification](https://github.com/colcon/colcon-notification/blob/e51396d85c24c21a57077eebb8c9a0a315e2a4e5/setup.py)

I will go ahead and close this ticket for now since I don't think such a hook should be added to `colcon`. Please feel free to continue commenting and it can always be reconsidered / reopened.

closed 2018-12-26 by dirk-thomas

spiderkeys, 2019-03-12:
We also have a use-case for this in our C++ projects, as we use Conan for many of our C++ dependencies and generally need to run "conan install" for each project before running build. 

"conan install" generates a conanbuildinfo.cmake file which has a collection of variables describing the targets that the main CMake project intends to import.

ahamez, 2022-10-25:
@dirk-thomas Actually, like @spiderkeys, we need to call Conan before colcon calls CMake. Can this hook be reconsidered?

spiderkeys, 2023-02-26:
Pinging this issue again, as now that Conan 2.0 has been released, and the cmake-conan approach of installing dependencies as a step within your CMakeLists.txt is no longer available/ever planned to be supported again, there is no longer a way to make sure each package can install conan dependencies before being built when using colcon.

Open to other suggestions of how to support this type of logic, if there are suggestions...